coolsnowwolf/lede #9403: 5.15 内核，ifup wan 几率导致 br-lan 一并重启引发无法上网

violarulan, 2022-05-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

本地环境：v4/v6 双栈公网 ip，手动删除 wan6 接口，在 br-lan 手动设定 /64 的 ipv6 分配，开启 openclash 和 zerotier，未开启 mwan/负载均衡

宽带掉线/手动重拨（/sbin/ifup wan）时，有几率（不是100%触发）同时把 br-lan 也重启，引发无法上网的问题

症状表现为：pppoe wan 口可以正常获取 ipv4 地址，br-lan 未获取到 ipv6 地址（此时 luci 首页 wan6 状态也是未连接），路由器自身和下边设备均无网络

 ### (2) 路由器型号和固件版本

A：

R4S
OpenWrt R22.5.5 / LuCI Master (git-22.121.65028-2a5da72)

### (3) 详细日志

A：

（以下为附近时间的内核日志）
eth0：wan
eth1、eth2：br-lan 桥接

```
43797.764181] br-lan: port 2(eth2) entered disabled state
[43797.764773] br-lan: port 1(eth1) entered disabled state
[43797.770711] device eth2 left promiscuous mode
[43797.771260] br-lan: port 2(eth2) entered disabled state
[43797.797148] device eth1 left promiscuous mode
[43797.797612] br-lan: port 1(eth1) entered disabled state
[43797.886184] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
[43797.972828] cma: cma_alloc: reserved: alloc failed, req-size: 4 pages, ret: -12
[43797.973525] cma: cma_alloc: reserved: alloc failed, req-size: 4 pages, ret: -12
[43797.975399] eth1: 0xffff800009850000, be:52:f6:26:97:1c, IRQ 87
[43798.033509] br-lan: port 1(eth1) entered blocking state
[43798.033980] br-lan: port 1(eth1) entered disabled state
[43798.034621] device eth1 entered promiscuous mode
[43798.037781] br-lan: port 1(eth1) entered blocking state
[43798.038248] br-lan: port 1(eth1) entered forwarding state
[43798.382294] br-lan: port 2(eth2) entered blocking state
[43798.382778] br-lan: port 2(eth2) entered disabled state
[43798.383444] device eth2 entered promiscuous mode
[43798.383978] br-lan: port 2(eth2) entered blocking state
[43798.384447] br-lan: port 2(eth2) entered forwarding state
[43798.434507] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=85)
[43798.436562] cma: cma_alloc: reserved: alloc failed, req-size: 2 pages, ret: -12
[43798.437291] cma: cma_alloc: reserved: alloc failed, req-size: 2 pages, ret: -12
[43798.438119] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[43798.478844] ax88179_178a 8-1:1.0 eth2: ax88179 - Link status is: 1
[43799.983048] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[43799.983769] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[43799.984396] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[43799.986691] br-lan: port 1(eth1) entered disabled state
[43799.987744] br-lan: port 2(eth2) entered disabled state
[43801.806856] ax88179_178a 8-1:1.0 eth2: ax88179 - Link status is: 1
[43801.819957] br-lan: port 2(eth2) entered blocking state
[43801.820452] br-lan: port 2(eth2) entered forwarding state
[43802.132852] r8168: eth1: link up
[43802.133349] br-lan: port 1(eth1) entered blocking state
[43802.133818] br-lan: port 1(eth1) entered forwarding state
[43802.738702] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[43802.739553] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[43807.997905] pppoe-wan: renamed from ppp0
```

coolsnowwolf, 2022-05-17:
这里没看出什么问题

shu0802, 2022-05-18:
> 这里没看出什么问题

双线多拨系统日志里全是这个内容
[image]

violarulan, 2022-05-21:
> 我也出现了这样的问题，但是只有一次，凌晨全部断网，还需再观察会不会复现

经过三天的尝试，发现关闭 upnp 服务即恢复正常

closed 2022-06-10 by violarulan

yishion819, 2022-11-03:
> 已找到解决方法，经过搜寻资料得知我的设备是无 EEPROM 的 R4C，所有网口的 Mac 地址都是一样的，是的都是相同，在设置里手动指定一个网口的 Mac 让它不一样就解决问题了。

请问是进后台用vi进行修改么？luci界面没有设置mac的地方，谢谢

violarulan, 2022-11-05:
> > 已找到解决方法，经过搜寻资料得知我的设备是无 EEPROM 的 R4C，所有网口的 Mac 地址都是一样的，是的都是相同，在设置里手动指定一个网口的 Mac 让它不一样就解决问题了。
> 
> 请问是进后台用vi进行修改么？luci界面没有设置mac的地方，谢谢

@yishion819 接口-LAN-高级设置-重设MAC地址

yishion819, 2022-11-05:
> > > 已找到解决方法，经过搜寻资料得知我的设备是无 EEPROM 的 R4C，所有网口的 Mac 地址都是一样的，是的都是相同，在设置里手动指定一个网口的 Mac 让它不一样就解决问题了。
> 
> > 
> 
> > 请问是进后台用vi进行修改么？luci界面没有设置mac的地方，谢谢
> 
> 
> 
> @yishion819 接口-LAN-高级设置-重设MAC地址

LUCI没有这个是不是要用vi去修改？