coolsnowwolf/lede #9409: 连接无线WIFI的客户端看不到了

chr09670, 2022-05-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天安装了当前最新版本的系统后，无线模块下看不到连接的客户端了；用的上一个版本是两年前编译的，下面会显示连接到2.4g和5g的客户端，感觉还是挺方便的。怎样能让它显示出来，或者有其它包可以有这个功能吗？
[image]

### (2) 路由器型号和固件版本

A：Newifi-D2  OpenWrt R22.5.5 / LuCI Master (git-22.136.55852-6698dc8)

### (3) 详细日志

A：

EnnawYang, 2022-05-18:
目前的闭源驱动是这样的

wp-chen, 2022-05-19:
请问解决了嘛？应该是需要开启某个编译选项，默认不编译所以看不到了

winneis, 2022-05-31:
不能用了
我现在用的是这个无线网卡加openwrtQualcomm Atheros QCA9880 802.11nac (radio0)

coolsnowwolf, 2026-09-05:
当前 LuCI 无线状态页面已通过 ubus 无线关联列表显示客户端；旧版页面看不到客户端的问题不再对应现行实现，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf