coolsnowwolf/lede #9413: 新源码砍掉防火墙的全锥形NAT了吗

lisonglin980, 2022-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
新源码砍掉防火墙的全锥形NAT了吗
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
Turbo ACC里面的全锥形NAT对x86没效果，但是防火墙里面打开是有效果的，前几天的版本防火墙里打开全锥形NAT使用NatTypeTester检测网络是全锥形，现在新版本里防火墙中没有这个设置后，用Turbo ACC无法实现全锥形NAT
### (1) 具体问题

A：
[image]
[image]

### (2) 路由器型号和固件版本

A：x86 git-22.138.16277-772b531

### (3) 详细日志

A：

hcym, 2022-05-20:
昨天的看到这个，不知道有什么用，x86

dsyo2008, 2022-05-21:
表示正常，防火墙已经去掉这NAT，Turbo ACC开启NAT。
[image]

Never2333, 2022-05-24:
同样问题... turbo acc里打开nat 1之后再测还是nat 3

WYC-2020, 2022-06-10:
如果怀疑是那个的问题，就在luci里吧对luci-app-turbo的修改撤销，然后编译再试，我是没有去掉防火墙哪里的设置的

coolsnowwolf, 2026-09-05:
全锥形 NAT 依赖特定内核/防火墙实现；当前树仍包含 nftables fullcone 表达式支持，但旧版 Turbo ACC/iptables LuCI 选项不能代表现行 firewall4 配置。报告没有当前防火墙规则和内核支持对照，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf