coolsnowwolf/lede #9415: R5S出来了，3口，已下单

ostrichii, 2022-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
r5s出来了，已上车，等发货，蹲固件
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

ForgotFun, 2022-05-20:
佐大的进阶班22讲以Rockchip为例子讲解了如何移植OpenWrt，参加佐大培训班自己学习移植OpenWrt吧。
B站上有试听课 [https://www.bilibili.com/video/BV1wA4y1D7wf](url)

Hiyajomaho-num9, 2022-05-20:
另请高明

1054009064, 2022-05-20:
找他们代理给固件就行，杂交固件它不香不好用了吗？

Aprikoze, 2022-05-20:
为什么不找BIGDONGDONG要固件

okyipeng, 2022-05-20:
我偷偷告诉你有一个大神叫**bigdongdong**，他的固件贼牛逼的！
记住固件没有“by bigdongdong”的都是垃圾！

jimmy60-di, 2022-05-20:
万物基于bigdongdong......

shousiji, 2022-05-20:
> 佐大的进阶班22讲以Rockchip为例子讲解了如何移植OpenWrt，参加佐大培训班自己学习移植OpenWrt吧。
> B站上有试听课 https://www.bilibili.com/video/BV1wA4y1D7wf

大佐yyds

AmadeusGhost, 2022-05-20:
固件开发费五位数 RMB，给开发组打钱或者报名佐大培训班自己适配，你自己选一个。
[image]

littlegunboom, 2022-05-20:
还是找BCIE吧，这个靠谱🤡

HHJLKK, 2022-05-20:
> 佐大的进阶班22讲以Rockchip为例子讲解了如何移植OpenWrt，参加佐大培训班自己学习移植OpenWrt吧。 B站上有试听课 https://www.bilibili.com/video/BV1wA4y1D7wf

大佐yyds

hxlls, 2022-05-20:
比尔麦克，你值得拥有，速度找个BCIE认证固件，顺便找比尔大师开光
[image]
[image]

anmonick, 2022-05-20:
不要在这里提这种傻逼问题搞污染，你可以看看别人有问这种问题的吗。哪买的找哪去

closed 2022-05-20 by ostrichii

iKira, 2022-06-15:
> 
有什么典故你们可以讲出来，别人也不知道，仗着自己知道的多点就这么埋汰人，也不是很体面吧？
看了上面回复，大概率是那个商家拿了代码打上自己的logo吧，你们明明可以光明正大的说出来，为什么要对一个不知情的人进行攻击呢？凸显出你们的优越感？