coolsnowwolf/lede #9422: x86_64 pve虚拟机openwrt时常出现异常上传且伴随着CPU占用率提升至百分之四五十, 断网现象

Never2333, 2022-05-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：时常出现异常上传且伴随着CPU占用率提升至百分之四五十, 断网现象，而且猫的无线网也受影响。试图重新编译固件，不保留配置重新刷机和重建虚拟机无果。网络是ikuai+openwrt, ikuai负责dhcp, openwrt相当于旁路由, 负责富强。该问题可通过重启openwrt恢复，但是在统计和监控中未发现任何异常。ikuai的统计中显示的是openwrt异常上传，协议为“常用协议”。

[image]
[image]

### (2) 路由器型号和固件版本

A：x86_64. <html>
<body>

OpenWrt R22.5.5 / LuCI Master (git-22.138.16277-772b531) 内核: 5.15.41

</body>
</html>

### (3) 详细日志

A：（发生问题时内核无新日志，现附上全部日志）
[    0.000000] Linux version 5.15.41 (never@never-virtual-machine-2) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4520-178730014) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Fri May 20 05:20:52 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=f271b0be-fe52-91a5-1a62-57ab3315e802 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] signal: max sigframe size: 1040
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffddfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffde000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.15.0-0-g2dd4b9b3f840-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 2483001, primary cpu clock
[    0.000002] kvm-clock: using sched offset of 3815649498 cycles
[    0.000009] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000015] tsc: Detected 2807.998 MHz processor
[    0.000499] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000501] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000504] last_pfn = 0x7ffde max_arch_pfn = 0x400000000
[    0.000533] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.006136] found SMP MP-table at [mem 0x000f5b50-0x000f5b5f]
[    0.006375] ACPI: Early table checksum verification disabled
[    0.006424] ACPI: RSDP 0x00000000000F5980 000014 (v00 BOCHS )
[    0.006429] ACPI: RSDT 0x000000007FFE4A46 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006439] ACPI: FACP 0x000000007FFE476C 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006443] ACPI: DSDT 0x000000007FFDF040 00572C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006446] ACPI: FACS 0x000000007FFDF000 000040
[    0.006449] ACPI: APIC 0x000000007FFE4860 000080 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006451] ACPI: SSDT 0x000000007FFE48E0 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.006454] ACPI: HPET 0x000000007FFE49AA 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006457] ACPI: MCFG 0x000000007FFE49E2 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006459] ACPI: WAET 0x000000007FFE4A1E 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006462] ACPI: Reserving FACP table memory at [mem 0x7ffe476c-0x7ffe485f]
[    0.006463] ACPI: Reserving DSDT table memory at [mem 0x7ffdf040-0x7ffe476b]
[... 812 lines cut ...]
Sun May 22 08:00:00 2022 cron.err crond[10333]: USER root pid 5270 cmd /etc/init.d/frp restart
Sun May 22 08:00:00 2022 user.notice FRPC: Shutting down frp service
Sun May 22 08:00:00 2022 user.notice FRPC: Starting frp service
Sun May 22 08:40:00 2022 cron.err crond[10333]: USER root pid 32198 cmd /etc/init.d/frp restart
Sun May 22 08:40:00 2022 user.notice FRPC: Shutting down frp service
Sun May 22 08:40:00 2022 user.notice FRPC: Starting frp service
Sun May 22 09:00:00 2022 cron.err crond[10333]: USER root pid 14992 cmd /etc/init.d/frp restart
Sun May 22 09:00:00 2022 user.notice FRPC: Shutting down frp service
Sun May 22 09:00:00 2022 user.notice FRPC: Starting frp service
Sun May 22 09:24:15 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 2 (fd 52) from 192.168.233.125 port 5578 is idle for more than 60 seconds - closing it.
Sun May 22 09:24:15 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 4 (fd 49) from 192.168.233.125 port 5546 is idle for more than 60 seconds - closing it.
Sun May 22 09:24:15 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 5 (fd 50) from 192.168.233.125 port 5544 is idle for more than 60 seconds - closing it.
Sun May 22 09:24:15 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.233.125 port 5547 is idle for more than 60 seconds - closing it.
Sun May 22 09:24:15 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 5 (fd 47) from 192.168.233.125 port 5542 is idle for more than 60 seconds - closing it.
Sun May 22 09:30:07 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.233.125 port 10096 is idle for more than 60 seconds - closing it.
Sun May 22 09:30:07 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 4 (fd 48) from 192.168.233.125 port 5543 is idle for more than 60 seconds - closing it.
Sun May 22 09:30:07 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 5 (fd 52) from 192.168.233.125 port 10097 is idle for more than 60 seconds - closing it.
Sun May 22 09:30:07 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 4 (fd 49) from 192.168.233.125 port 10094 is idle for more than 60 seconds - closing it.
Sun May 22 09:30:07 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 6 (fd 50) from 192.168.233.125 port 10095 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 3 (fd 51) from 192.168.233.125 port 11234 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 4 (fd 52) from 192.168.233.125 port 11235 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 4 (fd 49) from 192.168.233.125 port 11232 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 5 (fd 48) from 192.168.233.125 port 11231 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 5 (fd 47) from 192.168.233.125 port 10093 is idle for more than 60 seconds - closing it.
Sun May 22 09:37:27 2022 daemon.info netdata[5876]: POLLFD: LISTENER: client slot 6 (fd 50) from 192.168.233.125 port 11233 is idle for more than 60 seconds - closing it.

3mile, 2022-05-23:
同样的情况，同样的固件下物理机正常。PVE经常出现CPU异常占用，此时ssh已经无法登录，op网页无法打开，只能重启，日志无异常。
由于重启后无日志，所以还搭建了rsyslog服务器接收日志。
以下是异常时rsyslog,`07:31:27`正常，`09:31:27`后就没有日志接收了。`09:34:51`重启
```
May 22 07:31:26 OpenWrt netifd: IPTV (2641): udhcpc: sending renew to server 10.97.64.1
May 22 07:31:27 OpenWrt netifd: IPTV (2641): udhcpc: lease of 10.131.128.146 obtained from 10.97.64.1, lease time 14400
May 22 09:31:27 OpenWrt netifd: IPTV (2641): udhcpc: sending renew to server 10.97.64.1
May 22 09:34:51 OpenWrt kernel: [43429.104315] rcu: INFO: rcu_sched self-detected stall on CPU
May 22 09:34:51 OpenWrt kernel: [43429.105278] rcu: #0110-....: (58822 ticks this GP) idle=351/1/0x4000000000000000 softirq=1562476/1562476 fqs=29816
May 22 09:34:51 OpenWrt kernel: [43429.106559] #011(t=60003 jiffies g=2350289 q=41486)
May 22 09:34:51 OpenWrt kernel: [43429.107468] NMI backtrace for cpu 0
May 22 09:34:51 OpenWrt kernel: [43429.108288] CPU: 0 PID: 7007 Comm: nfsd Not tainted 5.15.32 #0
May 22 09:34:51 OpenWrt kernel: [43429.109284] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.15.0-0-g2dd4b9b3f840-prebuilt.qemu.org 04/01/2014
```

coolsnowwolf, 2026-09-05:
x86/PVE 异常上传没有连接追踪、进程、接口统计或抓包证据，且旁路由拓扑包含其他设备；无法归因 LEDE，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf