coolsnowwolf/lede #9435: luci系统进程界面cpu使用率及内存使用率一直为0，不变化

Rivens7, 2022-05-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
luci系统进程界面cpu使用率及内存使用率一直为0不发生变化，谷歌了好久没有找到问题所在，是编译的时候少勾选依赖了吗
Q：是否搜索了issue (使用 "x" 选择)
* [是] 没有类似的issue

### (2) 路由器型号和固件版本

A：X86 R22.5.5

### (3) 详细日志

A：
[image]

WYC-2020, 2022-06-07:
[image]

ndevel, 2022-06-17:
`root@OpenWrt:/usr/lib/lua/luci/view/admin_status# pwd
/usr/lib/lua/luci/view/admin_status
root@OpenWrt:/usr/lib/lua/luci/view/admin_status# ll
drwxr-xr-x    3 root     root          4096 Jun 17 11:38 ./
drwxr-xr-x   25 root     root          4096 Apr 19 23:00 ../
-rw-r--r--    1 root     root          8285 Apr 19 23:00 bandwidth.htm
-rw-r--r--    1 root     root         11339 Apr 19 23:00 connections.htm
-rw-r--r--    1 root     root           377 Apr 19 23:00 dmesg.htm
drwxr-xr-x    2 root     root          4096 Apr 19 23:00 index/
-rw-r--r--    1 root     root         27124 Apr 19 23:00 index.htm
-rw-r--r--    1 root     root         27124 Jun 17 11:38 index.htm.old
-rw-r--r--    1 root     root          4937 Apr 19 23:00 iptables.htm
-rw-r--r--    1 root     root          8586 Apr 19 23:00 load.htm
-rw-r--r--    1 root     root          5010 Apr 19 23:00 routes.htm
-rw-r--r--    1 root     root           379 Apr 19 23:00 syslog.htm
-rw-r--r--    1 root     root         12201 Apr 19 23:00 wireless.htm

#index.htm文件中有这一行：
#local cpu_usage = (luci.sys.exec("expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')") or "6") .. "%"

root@OpenWrt:~# top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}'
97.7
root@OpenWrt:~# top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}'
100
root@OpenWrt:~# expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')
expr: non-numeric argument
root@OpenWrt:~# expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')
expr: non-numeric argument
root@OpenWrt:~# expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')
expr: non-numeric argument
root@OpenWrt:~# expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')
0

root@OpenWrt:~# expr 100 - 97.7
expr: non-numeric argument
root@OpenWrt:~# 
root@OpenWrt:~# expr 100 - 97
3

#原因是expr数学运算只能对整数，不支持浮点数

change line:
expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')
to:
echo "scale=1; 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')" | bc
result:
local cpu_usage = (luci.sys.exec("echo \"scale=1; 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')\" | bc")) .. "%"
or:
local cpu_usage = (luci.sys.exec("echo \"scale=1; 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')\" | bc") or "6") .. "%"`

ndevel, 2022-06-17:
注意busybox需要加入bc命令的支持，好像是默认选中的

ndevel, 2022-06-17:
`root@dev:/home/dev/dev/lede_20220412/lede/feeds/luci# git diff
diff --git a/modules/luci-base/luasrc/sys.lua b/modules/luci-base/luasrc/sys.lua
index 78f6591..c8367e9 100644
--- a/modules/luci-base/luasrc/sys.lua
+++ b/modules/luci-base/luasrc/sys.lua
@@ -395,7 +395,7 @@ function process.list()
 
        for line in ps do
                local pid, ppid, user, stat, vsz, mem, cpu, cmd = line:match(
-                       "^ *(%d+) +(%d+) +(%S.-%S) +([RSDZTW][W ][<N ]) +(%d+) +(%d+%%) +(%d+%%) +(.+)"
+                       "^ *(%d+) +(%d+) +(%S.-%S) +([RSDZTW][W ][<N ]) +(%S+) +(%d+%%) +(%d+%%) +(.+)"
                )
 
                local idx = tonumber(pid)
diff --git a/modules/luci-base/root/etc/config/luci b/modules/luci-base/root/etc/config/luci
index baa3ac5..4a46642 100644
--- a/modules/luci-base/root/etc/config/luci
+++ b/modules/luci-base/root/etc/config/luci
@@ -1,5 +1,5 @@
 config core main
-       option lang auto
+       option lang en
        option mediaurlbase /luci-static/bootstrap
        option resourcebase /luci-static/resources
        
diff --git a/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm b/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm
index fc29487..cdc4724 100644
--- a/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm
+++ b/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm
@@ -48,7 +48,7 @@
                
                local user_info = luci.sys.exec("cat /proc/net/arp | grep 'br-lan' | grep '0x2' | wc -l")
                
-               local cpu_usage = (luci.sys.exec("expr 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')") or "6") .. "%"
+               local cpu_usage = (luci.sys.exec("echo \"scale=1; 100 - $(top -n 1 | grep 'CPU:' | awk -F '%' '{print$4}' | awk -F ' ' '{print$2}')\" | bc") or "6") .. "%"
 
                local rv = {
                        cpuusage    = cpu_usage,
root@dev:/home/dev/dev/lede_20220412/lede/feeds/luci# `

coolsnowwolf, 2026-09-05:
当前 LuCI 已使用 `luci-rpc` 的进程列表接口并显示 CPU usage 与 Memory usage 列；旧版页面数值不更新的问题不再对应现行实现，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf