coolsnowwolf/lede #9437: 遇到一个诡异的问题，HC5962，有线插到中间的lan口上，过会2.4G就上不了网了

modelsun, 2022-05-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有线插到中间的lan口上（一共3个），过会2.4G就上不了网了，5G没问题，但插其它lan口也没问题

### (2) 路由器型号和固件版本

A：HIWIFI  HC5962

### (3) 详细日志

A：  版本5.10.115

catsimple, 2022-05-25:
提问不贴日志是什么成分

modelsun, 2022-06-02:
> 提问不贴日志是什么成分

大佬，已经增加日志了。
在系统日志中看，有在反复打印如下日志， 连上2.4G 提示，网络不可用，再连5G提示网络不可用，过了一会，都正常了。

Thu Jun 2 08:25:21 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED f2:e7:b3:f9:ed:6d
Thu Jun 2 08:25:21 2022 daemon.info hostapd: wlan0: STA f2:e7:b3:f9:ed:6d IEEE 802.11: disassociated
Thu Jun 2 08:25:21 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:25:21 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:25:21 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED f2:e7:b3:f9:ed:6d 2

Thu Jun 2 08:25:44 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f2:e7:b3:f9:ed:6d
Thu Jun 2 08:25:44 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: disassociated
Thu Jun 2 08:25:44 2022 daemon.info hostapd: wlan0: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:25:44 2022 daemon.info hostapd: wlan0: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:25:44 2022 daemon.info hostapd: wlan0: STA f2:e7:b3:f9:ed:6d IEEE 802.11: associated (aid 

Thu Jun 2 08:26:51 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED f2:e7:b3:f9:ed:6d
Thu Jun 2 08:26:51 2022 daemon.info hostapd: wlan0: STA f2:e7:b3:f9:ed:6d IEEE 802.11: disassociated
Thu Jun 2 08:26:51 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:26:51 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated
Thu Jun 2 08:26:51 2022 daemon.info hostapd: wlan1: STA f2:e7:b3:f9:ed:6d IEEE 802.11: authenticated

modelsun, 2022-06-02:
> 提问不贴日志是什么成分

大佬，已经更新日志了

modelsun, 2022-06-02:
> 
> 贴内核日志

[    0.000000] Linux version 5.10.117 (runner@fv-az252-680) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r0-bcfcad5) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Sun May 29 16:05:18 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is HiWiFi HC5962
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30608 r8192 d22640 u61440
[    0.000000] pcpu-alloc: s30608 r8192 d22640 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00032600
[    0.000000] Readback ErrCtl register=00032600
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249000K/262144K available (6800K kernel code, 625K rwdata, 1412K rodata, 1260K init, 235K bss, 13144K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x3d0/0x5e4 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000013] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.007949] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.016917] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.083094] pid_max: default: 32768 minimum: 301
[    0.087861] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.095063] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.105475] rcu: Hierarchical SRCU implementation.
[    0.110473] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.118491] smp: Bringing up secondary CPUs ...
[    0.123678] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.123689] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.123702] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.123779] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.178384] Synchronize counters for CPU 1: done.
[    0.210711] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.210720] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.210728] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.210781] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.269678] Synchronize counters for CPU 2: done.
[    0.300168] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.300177] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.300185] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.300238] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.354865] Synchronize counters for CPU 3: done.
[    0.384735] smp: Brought up 1 node, 4 CPUs
[    0.392782] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.402602] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[... 445 lines cut ...]
[   98.250239] recv user msg
[   98.252974] oaf msg handle, action = 0
[   98.256737] module init.........
[  111.812610] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  111.820525] NFSD: Using legacy client tracking operations.
[  111.826069] NFSD: starting 10-second grace period (net f00000a0)
[92515.344566] mt76x2e 0000:02:00.0: MCU message 1f (seq 12) timed out
[92515.585123] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[92515.590671] mt76x2e 0000:02:00.0: Build: 1
[92515.595155] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[92515.614535] mt76x2e 0000:02:00.0: Firmware running!
[92515.624963] ieee80211 phy1: Hardware restart was requested
[93519.832194] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[93519.837684] mt76x2e 0000:02:00.0: Build: 1
[... cut ...]
[137534.750572] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[137534.758254] br-lan: port 2(lan2) entered blocking state
[137534.763926] br-lan: port 2(lan2) entered forwarding state
[137544.109842] mt7530 mdio-bus:1f lan2: Link is Down
[137544.116429] br-lan: port 2(lan2) entered disabled state
[137546.190401] mt7530 mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[137546.198589] br-lan: port 2(lan2) entered blocking state
[137546.204279] br-lan: port 2(lan2) entered forwarding state

closed 2022-07-25 by aiamadeus