coolsnowwolf/lede #9441: 老大能否更新mtk7621的hwnat硬件加速

lwb1978, 2022-05-25:
mtk开源了新的hwnat和有线驱动，有网友测试效果明显，老大能否抽空跟进？谢谢。

siyu0, 2022-05-26:
支持，目前7621的hwnat太残废了。

nanchuci, 2022-05-26:
不要着急，才出来，会有的！现目前相对说改动太大，还不能兼顾到各个机型！

chobitsos, 2022-05-26:
> mtk开源了新的hwnat和有线驱动，有网友测试效果明显，老大能否抽空跟进？谢谢。

在guyhub这里吗？

uniessey, 2022-05-26:
> > mtk开源了新的hwnat和有线驱动，有网友测试效果明显，老大能否抽空跟进？谢谢。
> 
> 在guyhub这里吗？ 话说之前不是搞过一次吗？#5123

我猜测他说的可能是这个：https://forum.openwrt.org/t/use-mtk-sdk-hwnat-driver-to-replace-hw-flow-offload/128120

lwb1978, 2022-05-26:
> > > mtk开源了新的hwnat和有线驱动，有网友测试效果明显，老大能否抽空跟进？谢谢。
> > 
> > 
> > 在guyhub这里吗？ 话说之前不是搞过一次吗？#5123
> 
> 我猜测他说的可能是这个：https://forum.openwrt.org/t/use-mtk-sdk-hwnat-driver-to-replace-hw-flow-offload/128120

恩山论坛的链接在这：https://www.right.com.cn/forum/thread-8235037-1-1.html

liwenjie119, 2022-05-26:
期待优化后的新驱动可用

ttmi9002, 2022-05-26:
有了这个驱动，MTK7621又能再战三年。现在用R3G组网就是HWNAT不稳定无线速度只能跑到300M，新驱动有望提升到500M

AmadeusGhost, 2022-05-26:
在做了，别急
[image]

chobitsos, 2022-05-27:
> 在做了，别急
> [image]

大佬，顺便把mt7260的也加上哈

286700234, 2022-05-28:
期待做完了之后的效果怎么样

closed 2022-06-12 by AmadeusGhost