coolsnowwolf/lede #9442: 5.15.42和43这2版一直循环重启，各位大神看看

rongcenter, 2022-05-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

5.15.42和43这2版一直循环重启，各位大神看看

### (2) 路由器型号和固件版本

X86

### (3) 详细日志

[image]

19901019, 2022-05-25:
我刚编译5.15.43，WAN口拨号成功就一直重启，现在一登陆进去立马停用WAN口刷回5.15.40了。

qqsir-dev, 2022-05-26:
我的也是这个问题，软路由。

rongcenter, 2022-05-26:
> 我刚编译5.15.43，WAN口拨号成功就一直重启，现在一登陆进去立马停用WAN口刷回5.15.40了。

41没问题的

mitsukileung, 2022-05-26:
同样的问题

pangfeng161, 2022-05-26:
43版本暂时未发现,或许是编译的问题吧

Eay5, 2022-05-26:
我的也是重启

xiaiohuan, 2022-05-26:
哇原来不止我一个有这问题

xiaiohuan, 2022-05-26:
> > 我刚编译5.15.43，WAN口抛硬币就一直在重启，现在一登陆立马用5.1.40万口刷回5.1.40了。
> 
> 41 没有问题的

你好问下 怎么改编译内核版本

xiaiohuan, 2022-05-26:
> 同样问题，只要设置了pppoe拨号，拨上就重启。我暂时退回41了

怎么改编译内核版本的 我直接改了 文件 信息 就无限报错了

19901019, 2022-05-26:
> > 同样问题，只要设置了pppoe拨号，拨上就重启。我暂时退回41了
> 
> 
> 
> 怎么改编译内核版本的 我直接改了 文件 信息 就无限报错了

X86改成5.4的话
lede-target-linux-x86-makefile里修改KERNEL_PATCHVER:=5.4

suizhenmin, 2022-05-26:
5.15.43 刚编译的，没有问题，也有可能我集成的东西少，就一个ssr

coolsnowwolf, 2022-05-26:
看起来是 flow offload这边的问题

xiaiohuan, 2022-05-26:
> 5.15.43 刚编译的，没有问题，也有可能我集成的东西少，就一个ssr

X86 软路由版本会重启

3mile, 2022-05-26:
同样的问题,回退到5.15.32了

WYC-2020, 2022-05-26:
没问题，
[image]
[image]
自己吧ipv6 选上看下

coolsnowwolf, 2022-05-26:
暂时先 revert ，充分测试再提交

rongcenter, 2022-05-26:
> 暂时先 revert ，充分测试再提交

大神看一下，切到sfe就正常了。
![图像 006](https://user-i
[image]
mages.githubusercontent.com/6003357/170472784-3110b758-556a-4791-8409-2a857e141ec5.jpg)

WYC-2020, 2022-05-26:
切不切都正常

19901019, 2022-05-26:
刚新编的5.15.41终于不会重启了。

jwkinko, 2022-05-26:
> > 暂时先 revert ，充分测试再提交
> 
> 
> 
> 大神看一下，切到sfe就正常了。
> 
> [image]
> 
> [image]
> 
> 
> 
> 

这个插件包上次更新就有问题了。

suizhenmin, 2022-05-26:
> > 5.15.43 刚编译的，没有问题，也有可能我集成的东西少，就一个ssr
> 
> X86 软路由版本会重启

我的就是X86的，应该是我极度精简吧，就加了ssr 。turbo  acc 我没有编译。

rongcenter, 2022-06-01:
各位最新的都好了吗？编译了最新的还是自动无限重启

closed 2022-06-01 by rongcenter

19901019, 2022-06-01:
> 各位最新的都好了吗？编译了最新的还是自动无限重启

我刚编译好，听你这么一说，我不敢上传更新了

rongcenter, 2022-06-01:
> > 各位最新的都好了吗？编译了最新的还是自动无限重启
> 
> 我刚编译好，听你这么一说，我不敢上传更新了
我试试make clean，再试一次

19901019, 2022-06-01:
> > > 各位最新的都好了吗？编译了最新的还是自动无限重启
> 
> > 
> 
> > 我刚编译好，听你这么一说，我不敢上传更新了
> 
> 
> 
> 我试试make clean，再试一次

老哥行不行说一下，我都不敢上传更新了

jwkinko, 2022-06-01:
别选turboacc 就好了，我现在用openwrt官方版本了

rongcenter, 2022-06-01:
> > > > 各位最新的都好了吗？编译了最新的还是自动无限重启
> > 
> > 
> > > 
> > 
> > 
> > > 我刚编译好，听你这么一说，我不敢上传更新了
> > 
> > 
> > 我试试make clean，再试一次
> 
> 老哥行不行说一下，我都不敢上传更新了

别传吧，我的无限重启了，再看看是不是个例，要不你试试？

rongcenter, 2022-06-01:
> 

是的，不选这个一点问题没有

rongcenter, 2022-06-01:
> 

大神是默认的配置，就是勾上了IPv6选项，不选turboacc没有问题

WYC-2020, 2022-06-01:
> > > 一开始就一直没问题86，这次也没改任何东西，只是revert回来而已，十之八九和自己定制的什么有关系，建议自己先默认，不保留配置刷
> > 
> > 
> > 大神就默认配置，就是勾上了ipv6，不选turboacc或者选sfe就没有问题，先不勾turboacc用着了
> 
> 把那个tur补丁回复试下，我的是一直保留这个补丁的，我这边测试是带上sfe也没问题

ipv6记得把防火墙里面那两个模块也选上不是简单只选一个地方就玩了的

rongcenter, 2022-06-02:
已经修复了，快去编