coolsnowwolf/lede #9445: R4S 5.15编译，总是不能正常运行

gswangmd-gssi, 2022-05-27:
R4S 5.15编译，总是不能正常运行，到底和那些插件冲突，也没办法查。

Boos4721, 2022-05-28:
[image]

gswangmd-gssi, 2022-05-28:
我反复编译，都是WEB页面打不开。

1054009064, 2022-05-28:
部分 r4s 减配不减价，以至于你以为你买到了 r4s。

minlang112, 2022-05-28:
你们说的R4S可能根本就不是一个东西

Boos4721, 2022-05-28:
[image]

BenjaminX, 2022-05-28:
我也遇到了 Web 管理打不开问题

gswangmd-gssi, 2022-05-29:
> [image]

下载您的代码编译R4S，提示补丁有问题，编译不下去。

coolsnowwolf, 2022-05-29:
你们手上的 R4S 其实真正的身份是 R4C （阉割 epprom 和用国产网卡芯片的版本）

你们手上现在 +50 所谓的企业版，其实真正的身份是原来的  R4S

而 R4C 我目前无暇修复在 5.15 上面的网卡问题

总之：
R4C = 原来的 R4S 阉割 epprom 和用国产网卡芯片的版本 ， 价格不变
R4S 企业版 = 原来的 R4S  ，价格 +50 

这么说能明白了么 ？

gswangmd-gssi, 2022-05-29:
啊？

1054009064, 2022-05-29:
重点：无暇修复

AmadeusGhost, 2022-05-29:
这玩意建议去找卖你们东西的商家解决。

LPDDR6-10000MHz, 2022-05-30:
拆机，拍照pcie网卡看看

biggerdogs, 2022-06-11:
我也是被忽悠了，买了个家庭版，发现刷不了固件，退了

showme2005, 2022-06-14:
> 你们手上的 R4S 其实真正的身份是 R4C （阉割 epprom 和用国产网卡芯片的版本）
> 
> 你们手上现在 +50 所谓的企业版，其实真正的身份是原来的 R4S
> 
> 而 R4C 我目前无暇修复在 5.15 上面的网卡问题
> 
> 总之： R4C = 原来的 R4S 阉割 epprom 和用国产网卡芯片的版本 ， 价格不变 R4S 企业版 = 原来的 R4S ，价格 +50
> 
> 这么说能明白了么 ？

我C 还有这操作...MD..怪说不得..R4S才出来的时候 很多大佬不推荐..但是问他们也不说..MD 你这一说..结果 原因在这里...我艹...随便在问一下...只有拆机才可以看?还是通过 什么可以查看自己是不是买的阉割的呀......

bigdongdongCLUB, 2022-06-15:
真的 不是有人问我  对R4S标准版 嗤之以鼻并且发来你这发言的截图。我真的不晓得你背地里会这么的张嘴就来。
我本身一个销售商 对于友善 对于野火 对于任何产品 我对事不对人！但你这真的站在自己有点技术，任意的带节奏！
真的让我有点看不下去了！！

先说R4S 标准版 。OK 是阉割了 MAC芯片的 也就是你说的 epprom。 这个确实不假，但 价格不变？！你哪里得出来的结论！！
有任何R4S标准版本也就是所谓的“R4C”和R4S在不分版本之前 价格是一样的证据 我直接个人奖励5000！
公开出视频给你赔礼道歉！

R4S在 原版的基础上去掉这个epprom芯片 价格是降了50好吗！！！我是个卖家 价格我太了解了！
然后你还加了一句改用了国产芯片！OK 这个你就敢这么信誓旦旦的张嘴就来？？你是个开发者 芯片你不认识吗？
你就欺负小白 他们不懂是吧！ R4S企业版 和你所说的“R4C”也就是 R4S标准 网卡上有任何区别

coolsnowwolf, 2022-06-15:
1. 首先这个问题的时间我回复是 5月29号，那时候麻烦楼上自己回答下，所谓的 R4S 非企业版是什么价格，是不是后来才降的 50？
2. R4S 的 5.15 内核代码是我们比 OP 官方早得多的时间适配的，我们有和群友测试过他们手上的 R4S 网口是能正常使用的。直到今天 （6月15日）的 commit，使用 test kernel 编译出来的 R4S 固件网络接口依然一切正常，我也找了骷髅头（R4S固件发布者之一) 确认是网口可用的。
3. 那么请问如果不是网络硬件变换了，应该做如何的解释呢？
4. 你问我们的居心，那你的居心何在呢？就事论事不好么？ 阴阳怪气谁不会？？

[image]

[image]

ForgotFun, 2022-06-15:
大家别吵了，参加佐大OpenWrt开发培训班，咱们就是一家人了，都是OpenWrt开发者了。
都是开发者后，这些就不是吵架了，而是技术交流了，所以得先参加佐大OpenWrt培训班。
https://forgotfun.org/2018/04/openwrt-training-2018.html

bigdongdongCLUB, 2022-06-15:
> 
1： 价格我最明白了。5月29日的时候 R4S标准版（非企业版）单机 价格 479 在6月6日 开始618活动 降价30 18日之后会恢复479

dahetaoa, 2022-06-15:
要我说，x86挺好的……

asushugo, 2022-06-15:
[image]

EnnawYang, 2022-06-15:
容我说一句：屑 rockchip

coolsnowwolf, 2022-06-15:
> 全社交媒体

哇 全社交媒体啊，我真的好怕，球球你不要发好不好？？ 

R4S 天下无敌~~

我可以理解成要挟和恐吓么？ ？

dahetaoa, 2022-06-15:
海鲜佬m2 yyds（）

xiao9jun, 2022-06-15:
[image]
到达github最高层辩论赛，太美丽啦，还是报佐大培训班吧家人们

JiangGeek, 2022-06-15:
真不明白这些纯ARM软路由定位在哪，一堆博主推，一堆人买，性能真干不过x86，还不如买高通，NSS yyds

susx-net, 2022-06-15:
这辣鸡arm就不该做软路由，没有硬件转发就是耍流氓。

susx-net, 2022-06-15:
直接删库可解决此问题。

hxlls, 2022-06-15:
删掉R4S支持和关掉评论把，UP主是有本事的人，能号令全社交媒体，应该能自己提供技术支持

dahetaoa, 2022-06-15:
> 删掉R4S支持和关掉评论把，UP主是有本事的人，能号令全社交媒体，应该能自己提供技术支持

河南拔智齿

smeor, 2022-06-15:
**号外号外，某代商公然威胁致命开发者，以谋求关注流量，号外号外**

coolsnowwolf, 2022-06-15:
1. 直到今天为止的 commit，同样代码出的固件，在我借测已还给群友的，和骷髅头用的老版本 R4S 网口一切正常，我为什么不能觉得是硬件换了呢？这不是很自然的吗？

2. 能号令全社交媒体的人，除了我们国家我真正尊敬的党和领导，我还不知道谁敢说这话？ 好大的威风啊！~~ 我真的怕了

BenjaminX, 2022-06-15:
这么热闹？R5s驱动拿出来走一下友善

howie-wu, 2022-06-15:
> 删掉R4S支持和关掉评论把，UP主是有本事的人，能号令全社交媒体，应该能自己提供技术支持

河南拔智齿

bigdongdongCLUB, 2022-06-15:
> 1. 直到今天为止的 commit，同样代码出的固件，在我借测已还给群友的，和骷髅头用的老版本 R4S 网口一切正常，我为什么不能觉得是硬件换了呢？这不是很自然的吗？
> 2. 能号令全社交媒体的人，除了我们国家我真正尊敬的党和领导，我还不知道谁敢说这话？ 好大的威风啊！~~ 我真的怕了

我号令？我是说我发我的全社交媒体！

Evarone, 2022-06-15:
快进到友善系列删库跑路

littlegunboom, 2022-06-15:
要证明是不是换硬件了（最早期的R4S 和 现在卖的 R4S标准版），最简单不应该是拿出拆机什么的直接到证据“打脸雕姓网友”+自证清白吗？还是说碧哥东东你觉得是code问题导致新R4S除了问题？要不你提个PR维护下？一起做些贡献？

xray-bit, 2022-06-15:
说到删库，怂了怂了

bigdongdongCLUB, 2022-06-15:
> 说到删库就怂了？别吧

啊？？兄弟快删吧！当是谷歌制裁华为呢啊 ~~我的天

hxlls, 2022-06-15:
别怂，干就完事了

susx-net, 2022-06-15:
> > 
> 
> 我真没想和你杠。我能在你的地界跟你说这个，我只想你解释一下你所说的。我作为一个经销商。没必要站任何队。 我完全是平心而论！你站队 电犀牛 一点毛病没有！你会技术 靠技术赚钱 没任何问题！但请别有意无意的污蔑！ 这个圈子能和谐一些最好和谐一些！ 你的原话 你自己品品
> 
> 你们手上的 R4S 其实真正的身份是 R4C （阉割 epprom 和用国产网卡芯片的版本）
> 
> 你们手上现在 +50 所谓的企业版，其实真正的身份是原来的 R4S
> 
> 而 R4C 我目前无暇修复在 5.15 上面的网卡问题
> 
> 总之： R4C = 原来的 R4S 阉割 epprom 和用国产网卡芯片的版本 ， 价格不变 R4S 企业版 = 原来的 R4S ，价格 +50

小伙子，不用说了，争来争去没啥意义，我们直接删除所有友善系列支持，你自寻出路吧，你觉得呢？

bigdongdongCLUB, 2022-06-15:
就事论事！自己发言是带节奏的 自己还不了解？

littlegunboom, 2022-06-15:
> 就事论事！自己发言是带节奏的 自己还不了解？

没见到就事论事，只看见你在浑水摸鱼，哗众取宠

smeor, 2022-06-15:
### **搞起来**

coolsnowwolf, 2022-06-15:
退一万步，我们觉得友善这型号硬件不同导致的，那也是友善出来和我们说明白。

你一个经销商你能代表厂家吗？ 你着什么急？

不要干不是自己身份的事情

xiao9jun, 2022-06-15:
你好像很在意说友善被说成割韭菜厂商这件事，说到底还是硬件方案的问题，这事得厂家解释吧

Tridagger, 2022-06-15:
的确，面对别人的质疑，拿出证据自证一下！

susx-net, 2022-06-15:
[image]

xiaozhou1210, 2022-06-15:
别骂了别骂了，再骂我的r4s怎么卖

minlang112, 2022-06-15:
建议经销商同志拿出不同批次板子的高清无码截图，以证网卡PHY未做任何改动，开发者和本库的协力维护者都是讲道理的人，以德服人

susx-net, 2022-06-15:
[image]

Evarone, 2022-06-15:
大伙快买R4s 我教你们在恩山论坛下载固件

Kaze1027, 2022-06-15:
纯路人，这bigdongdongCLUB是谁啊，请问是RK官方吗。

MarksonHon, 2022-06-15:
> 纯路人，这bigdongdongCLUB是谁啊，请问是RK官方吗。

一个 UP 主啦

FUjr, 2022-06-15:
他不是测评r5s那位嘛？怎么经销商能测评自家产品了？还公正客观？嘴里各种阴阳友商？

bigdongdongCLUB, 2022-06-15:
> > > 
> > 
> > 
> > 1： 价格我最明白了。5月29日的时候 R4S标准版（非企业版）单机 价格 479 在6月6日 开始618活动 降价30 18日之后会恢复479 而R4S 企业版 单机 价格一直都是在529。 而你对价格是怎么描述的呢？ R4C 标准版 简配 用R4S企业版的价格。而原本的R4S 摇身一变 上涨50 ！ 友善这韭菜割的漂亮 是这个意思吧！ 2：你的意思是简明一下 就是 R4S 之前网卡是能用的 后来不能用了 怀疑硬件换了对吧 3：同上，因为固件问题 怀疑硬件换了是么？你作为一个开发者硬件换没换 不会不知道吧？所以你完全没确定事情 就说 网卡被替换成国产的了是呗？ 我就请问 网卡是不是换了 芯片是不是换 R4S 非企业版和企业版 网卡是不是不一样！？ 4：我居心何在？我需要什么居心，你这番话明眼人都知道你的用意是什么啊！我可以全社交媒体发一下 让大家评论
> 
> 快发，不发是小狗！

6月18日 直播来聊吧！

Kaze1027, 2022-06-15:
> > 纯路人，这bigdongdongCLUB是谁啊，请问是RK官方吗。
> 
> 一个 UP 主啦

哦哦，我看他这么着急，我还以为是他画的R4S的板子呢，嘻嘻🤭

Yixie2020, 2022-06-15:
> 他不是测评r5s那位嘛？怎么经销商能测评自家产品了？还公正客观？嘴里各种阴阳友商？

不要说出来，说出来影响人家恰烂钱了

bigdongdongCLUB, 2022-06-15:
> [image] 请看。。。。

so???

FUjr, 2022-06-15:
这位哥，咋吵架都要给自己引流啊…

bigdongdongCLUB, 2022-06-15:
> > > > > 
> > > > 
> > > > 
> > > > 1： 价格我最明白了。5月29日的时候 R4S标准版（非企业版）单机 价格 479 在6月6日 开始618活动 降价30 18日之后会恢复479 而R4S 企业版 单机 价格一直都是在529。 而你对价格是怎么描述的呢？ R4C 标准版 简配 用R4S企业版的价格。而原本的R4S 摇身一变 上涨50 ！ 友善这韭菜割的漂亮 是这个意思吧！ 2：你的意思是简明一下 就是 R4S 之前网卡是能用的 后来不能用了 怀疑硬件换了对吧 3：同上，因为固件问题 怀疑硬件换了是么？你作为一个开发者硬件换没换 不会不知道吧？所以你完全没确定事情 就说 网卡被替换成国产的了是呗？ 我就请问 网卡是不是换了 芯片是不是换 R4S 非企业版和企业版 网卡是不是不一样！？ 4：我居心何在？我需要什么居心，你这番话明眼人都知道你的用意是什么啊！我可以全社交媒体发一下 让大家评论
> > > 
> > > 
> > > 快发，不发是小狗！
> > 
> > 
> > 6月18日 直播来聊吧！
> 
> 倒也不必直接引流，直接发图吧

> 这位哥，咋吵架都要给自己引流啊…

大哥了 这里能有几个人看见 我来这引流  我饿疯了吧

[128 more replies]

closed 2022-06-17 by coolsnowwolf