coolsnowwolf/lede #9447: R4S CGI Web 管理访问异常

BenjaminX, 2022-05-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：间歇性 luci CGI 失败

### (2) 路由器型号和固件版本

A：
R4S 4G
OpenWrt R22.5.5 / LuCI Master (git-22.146.34683-df8288f)
内核 5.15.41

### (3) 详细日志

Sat May 28 16:44:42 2022 daemon.err uwsgi[1837]: Sat May 28 16:44:42 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Sat May 28 16:44:47 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 2 (new pid: 1202)
Sat May 28 16:45:38 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:45:43 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:45:43 2022 daemon.err uwsgi[1837]: Sat May 28 16:45:43 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:45:47 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:45:47 2022 daemon.err uwsgi[1837]: Sat May 28 16:45:47 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:45:51 2022 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Sat May 28 16:46:38 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:46:56 2022 daemon.info uwsgi-luci: uWSGI worker 3 cheaped.
Sat May 28 16:46:59 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 1 (new pid: 3064)
Sat May 28 16:46:59 2022 daemon.err uwsgi[1837]: Sat May 28 16:46:59 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Sat May 28 16:47:56 2022 daemon.err nginx[2168]: 2022/05/28 16:47:56 [error] 2273#0: *113 upstream timed out (110: Operation timed out) while reading upstream, client: 10.1.1.70, server: _lan, request: "GET /cgi-bin/luci HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "10.1.1.1", referrer: "https://10.1.1.1/"
Sat May 28 16:47:56 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:49:00 2022 daemon.info uwsgi-luci: Sat May 28 16:49:00 2022 - worker 2 (pid: 1202) is taking too much time to die...NO MERCY !!!
Sat May 28 16:49:01 2022 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Sat May 28 16:50:12 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 2 (new pid: 5499)
Sat May 28 16:51:09 2022 daemon.err nginx[2168]: 2022/05/28 16:51:09 [error] 2273#0: *116 upstream timed out (110: Operation timed out) while reading upstream, client: 10.1.1.70, server: _lan, request: "GET /cgi-bin/luci HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "10.1.1.1", referrer: "https://10.1.1.1/"
Sat May 28 16:51:09 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:51:09 2022 daemon.err uwsgi[1837]: Sat May 28 16:51:09 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:52:13 2022 daemon.info uwsgi-luci: Sat May 28 16:52:13 2022 - worker 1 (pid: 3064) is taking too much time to die...NO MERCY !!!
Sat May 28 16:52:14 2022 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Sat May 28 16:53:02 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 1 (new pid: 7602)
Sat May 28 16:53:02 2022 daemon.err uwsgi[1837]: Sat May 28 16:53:02 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Sat May 28 16:53:54 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 3 (new pid: 8228)
Sat May 28 16:53:59 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:53:59 2022 daemon.err uwsgi[1837]: Sat May 28 16:53:59 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:54:19 2022 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Sat May 28 16:54:22 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 2 (new pid: 8549)
Sat May 28 16:54:50 2022 user.warn ddns-scripts[26284]: cloudflare: Updating IP at DDNS provider failed - starting retry 1/0
Sat May 28 16:54:51 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:54:51 2022 daemon.err uwsgi[1837]: Sat May 28 16:54:51 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:55:19 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:55:19 2022 daemon.err uwsgi[1837]: Sat May 28 16:55:19 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:55:50 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:55:50 2022 daemon.err uwsgi[1837]: Sat May 28 16:55:50 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:55:51 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:56:45 2022 daemon.err nginx[2168]: 2022/05/28 16:56:45 [error] 2273#0: *133 upstream timed out (110: Operation timed out) while reading upstream, client: 10.1.1.70, server: _lan, request: "GET /cgi-bin/luci HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "10.1.1.1", referrer: "https://10.1.1.1/"
Sat May 28 16:56:45 2022 daemon.info uwsgi-luci: CGI timeout !!!
Sat May 28 16:56:55 2022 daemon.info uwsgi-luci: Sat May 28 16:56:55 2022 - worker 1 (pid: 7602) is taking too much time to die...NO MERCY !!!
Sat May 28 16:56:56 2022 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Sat May 28 16:57:49 2022 daemon.info uwsgi-luci: Sat May 28 16:57:49 2022 - worker 3 (pid: 8228) is taking too much time to die...NO MERCY !!!
Sat May 28 16:57:50 2022 daemon.info uwsgi-luci: uWSGI worker 3 cheaped.
Sat May 28 16:58:47 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 1 (new pid: 14039)
Sat May 28 16:58:47 2022 daemon.err uwsgi[1837]: Sat May 28 16:58:47 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Sat May 28 16:58:51 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 3 (new pid: 14049)
Sat May 28 16:58:52 2022 daemon.info uwsgi-luci: Sat May 28 16:58:52 2022 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci (ip 10.1.1.70) !!!
Sat May 28 16:58:52 2022 daemon.info uwsgi-luci: Sat May 28 16:58:52 2022 - uwsgi_response_write_body_do(): Broken pipe [core/writer.c line 429] during GET /cgi-bin/luci (10.1.1.70)
Sat May 28 16:58:52 2022 daemon.err uwsgi[1837]: Sat May 28 16:58:52 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:58:55 2022 daemon.info uwsgi-luci: Sat May 28 16:58:55 2022 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci (ip 10.1.1.70) !!!
Sat May 28 16:58:55 2022 daemon.info uwsgi-luci: Sat May 28 16:58:55 2022 - uwsgi_response_write_body_do(): Broken pipe [core/writer.c line 429] during GET /cgi-bin/luci (10.1.1.70)
Sat May 28 16:58:55 2022 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Sat May 28 16:58:55 2022 daemon.err uwsgi[1837]: Sat May 28 16:58:55 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Sat May 28 16:59:40 2022 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.

gswangmd-gssi, 2022-06-05:
反映过多次均无回应。

closed 2022-06-15 by BenjaminX

KennnyCheung, 2023-11-02:
解决了吗