coolsnowwolf/lede #9448: GFW列表模式下，如何让二级主域名走ssr plus，三级域名不走

terrellgf, 2022-05-28:
## 2. 详细叙述

### (1) 具体问题

A：比如我想让https://test.com走代理，https://abc.test.com不走，在GFW列表模式下应该怎么配置？
如果强制走代理的域名中，直接配置test.com，则实际测试test.com和abc.test.com都走代理。
如果强制走代理的域名中，配置test.com。然后不走代理的域名中，配置abc.test.com。则这两个实际测试都不会走代理。

### (2) 路由器型号和固件版本

A：软路由; OpenWrt R20.2.15 / LuCI Master (git-20.047.54228-e6b2608)

chobitsos, 2022-05-30:
隔壁提问，但是我要告诉你，目前无法GFW模式无法实现你想要的效果，你自己去看一下规则源码，你只能用大陆白名单模式可以实现

terrellgf, 2022-05-31:
> 隔壁提问，但是我要告诉你，目前无法GFW模式无法实现你想要的效果，你自己去看一下规则源码，你只能用大陆白名单模式可以实现

谢谢您，IP模式确实是可以的，但是由于我有PT服务，用了IP模式我的桥头堡流量巨大，流量成本上承受不住。我去隔壁问问，感谢！

fengqi, 2022-06-07:
> > 隔壁提问，但是我要告诉你，目前无法GFW模式无法实现你想要的效果，你自己去看一下规则源码，你只能用大陆白名单模式可以实现
> 
> 谢谢您，IP模式确实是可以的，但是由于我有PT服务，用了IP模式我的桥头堡流量巨大，流量成本上承受不住。我去隔壁问问，感谢！

部分PT的trakcer和主站是一个域名，我个人的做法是PT是单独的机器，然后完全不过ssr。

coolsnowwolf, 2026-09-05:
SSR Plus 的域名层级匹配属于第三方规则语义，报告基于 R20.2.15，缺少当前规则生成结果和日志，无法在 LEDE 核心中定位，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf