coolsnowwolf/lede #9449: 云编译后在web升级固件失败：413 Request Entity Too Large nginx/1.21.3

lisankai93, 2022-05-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译后在web升级固件【openwrt-x86-64-generic-squashfs-combined-efi】失败提示：413 Request Entity Too Large nginx/1.21.3

### (2) 路由器型号和固件版本

A：j1900软路由，最新源码，相比上次增加了usb无线路由器的驱动，但是就算用上次的固件也是失败

### (3) 详细日志

A：

lisankai93, 2022-05-29:
如果跨版本内核必须得重新刷固件，请问如何快速恢复：
1. 如何保留目前的不同终端ip设置（涉及到不少智能家居得用静态ip，如果乱了得重新设置很麻烦）
2. 如何保留目前的端口转发（同上）

谢谢！

deplives, 2022-05-30:
> 如果跨版本内核必须得重新刷固件，请问如何快速恢复：
> 
> 1. 如何保留目前的不同终端ip设置（涉及到不少智能家居得用静态ip，如果乱了得重新设置很麻烦）
> 2. 如何保留目前的端口转发（同上）
> 
> 谢谢！

DHCP 静态地址分配的配置保存在 `/etc/config/dhcp` 中，可以保存然后新系统直接覆盖
端口转发的配置在 `/etc/config/firewall ` 中 `config redirect` 的内容，方法同上。
吐槽你一句，你问的这两个问题，随便在google关键字 `openwrt dhcp 配置文件` `openwrt 端口转发 配置文件`搜一下都能出结果，请善用搜索引擎。
[image]
[image]

lisankai93, 2022-05-30:
> DHCP 静态地址分配的配置保存在 `/etc/config/dhcp` 中，可以保存然后新系统直接覆盖 端口转发的配置在 `/etc/config/firewall ` 中 `config redirect` 的内容，方法同上。 吐槽你一句，你问的这两个问题，随便在google关键字 `openwrt dhcp 配置文件` `openwrt 端口转发 [...]

非常感谢！

现在我学会了这些搜索方法，因为以前不懂这个在dhcp里面，还是感谢感谢！
我今晚就重新刷固件，谢谢！

closed 2022-05-30 by lisankai93