coolsnowwolf/lede #9450: mt7621 全新 hwnat 驱动测试帖

AmadeusGhost, 2022-05-29:


liwenjie119, 2022-05-30:
cr660x，正常刷入，作为ap，有线可用，无线网无法获取dns，ip直连可用
<details>
<summary>kernel日志</summary>
<pre><code>  
&nbsp;
[    0.000000] Linux version 5.4.195 (liwenjie@localhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-04f39ae)) #0 SMP Sun May 29 11:39:21 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router CR660x
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27056 r8192 d22096 u57344
[    0.000000] pcpu-alloc: s27056 r8192 d22096 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00056b00
[    0.000000] Readback ErrCtl register=00056b00
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249892K/262144K available (6331K kernel code, 269K rwdata, 1320K rodata, 1280K init, 237K bss, 12252K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007805] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073978] pid_max: default: 32768 minimum: 301
[    0.078775] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085975] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096550] rcu: Hierarchical SRCU implementation.
[    0.101895] smp: Bringing up secondary CPUs ...
[    3.284789] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.284800] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    3.284812] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    3.284920] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.166692] Synchronize counters for CPU 1: done.
[    3.378128] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.378136] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    3.378144] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    3.378200] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.257450] Synchronize counters for CPU 2: done.
[    3.465483] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.465491] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    3.465499] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    3.465566] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.342638] Synchronize counters for CPU 3: done.
[    0.372507] smp: Brought up 1 node, 4 CPUs
[    0.381421] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.391218] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[... 306 lines cut ...]
[   46.629838] device wlan0 entered promiscuous mode
[   46.635414] br-lan: port 2(lan2) entered blocking state
[   46.640716] br-lan: port 2(lan2) entered forwarding state
[   46.667099] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   46.673943] br-lan: port 4(wlan0) entered blocking state
[   46.679387] br-lan: port 4(wlan0) entered forwarding state
[   46.687400] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[   46.696121] br-lan: port 3(lan3) entered blocking state
[   46.701469] br-lan: port 3(lan3) entered forwarding state
[   46.930636] extif_set_dev(wlan1)
[   46.937889] br-lan: port 5(wlan1) entered blocking state
[   46.943256] br-lan: port 5(wlan1) entered disabled state
[   46.950888] device wlan1 entered promiscuous mode
[   46.956186] br-lan: port 5(wlan1) entered blocking state
[   46.961522] br-lan: port 5(wlan1) entered forwarding state
[   47.547233] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   47.584908] mediatek_soc_hnat 1e100000.hnat: hnat_ipv6_get_nexthop:No neigh (daddr=ff02:0000:0000:0000:0000:0000:0000:0016)
[   62.257048] appfilter version:5.0.1
[   62.267306] init oaf sysctl...ok
[   62.997098] recv user msg
[   62.999741] oaf msg handle, action = 0
[   63.003509] module init.........
&nbsp;
</code></pre>
</details>

Hiyajomaho-num9, 2022-05-30:
MSG1500同上问题，有线可用，无线网无法获取dns，ip直连可用，可以互相ping同网段客户端
<details>
<summary>Syslog</summary>

```
Mon May 30 14:22:53 2022 kern.notice kernel: [    0.000000] Linux version 5.4.195 (th1996@cvpr-002) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4549-04f39aef2)) #0 SMP Sun May 29 11:39:21 2022
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] MIPS: machine is RAISECOM MSG1500 X.00
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon May 30 14:22:53 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon May 30 14:22:53 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon May 30 14:22:53 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Zone ranges:
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000]   HighMem  empty
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Early memory node ranges
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27024 r8192 d22128 u57344
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27024 r8192 d22128 u57344 alloc=14*4096
Mon May 30 14:22:53 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Mon May 30 14:22:53 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00005831
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00005831
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] Memory: 250088K/262144K available (6168K kernel code, 274K rwdata, 1300K rodata, 1268K init, 235K bss, 12056K reserved, 0K cma-reserved, 0K highmem)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon May 30 14:22:53 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Mon May 30 14:22:53 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Mon May 30 14:22:53 2022 kern.info kernel: [    0.007798] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.073972] pid_max: default: 32768 minimum: 301
Mon May 30 14:22:53 2022 kern.info kernel: [    0.078764] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.085964] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon May 30 14:22:53 2022 kern.info kernel: [    0.096463] rcu: Hierarchical SRCU implementation.
Mon May 30 14:22:53 2022 kern.info kernel: [    0.101795] smp: Bringing up secondary CPUs ...
Mon May 30 14:22:53 2022 kern.warn kernel: [    9.623169] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... cut ...]
[   40.926611] br-lan: port 2(wlan1) entered forwarding state
[   97.469828] extif_set_dev(wlan0)
[   97.479058] br-lan: port 3(wlan0) entered blocking state
[   97.489838] br-lan: port 3(wlan0) entered disabled state
[   97.501388] device wlan0 entered promiscuous mode
[  104.260839] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  104.273866] br-lan: port 3(wlan0) entered blocking state
[  104.284567] br-lan: port 3(wlan0) entered forwarding state
[ 1497.437914] mt753x 1e110000.gsw: Port 2 Link is Up - 1Gbps/Full
```

</details>

AmadeusGhost, 2022-05-30:
@liwenjie119 @Hiyajomaho-num9 现在应该修好了（

286700234, 2022-05-30:
这个是要重新拉源码还是直接更新大雕的源码就可以?

NagaseKouichi, 2022-05-30:
小娱路由器可以测试吗，没有无线是不是测试有线NAT效率就好了

triatomic, 2022-05-30:
mt7622 ? AX3200 支持的?

AmadeusGhost, 2022-05-31:
> 小娱路由器可以测试吗

可以

crg1433, 2022-05-31:
mt7621 redmi ac2100 支持吗?

AmadeusGhost, 2022-05-31:
我不知道你们是在捣乱还是在干啥，我只想我特地改回 switch 驱动的那些机子都能有人测一遍没问题我好合并。mt7620 我不知道还有什么搞的价值，mt7622（如 ax6s 之类）太贵了我也买不起，再说标题也只写了 mt7621，本菜鸡实在是无法理解 , , , 之前 5441 那个 issues 一堆人急着要新驱动好，我们搞了，怎么现在搞好就没人测了呢？

chobitsos, 2022-05-31:
> 我真的是忍不了，请一些人好好看一下issue内容
> 
> > 这是你们不能做的：
> > 不建议水帖，此帖为测试与反馈帖
> > 不要催更，有时候不是不回，是太忙了没空
> > 无脑催更想上新内核（如 5.10），除非你给我打钱改变做事优先级
> > 禁止任何吵架或者骂人行为，看到了就直接 ban，为什么 issues 现在少有人管懂得都懂
> 
> 这里是给人上传错误Log、找出问题、提出7621机子设配要求用的，是issues,不是discussion！ 才开放两天，又有人当这里是discussion帖子来水！ 请正常提交问题的人原谅我无礼的行为，鄙人是真的无法忍受这种无趣的现象！
> 
> > 看代码好像没有mt7620什么事啊，其实我想搞一下7620的驱动，目前7620有个致命的断流bug😪
> 
> 这里7621驱动测试帖子，不是给你拿来水7620用的！

我只是提一句而已，再说楼主回复了我一定会删comment啊，我搞不懂这有什么好喷的，而且你也完全没有按照楼主的提议做出任何有价值的回复，纯粹在宣泄个人情绪，到此为止吧！

```
这是你们能做的：

我想测试某个机子但源码还没改上，提出需求
测试了某个机子 ok 没问题，留个言
暂时没想到
```

手上有几台K2P，昨天看你更新支持了K2P,第一时间编译出来测试了目前没有什么问题
但是两台K2P其中一台在重启的时候不知道为什么偶尔是获取不到IP地址，大约重启10次里有2两次是无法获取IP地址的情况，后来我在dts文件里面https://github.com/AmadeusGhost/lede/blob/3f9bed0669cf0236930b6f4bc257e08dbd1ba8e2/target/linux/ramips/dts/mt7621_phicomm_k2p.dts#L53
将spi-max-frequency = <80000000>;改成了50000000（参照其他的SDK改的）之后反复测试了重启又没在复现了，我觉得可能是闪存兼容的问题，跟新的hwnat大概率没什么关系
目前的话我手上还有几台小米R4A千兆版，也是MT7621的，但是还没看到支持，如果有时间的话你看看可以加上支持吗？
因为我按照恩山那个帖子自己改了DTS，编译出来之后启动不了，自己又折腾了一晚上搞不定，所以你有时间的话可以加上看看

AmadeusGhost, 2022-05-31:
> 目前的话我手上还有几台小米R4A千兆版，也是MT7621的，但是还没看到支持，如果有时间的话你看看可以加上支持吗？

这种默认用 dsa hwnat 应该也没啥问题。如果是原本就启不动的话（使用新驱动之前），大概率是 breed 的分区与设备 dts 对不上所致。

lovehackintosh, 2022-06-01:
> redmi2100，可获取连接获取ip，蓝牙后手机提示无法到互联网，城市没有问题设置连接ip地址后连接wifi，可获取连接，手机ftp，电脑无法ftp到手机。

你需要提供日志，才能让大佬解决问题

uniessey, 2022-06-02:
> breed

> > 这里7621驱动测试帖子，不是给你拿来水7620用的！
> 
> 我只是提一句而已，再说楼主回复了我一定会删comment啊，我搞不懂这有什么好喷的，而且你也完全没有按照楼主的提议做出任何有价值的回复，纯粹在宣泄个人情绪，到此为止吧！
> 
> 手上有几台K2P，昨天看你更新支持了K2P,第一时间编译出来测试了目前没有什么问题 但是两台K2P其中一台在重启的时候不知道为什么偶尔是获取不到IP地址，大约重启10次里有2两次是无法获取IP地址的情况，后来我在dts文件里面https://github.com/AmadeusGhost/lede/blob/3f9bed0669cf0236930b6f4bc257e08dbd1ba8e2/target/linux/ramips/dts/mt7621_phicomm_k2p.dts#L53 将spi-max-frequency = <80000000>;改成了50000000（参照其他的SDK改的）之后反复测试了重启又没在复现了，我觉得可能是闪存兼容的问题，跟新的hwnat大概率没什么关系 目前的话我手上还有几台小米R4A千兆版，也是MT7621的，但是还没看到支持，如果有时间的话你看看可以加上支持吗？ 因为我按照恩山那个帖子自己改了DTS，编译出来之后启动不了，自己又折腾了一晚上搞不定，所以你有时间的话可以加上看看

参考之前的经验，小米的产品全部需要清除breed配置，要不然启动不了。

lovehackintosh, 2022-06-03:
编译linux内核时报错,日志如下
<details>
<summary>make V=s日志</summary>
<pre><code>  
&nbsp;
 Building modules, stage 2.
  MODPOST 103 modules
  AS      arch/mips/boot/compressed/head.o
  CC      arch/mips/boot/compressed/decompress.o
  CC      arch/mips/boot/compressed/string.o
  SHIPPED arch/mips/boot/compressed/bswapsi.c
  SHIPPED arch/mips/boot/compressed/ashldi3.c
  CC      arch/mips/boot/compressed/dummy.o
  OBJCOPY arch/mips/boot/compressed/vmlinux.bin
  HOSTCC  arch/mips/boot/compressed/calc_vmlinuz_load_addr
  CC      arch/mips/boot/compressed/bswapsi.o
  CC      arch/mips/boot/compressed/ashldi3.o
ERROR: "nd_tbl" [drivers/net/ethernet/mtk/mtk_hnat/mtkhnat.ko] undefined!make[6]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[5]: *** [Makefile:1324: modules] Error 2
make[5]: *** Waiting for unfinished jobs....
  XZKERN  arch/mips/boot/compressed/vmlinux.bin.z
  OBJCOPY arch/mips/boot/compressed/piggy.o
  LD      vmlinuz
  STRIP   vmlinuz
make[5]: Leaving directory '/home/clover/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.195'
make[4]: *** [Makefile:25: /home/clover/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.195/.modules] Error 2
make[4]: Leaving directory '/home/clover/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/clover/lede/target/linux'
time: target/linux/compile#651.13#51.14#72.29
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/clover/lede'
make[1]: *** [target/Makefile:18: /home/clover/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/clover/lede'
make: *** [/home/clover/lede/include/toplevel.mk:230: world] Error 2
&nbsp;
</code></pre>
</details>

AmadeusGhost, 2022-06-04:
> xiaomi redmi ac2100编译linux内核时报错,日志如下
> ERROR: "nd_tbl" [drivers/net/ethernet/mtk/mtk_hnat/mtkhnat.ko]

代码里面并没有这个 nd_tbl，所以我也不知道发生了什么 :(

lovehackintosh, 2022-06-04:
> > xiaomi redmi ac2100编译linux内核时报错,日志如下
> > ERROR: "nd_tbl" [drivers/net/ethernet/mtk/mtk_hnat/mtkhnat.ko]
> 
> 代码里面并没有这个 nd_tbl，所以我也不知道发生了什么 :(
我去build_dir相应的目录中,根本就没有mtkhnat.ko ,系统是archlinuc

lovehackintosh, 2022-06-07:
经测试,使用这套源代码时不能将Global build settings --> Enable IPv6 support in packages设为n,否则会导致编译Linux内核时ERROR: "nd_tbl" [drivers/net/ethernet/mtk/mtk_hnat/mtkhnat.ko]

lovehackintosh, 2022-06-07:
xiaomi redmi ac2100 全新hwnat驱动编译成功，但是WiFi不能ping外网,什么都访问不了(包括web后台管理)，可以ping通内网ip

AmadeusGhost, 2022-06-07:
> WiFi不能ping外网,什么都访问不了(包括web后台管理)，可以ping通内网ip

应该修好了，记得不保留配置升级

feng310, 2022-06-07:
`https://github.com/coolsnowwolf/lede/issues/9450#issue-1251768757`

feng310, 2022-06-07:
> https://github.com/coolsnowwolf/lede/issues/9450#issue-1251768757
1111
111

lovehackintosh, 2022-06-08:
> > xiaomi redmi ac2100 全新hwnat驱动编译成功，但是WiFi不能ping外网,什么都访问不了(包括web后台管理)，可以ping通内网ip
> 
> 请问是使用的大佬的源码吗？ 我使用lede的源码 将ramips文件夹替换过去也是出现了这个问题

是的

vsvkhs, 2022-06-18:
大佬辛苦了，膜拜

Stkrx, 2022-06-19:
Every few hours after flash from scratch (without preserve config and default .config) RM2100 auto-disabled wireless both bands, manually enable solves It but few hours late have same issue.

liwenjie119, 2022-06-20:
不保留配置升级，有线连接x86到cr660x或者newifi3作为ap并禁用dhcp，无线基本无网速或获取不到dns，x86的配置没有变化，不知道问题出在哪里

hyy-666, 2022-06-22:
有一个问题，不知道是不是个例，GHL用闭源的驱动开机没有启动wifi得手动启动，目前先刷padavan在用。
（ps：因为路由器急要用，拿了半小时测试忘记保存日志，周末补）

uniessey, 2022-06-22:
> 有一个问题，不知道是不是个例，GHL用闭源的驱动开机没有启动wifi得手动启动，目前先刷padavan在用。 （ps：因为路由器急要用，拿了半小时测试忘记保存日志，周末补）

可能是个古老的现象，试试在lan的自定义接口处添加ra0 rai0

Aes64X, 2022-06-24:
大佬，麻烦尝试增加下rb760igs，我试图抄一个来着。。。但这玩意有个SFP口，不知怎么处理。目前这个SFP口似乎不能用于VLAN拨号，也没有交换机选项，几乎没什么用。。。。。是我的备用路由，可以用来测试。

MageDelfador, 2022-06-24:
K2P刷了新固件发现一个小问题，设置端口转发后，外网可以成功连接转发端口，但本地内网无法连接。

VirtualboxWu, 2022-07-02:
不知道有没有人7621 玩单臂路由，新驱动好像严重影响速度。小白不知道怎么判断问题在哪

aiamadeus, 2022-07-03:
无线有问题的可以试试 revert 这个：[9cd22dd8293d8f071e7df0048949819dc06aeccc](https://github.com/coolsnowwolf/lede/commit/9cd22dd8293d8f071e7df0048949819dc06aeccc)
当然从这三周的测试数量来看我相信大家是一点都不在乎的，所以求别骂我 az

asiaon123, 2022-07-03:
@aiamadeus 

revert了 [9cd22dd8293d8f071e7df0048949819dc06aeccc](https://github.com/coolsnowwolf/lede/commit/9cd22dd8293d8f071e7df0048949819dc06aeccc) 无线还是不行，仍然没有速度，有线很好

机器：HC5962（B70），lede重新编译 完全重新刷入,已经取消勾选 软件流量负载 硬件流量负载

系统日志

```
Sun Jul  3 13:14:30 2022 kern.notice kernel: [    0.000000] Linux version 5.4.199 (zackgu@Zack-PC) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4618+1-706040d51)) #0 SMP Sun Jul 3 05:14:01 2022
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] MIPS: machine is HiWiFi HC5962
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun Jul  3 13:14:30 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Jul  3 13:14:30 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Jul  3 13:14:30 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Zone ranges:
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000]   HighMem  empty
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Early memory node ranges
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27024 r8192 d22128 u57344
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27024 r8192 d22128 u57344 alloc=14*4096
Sun Jul  3 13:14:30 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Sun Jul  3 13:14:30 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00010000
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00010000
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] Memory: 250084K/262144K available (6201K kernel code, 269K rwdata, 1304K rodata, 1236K init, 236K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Jul  3 13:14:30 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.007804] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.073975] pid_max: default: 32768 minimum: 301
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.078769] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.085968] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.096509] rcu: Hierarchical SRCU implementation.
Sun Jul  3 13:14:30 2022 kern.info kernel: [    0.101862] smp: Bringing up secondary CPUs ...
[... cut ...]
[  660.032670] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[  660.040326] br-lan: port 3(lan3) entered blocking state
[  660.045630] br-lan: port 3(lan3) entered forwarding state
[  891.947570] mt7530 mdio-bus:1f lan3: Link is Down
[  891.952401] br-lan: port 3(lan3) entered disabled state
[  981.385693] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[  981.393237] br-lan: port 3(lan3) entered blocking state
[  981.398485] br-lan: port 3(lan3) entered forwarding state
```

这是有线的速度， 无线的没打开过网页

[image]

asiaon123, 2022-07-04:
@lovehackintosh  好的，已修改 谢谢提醒

hcwang, 2022-07-05:
小米3Pro
新驱动下无线网络中5G正常工作，但2.4Ghz下，只有1M左右的下载速度，上传速度正常。在旧驱动下所有都正常。

286700234, 2022-07-06:
CR660X的WiFi现在修复好了吗？

liwenjie119, 2022-07-07:
没有进一步的更新修复，我就还停留在旧版本了。最后还能在家待一周，还有空测试，之后就彻底没空了

windybird, 2022-07-08:
new3-d2刷完后，需要无线重启才能正常工作

lovehackintosh, 2022-07-09:
> new3-d2刷完后，需要无线重启才能正常工作

可以尝试在系统->启动项->本地启动脚本的exit 0前面添加以下脚本
``` shell
# Wireless starter and Interface checker by Kenvix <i@kenvix.com>

# Start ra unconditionally
sleep 2s
ip link set rai0 up
ip link set ra0 up
echo "Wireless interface started"

# Lan Check
lanCheck=`uci get network.lan.ifname`
if [ $? -eq 0 ]; then
    echo $lanCheck | grep rai0 > /dev/null
    if [ $? -ne 0 ]; then
        uci set network.lan.ifname="$lanCheck rai0 ra0"
        uci commit
        echo "Updated wireless config of LAN Interface"
    fi
    echo "No need to update wireless config of LAN Interface"
else
    echo "wireless config of LAN Interface check failed. Interface may renamed." >&2
fi
```

windybird, 2022-07-09:
@lovehackintosh 可以用了，谢谢哈

windybird, 2022-07-09:
wifi 计划不起作用

modelsun, 2022-07-09:
> @lovehackintosh 可以用了，谢谢哈

用的是闭源WIFI驱动吗

lovehackintosh, 2022-07-09:
> > @lovehackintosh 可以用了，谢谢哈
> 
> 用的是闭源WIFI驱动吗

这只是一个WiFi启动脚本,跟驱动是否开源无关。驱动是否开源取决于你编译时选择的kmod,不过WiFi需要手动重启这个问题只在闭源驱动发生过

liwenjie119, 2022-07-09:
目前使用基本正常，但是在newifi3出现如下日志，不确定是否有bug
<details>
<summary>kernel</summary>
<pre><code>
[18537.344482] Rcv Wcid(2) AddBAReq
[18537.347721] Start Seq = 00000000
[18537.537765] Rcv Wcid(2) AddBAReq
[18537.541000] Start Seq = 00000000
[18778.030006] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[18778.037729] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(2)
[18778.044278] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[18778.044278] 
[19380.293230] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[19380.505960] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[19380.513695] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(1)
[19380.523411] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[19380.531117] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(1)
[19380.537868] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[19380.537868] 
[19380.575797] Rcv Wcid(1) AddBAReq
[19380.579017] Start Seq = 00000000
[19380.851782] Rcv Wcid(1) AddBAReq
[19380.855045] Start Seq = 00000001
[19444.971444] Rcv Wcid(1) AddBAReq
[19444.974675] Start Seq = 00000000
[22522.943510] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(20-F4-78-09-CB-25) !!!
[22522.951260] Receive EAPoL-Key frame from STA 20-F4-78-09-CB-25 wcid(1)
[22522.957787] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[22522.957787] 
[24732.191968] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[24733.527254] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24733.534999] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24733.607929] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24733.615663] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24733.622273] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[24733.622273] 
[24734.188023] Rcv Wcid(1) AddBAReq
[24734.191461] Start Seq = 00000018
[24734.638078] Rcv Wcid(1) AddBAReq
[24734.641473] Start Seq = 00000001
[24750.462971] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[24750.678054] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24750.685792] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24750.694707] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24750.702431] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24750.709020] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[24750.709020] 
[24750.768675] Rcv Wcid(1) AddBAReq
[24750.771935] Start Seq = 00000016
[24816.985735] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[24817.202677] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24817.210419] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24817.218897] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[24817.226621] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[24817.233223] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[24817.233223] 
[24817.893516] Rcv Wcid(1) AddBAReq
[24817.896747] Start Seq = 00000016
[25022.526457] Rcv Wcid(1) AddBAReq
[25022.529726] Start Seq = 00000001
[25022.692444] Rcv Wcid(1) AddBAReq
[25022.695777] Start Seq = 00000003
[25023.463894] Rcv Wcid(1) AddBAReq
[25023.467276] Start Seq = 00000003
[25179.568729] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[25179.783000] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[25179.790750] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[25179.799254] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[25179.806960] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[25179.813571] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[... 61 lines cut ...]
[26267.828375] 
[27893.622118] MIC Different in pairwise msg 2 of 4-way handshake!
[27894.664330] MIC Different in pairwise msg 2 of 4-way handshake!
[27895.747753] MIC Different in pairwise msg 2 of 4-way handshake!
[27896.820083] MIC Different in pairwise msg 2 of 4-way handshake!
[28764.630620] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[28764.849662] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[28764.857402] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[28764.865644] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[28764.873375] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[28764.879986] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[28764.879986] 
[28765.466998] Rcv Wcid(1) AddBAReq
[28765.470242] Start Seq = 00000017
[28855.129773] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[28855.348284] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[28855.356029] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[28855.364627] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(54-6C-EB-8B-3F-7B) !!!
[28855.372376] Receive EAPoL-Key frame from STA 54-6C-EB-8B-3F-7B wcid(1)
[28855.379033] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[28855.379033] 
[28855.791629] Rcv Wcid(1) AddBAReq
[28855.794907] Start Seq = 00000016
</code></pre>
</details>

aiamadeus, 2022-07-10:
> 目前使用基本正常，但是在newifi3出现如下日志，不确定是否有bug

你这个是 mtwifi 的日志，忽略即可。

eva2021hd, 2022-07-13:
# #昨天新编译了小娱C5固件，这次transmission正常。手头有个U盘，装了Centos。所以顺手做了HWNAT下的测试。
连接拓扑
PC---------（lan1）XY-C5（wan）--------PC(PPPoE)
PPPoE服务器那网卡IP是192.168.102.254，PPPoE程序使用rp-pppoe，拨号服务器IP为10.1.1.1，分配10.1.1.100-150。账号就是家里ISP账号。
路由配置已目前家用配置为主。测试以iperf3进行。
路由网络接口：
[image]

首先是网卡IP，不涉及PPPoE的测试。TCP和UDP都可以跑满1G带宽。具体见下图：
[image]

然后是PPPoE，TCP和UDP只可以跑满600M带宽。检查路由，负载很低只有3%，应该没过CPU。具体见下图：
[image]

最后以UDP测试，检查iperf3的服务器，发现服务器端口是TCP，指定UDP大小为1500后提示TCP MSS超过。按照提示的TCP MSS 1440为例，客户端发向PPPoE服务器可以满1G，从服务器发给客户端还是600M。具体见下图：
[image]

以上测试时，通过top监测路由CPU负载信息，CPU负载基本在0到3%。

还有拨号后，内核日志有大量的pppd错误日志。如下：
Tue Jul 12 12:54:56 2022 daemon.warn pppd[30439]: CCP: timeout sending Config-Requests
Tue Jul 12 12:54:59 2022 daemon.err pppd[30439]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:55:02 2022 daemon.err pppd[30439]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:55:05 2022 daemon.err pppd[30439]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:55:08 2022 daemon.err pppd[30439]: Received bad configure-rej:  12 06 00 00 00 00

<details><summary>具体pppd与接口wan的内核日志</summary>

<p>

```
Tue Jul 12 12:40:47 2022 daemon.notice netifd: Network device 'pppoe-wan' link is down
Tue Jul 12 12:40:47 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:40:47 2022 daemon.notice pppd[30675]: Connection terminated.
Tue Jul 12 12:40:48 2022 daemon.info pppd[30675]: Exit.
Tue Jul 12 12:40:48 2022 daemon.notice netifd: Interface 'wan' is now down
Tue Jul 12 12:40:48 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Jul 12 12:40:48 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Jul 12 12:40:48 2022 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jul 12 12:40:48 2022 daemon.notice netifd: wan (8033): uci: Entry not found
Tue Jul 12 12:40:48 2022 daemon.notice netifd: wan (8033): sh: out of range
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: Plugin pppoe.so loaded.
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: PPPoE plugin from pppd 2.4.9
Tue Jul 12 12:40:48 2022 daemon.notice pppd[8082]: pppd 2.4.9 started by root, uid 0
Tue Jul 12 12:40:48 2022 daemon.err pppd[8082]: Interface eth1 has MTU of 1492 -- should be at least 1500.
Tue Jul 12 12:40:48 2022 daemon.err pppd[8082]: This may cause serious connection problems.
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: PPP session is 5
Tue Jul 12 12:40:48 2022 daemon.warn pppd[8082]: Connected to 8c:89:a5:75:41:cd via interface eth1
Tue Jul 12 12:40:48 2022 kern.info kernel: [ 1488.570321] pppoe-wan: renamed from ppp0
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: Renamed interface ppp0 to pppoe-wan
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: Using interface pppoe-wan
Tue Jul 12 12:40:48 2022 daemon.notice pppd[8082]: Connect: pppoe-wan <--> eth1
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: syncppp not active
Tue Jul 12 12:40:48 2022 daemon.info pppd[8082]: CHAP authentication succeeded: Access granted
Tue Jul 12 12:40:48 2022 daemon.notice pppd[8082]: CHAP authentication succeeded
Tue Jul 12 12:40:48 2022 daemon.notice pppd[8082]: peer from calling number 8C:89:A5:75:41:CD authorized
Tue Jul 12 12:40:48 2022 daemon.err pppd[8082]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:40:48 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:40:49 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:40:49 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:40:49 2022 daemon.notice pppd[8082]: local  IP address 10.1.1.104
Tue Jul 12 12:40:49 2022 daemon.notice pppd[8082]: remote IP address 10.1.1.1
Tue Jul 12 12:40:49 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Tue Jul 12 12:40:49 2022 daemon.notice netifd: Interface 'wan' is now up
Tue Jul 12 12:40:51 2022 daemon.err pppd[8082]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:40:52 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Tue Jul 12 12:40:53 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Tue Jul 12 12:40:57 2022 daemon.err pppd[8082]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:41:01 2022 daemon.err pppd[8082]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:41:04 2022 daemon.err pppd[8082]: Received bad configure-rej:  12 06 00 00 00 00
[... 403 lines cut ...]
Tue Jul 12 12:56:14 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:14 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:56:14 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:56:14 2022 daemon.err miniupnpd[2684]: Failed to get IP for interface pppoe-wan
Tue Jul 12 12:56:14 2022 daemon.notice pppd[3425]: local  IP address 10.1.1.110
Tue Jul 12 12:56:14 2022 daemon.notice pppd[3425]: remote IP address 10.1.1.1
Tue Jul 12 12:56:14 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Tue Jul 12 12:56:14 2022 daemon.notice netifd: Interface 'wan' is now up
Tue Jul 12 12:56:15 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Tue Jul 12 12:56:16 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Tue Jul 12 12:56:17 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
[... cut ...]
Tue Jul 12 12:56:26 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:29 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:32 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:35 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:38 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00
Tue Jul 12 12:56:41 2022 daemon.err pppd[3425]: Received bad configure-rej:  12 06 00 00 00 00

```

</p>

</details>

eva2021hd, 2022-07-18:
最新发现。开启samba36或ftp。在打开大文件时会死机。无日志

lovehackintosh, 2022-07-19:
> 最新发现。开启samba36或ftp。在打开大文件时会死机。无日志

你超频了吗?是的话就把这个lede/target/linux/ramips/patches-5.4/991-mt7621-improve_cpu_clock.patch删掉重新编译再试试

lovehackintosh, 2022-07-20:
> > > 最新发现。开启samba36或ftp。在打开大文件时会死机。无日志
> > 
> > 
> > 你超频了吗?是的话就把这个lede/target/linux/ramips/patches-5.4/991-mt7621-improve_cpu_clock.patch删掉重新编译再试试
> 
> 刚刚看了下，lean代码修改里8天前这个文件已经被移除，我是7天前做的编译。编译时应该没这个文件。 小娱默认880mhz没有超频。

你看一下内核日志CPU Clock是不是880mhz

eva2021hd, 2022-07-21:
> > > > 最新发现。开启samba36或ftp。在打开大文件时会死机。无日志
> > > 
> > > 
> > > 你超频了吗?是的话就把这个lede/target/linux/ramips/patches-5.4/991-mt7621-improve_cpu_clock.patch删掉重新编译再试试
> > 
> > 
> > 刚刚看了下，lean代码修改里8天前这个文件已经被移除，我是7天前做的编译。编译时应该没这个文件。 小娱默认880mhz没有超频。
> 
> 你看一下内核日志CPU Clock是不是880mhz

前天重新云编译了一个。确认是880Mhz
出现问题很有意思，只要使用FTP（默认的那个）传输文件，设备立刻死机。日志看不到。

eva2021hd, 2022-07-21:
> > > > 最新发现。开启samba36或ftp。在打开大文件时会死机。无日志
> > > 
> > > 
> > > 你超频了吗?是的话就把这个lede/target/linux/ramips/patches-5.4/991-mt7621-improve_cpu_clock.patch删掉重新编译再试试
> > 
> > 
> > 刚刚看了下，lean代码修改里8天前这个文件已经被移除，我是7天前做的编译。编译时应该没这个文件。 小娱默认880mhz没有超频。
> 
> 你看一下内核日志CPU Clock是不是880mhz

前天重新云编译了一个。确认是880Mhz
	[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
	[    0.000000] Memory: 509288K/524288K available (6209K kernel code, 268K rwdata, 1304K rodata, 1292K init, 235K bss, 15000K reserved, 0K cma-reserved, 262144K highmem)
	[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
	[    0.000000] rcu: Hierarchical RCU implementation.
	[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
	[    0.000000] NR_IRQS: 256
	[    0.000000] CPU Clock: 880MHz
	[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
	[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
	[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
	[    0.007906] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
	[    0.074070] pid_max: default: 32768 minimum: 301
	
出现问题很有意思，只要使用FTP（默认的那个）传输文件，设备立刻死机。日志看不到。

付能看到的日志：

<details><summary>内核日志</summary>

<p>

```

[    0.000000] Linux version 5.4.203 (runner@fv-az71-530) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4727-6d79ece4c)) #0 SMP Tue Jul 19 06:51:16 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is XiaoYu XY-C5
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27152 r8192 d22000 u57344
[    0.000000] pcpu-alloc: s27152 r8192 d22000 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0000506c
[    0.000000] Readback ErrCtl register=0000506c
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509288K/524288K available (6209K kernel code, 268K rwdata, 1304K rodata, 1292K init, 235K bss, 15000K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[... 861 lines cut ...]
Thu Jul 21 08:31:35 2022 daemon.info smartd[5542]: Monitoring 0 ATA/SATA, 0 SCSI/SAS and 0 NVMe devices
Thu Jul 21 08:31:36 2022 daemon.err aria2c[6313]: Could not open /dev/crypto: Permission denied
Thu Jul 21 08:31:38 2022 daemon.notice procd: /etc/rc.d/S99nfsd: exportfs: /mnt does not support NFS export
Thu Jul 21 08:31:39 2022 kern.warn kernel: [   73.827858] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
Thu Jul 21 08:31:39 2022 kern.warn kernel: [   73.843743] NFSD: Using legacy client tracking operations.
Thu Jul 21 08:31:39 2022 kern.info kernel: [   73.854930] NFSD: starting 10-second grace period (net f00000a0)
Thu Jul 21 08:31:40 2022 daemon.notice rpc.mountd[7349]: Version 2.5.4 starting
Thu Jul 21 08:31:40 2022 daemon.notice procd: /etc/rc.d/S99set-irq-affinity: /etc/rc.common: line 30: can't create /proc/irq/24/smp_affinity: nonexistent directory
Thu Jul 21 08:31:40 2022 daemon.notice rpc.statd[7348]: Version 2.5.4 starting
Thu Jul 21 08:31:40 2022 daemon.warn rpc.statd[7348]: Flags: No-Daemon TI-RPC
Thu Jul 21 08:31:41 2022 daemon.notice rpc.statd[7348]: Initializing NSM state
Thu Jul 21 08:31:41 2022 daemon.warn rpc.statd[7348]: Running as root.  chown /var/lib/nfs to choose different user
Thu Jul 21 08:31:42 2022 user.notice transmission: Starting with 255290000 virt mem
Thu Jul 21 08:31:50 2022 daemon.info procd: - init complete -
Thu Jul 21 08:31:53 2022 daemon.err transmission-daemon[7874]: Could not open /dev/crypto: Permission denied
Thu Jul 21 08:32:10 2022 authpriv.info dropbear[13879]: Child connection from 192.168.100.111:49778
Thu Jul 21 08:32:14 2022 authpriv.notice dropbear[13879]: Password auth succeeded for 'root' from 192.168.100.111:49778
Thu Jul 21 08:32:14 2022 authpriv.info dropbear[14010]: Child connection from 192.168.100.111:49779
Thu Jul 21 08:32:16 2022 authpriv.notice dropbear[14010]: Password auth succeeded for 'root' from 192.168.100.111:49779

```

</p>

</details>

uniessey, 2022-07-26:
> > > > > 最新发现。开启samba36或ftp。在打开大文件时会死机。无日志
> > > > 
> > > > 
> > > > 你超频了吗?是的话就把这个lede/target/linux/ramips/patches-5.4/991-mt7621-improve_cpu_clock.patch删掉重新编译再试试
> > > 
> > > 
> > > 刚刚看了下，lean代码修改里8天前这个文件已经被移除，我是7天前做的编译。编译时应该没这个文件。 小娱默认880mhz没有超频。
> > 
> > 
> > 你看一下内核日志CPU Clock是不是880mhz
> 
> 前天重新云编译了一个。确认是880Mhz [ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off [ 0.000000] Memory: 509288K/524288K available (6209K kernel code, 268K rwdata, 1304K rodata, 1292K init, 235K bss, 15000K reserved, 0K cma-reserved, 262144K highmem) [ 0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 [ 0.000000] rcu: Hierarchical RCU implementation. [ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies. [ 0.000000] NR_IRQS: 256 [ 0.000000] CPU Clock: 880MHz [ 0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns [ 0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns [ 0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns [ 0.007906] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688) [ 0.074070] pid_max: default: 32768 minimum: 301
> 
> 出现问题很有意思，只要使用FTP（默认的那个）传输文件，设备立刻死机。日志看不到。
> 
> 付能看到的日志：
> 
> 内核日志
> 系统日志

我隐约记得一个古老的问题，SMB、NFS、FTP与acc里的bbr加速有冲突，古早之前的表现是限速。

lovehackintosh, 2022-07-27:
@uniessey 貌似设备性能差，传输流量大会导致死机

aiamadeus, 2022-08-09:
从 commit 665970a97d9f38d6c69fe73a52f447ccc123217e 开始使用 switch 驱动的设备已支持将某个网口设置为强制千兆模式，详见：https://www.right.com.cn/forum/thread-3640614-1-1.html

maxmib, 2022-08-10:
使用vlan在主路由和下级ap之间划分访客网络时，新版驱动的新路由三固件会导致访客无线网络速度只有两百kB/s 左右，而主无线网络正常，访客网络是新建接口里桥接访客无线和vlan网口，换回老版驱动固件则正常。

q397333, 2022-08-28:
大佬，我按照注释里写的，K2P编译的时候勾选了turboacc插件里的flow offload选项，但是刷好后进入turboacc界面并没有找到关闭flow offload的选项，是哪里出问题了吗？
[image]

lovehackintosh, 2022-08-29:
> 大佬，我按照注释里写的，K2P编译的时候勾选了turboacc插件里的flow offload选项，但是刷好后进入turboacc界面并没有找到关闭flow offload的选项，是哪里出问题了吗？

现在Luci  更新了,检测到mtkhwnat会自动禁用flow offload,不用手动设置

aiamadeus, 2022-08-30:
从 commit 1ccc3bfd54f7d9a2a2a02f2cb2b062452ea8d830 开始除极个别机子外 dsa 的 dual gmac 也已经支持，跟 gsw 驱动一样解锁 2Gbps 的有线速率上限。

uniessey, 2022-08-30:
> 从 commit [1ccc3bf](https://github.com/coolsnowwolf/lede/commit/1ccc3bfd54f7d9a2a2a02f2cb2b062452ea8d830) 开始除极个别机子外 dsa 的 dual gmac 也已经支持，跟 gsw 驱动一样解锁 2Gbps 的有线速率上限。

感谢大佬们的辛苦工作。
我猜测一下，D2应该不在此范围内吧？^^!

aiamadeus, 2022-08-30:
> 感谢大佬们的辛苦工作。
> 我猜测一下，D2应该不在此范围内吧？^^!

是的，因为新三不硬改开启不了 rgmii2

asiaon123, 2022-08-30:
hc5962 刷了最新的 无线速率应该是正常了 但是有线又没有速度了，跟原来无线的一样的表现只有1M左右

[14 more replies]

closed 2023-07-08 by aiamadeus