coolsnowwolf/lede #9461: Dnsmasq的“记录查询日志”选项无效

zhouym233, 2022-06-01:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
Dnsmasq的“记录查询日志”选项无效，启用该选项后系统日志中没有出现dnsmasq的DNS查询日志。
与OpenWrt官方版本的行为不一致，这不利于在上游DNS故障时排查问题。
[image]

### (2) 路由器型号和固件版本

A：
YouHua WR1200JS ;
OpenWrt R22.5.5，且在更早的版本中长期存在

### (3) 详细日志

A：
```
Wed Jun  1 17:45:11 2022 kern.notice kernel: [    0.000000] Linux version 5.4.195 (ubuntu@VM-20-11-ubuntu) (gcc version 9.3.0 (OpenWrt GCC 9.3.0 r4151-5384d96c5)) #0 SMP Tue May 31 15:00:41 2022
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] MIPS: machine is YouHua WR1200JS
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Wed Jun  1 17:45:11 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Jun  1 17:45:11 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Jun  1 17:45:11 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Zone ranges:
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000]   HighMem  empty
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Early memory node ranges
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Wed Jun  1 17:45:11 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Wed Jun  1 17:45:11 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00012090
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00012090
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] Memory: 120392K/131072K available (6115K kernel code, 251K rwdata, 1280K rodata, 1236K init, 238K bss, 10680K reserved, 0K cma-reserved, 0K highmem)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Wed Jun  1 17:45:11 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.007813] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.073974] pid_max: default: 32768 minimum: 301
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.078772] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.085973] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun  1 17:45:11 2022 kern.info kernel: [    0.096656] rcu: Hierarchical SRCU implementation.
[... 531 lines cut ...]
Wed Jun  1 17:45:30 2022 kern.info kernel: [   40.085213] br-lan: port 5(wlan0) entered forwarding state
Wed Jun  1 17:45:30 2022 daemon.notice netifd: Network device 'wlan0' link is up
Wed Jun  1 17:45:30 2022 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Wed Jun  1 17:45:30 2022 daemon.notice hostapd: wlan0: AP-ENABLED
Wed Jun  1 17:46:13 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** IEEE 802.11: authenticated
Wed Jun  1 17:46:13 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED **:**:**:**:**:** 2
Wed Jun  1 17:46:13 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** IEEE 802.11: associated (aid 2)
Wed Jun  1 17:46:13 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED **:**:**:**:**:**
Wed Jun  1 17:46:13 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** RADIUS: starting accounting session D4CD09EBE401BBE2
Wed Jun  1 17:46:13 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** WPA: pairwise key handshake completed (RSN)
Wed Jun  1 17:46:13 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED **:**:**:**:**:**
Wed Jun  1 17:46:47 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** IEEE 802.11: authenticated
Wed Jun  1 17:46:47 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** IEEE 802.11: associated (aid 3)
Wed Jun  1 17:46:47 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED **:**:**:**:**:**
Wed Jun  1 17:46:47 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** RADIUS: starting accounting session 868D56F63DCD4059
Wed Jun  1 17:46:47 2022 daemon.info hostapd: wlan1: STA **:**:**:**:**:** WPA: pairwise key handshake completed (RSN)
Wed Jun  1 17:46:47 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED **:**:**:**:**:**
Wed Jun  1 17:46:57 2022 daemon.info hostapd: wlan0: STA **:**:**:**:**:** IEEE 802.11: authenticated
Wed Jun  1 17:46:57 2022 daemon.info hostapd: wlan0: STA **:**:**:**:**:** IEEE 802.11: associated (aid 1)
Wed Jun  1 17:46:57 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED **:**:**:**:**:**
Wed Jun  1 17:46:57 2022 daemon.info hostapd: wlan0: STA **:**:**:**:**:** RADIUS: starting accounting session F8FF0DF705B0C57E
Wed Jun  1 17:46:57 2022 daemon.info hostapd: wlan0: STA **:**:**:**:**:** WPA: pairwise key handshake completed (RSN)
Wed Jun  1 17:46:57 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED **:**:**:**:**:**

```

chobitsos, 2022-06-02:
/etc/dnsmasq.conf里删除log-facility=/dev/null

lphgor, 2022-06-02:
源码删除这行：https://github.com/coolsnowwolf/lede/blob/2a33e979c850c9bf214e6713a9e39ada029e2381/package/lean/default-settings/files/zzz-default-settings#L54

云编译的话，`diy-part2.sh`里面加上这行：
```sh
sed -i '/log-facility/d' ./package/lean/default-settings/files/zzz-default-settings
```

zhouym233, 2022-06-02:
已解决，感谢大家的帮助

closed 2022-06-02 by zhouym233