coolsnowwolf/lede #9472: r8168的pcie网卡用一段时间就掉卡，搞了快2周，实在是不知道怎么整了，所以要在这里麻烦一下了

taoqiko, 2022-06-03:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)已经搜索，没有看到解决的

 没有类似的issue
2. 详细叙述
(1) 具体问题
A：运行一段时间之后，eth2网卡就掉了，用的dq77kb板子，两个板载，eth0是lan,eth1一个wan口，然后我还有一条网线所以，用了一个r8168的pcie网卡作为eht2做wan口拨号。eth2断了之后我就需要重启才能再次拨号连接。

(2) 路由器型号和固件版本
主机名	OpenWrt
型号	/DQ77KB - Intel(R) Core(TM) i5-3475S CPU @ 2.90GHz : 4 Core 4 Thread (CpuMark : 80108.814473 Scores)
CPU 信息	1658.285 MHz +30.0°C (high = +93.0°C, crit = +103.0°C)
固件版本	OpenWrt R22.5.5 / LuCI Master (git-22.138.16277-772b531)
内核版本	5.4.195

(3) 详细日志
系统日志

Tue May 31 06:28:27 2022 daemon.err insmod: module is already loaded - slhc
Tue May 31 06:28:27 2022 daemon.err insmod: module is already loaded - ppp_generic

Tue May 31 06:28:42 2022 daemon.info pppd[1062]: Exit.
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is now down
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is disabled
Tue May 31 06:28:42 2022 kern.info kernel: [40112.708722] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is enabled
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is setting up now
Tue May 31 06:28:42 2022 daemon.notice netifd: Network device 'eth2' link is down
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Tue May 31 06:28:42 2022 daemon.err insmod: module is already loaded - slhc
Tue May 31 06:28:42 2022 daemon.err insmod: module is already loaded - ppp_generic
Tue May 31 06:28:42 2022 daemon.err insmod: module is already loaded - pppox
Tue May 31 06:28:42 2022 daemon.err insmod: module is already loaded - pppoe
Tue May 31 06:28:42 2022 daemon.notice netifd: wan1 (1306): Command failed: Permission denied
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is now down
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is disabled
Tue May 31 06:28:42 2022 kern.info kernel: [40112.791870] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
Tue May 31 06:28:42 2022 daemon.notice netifd: Interface 'wan1' is enabled
Tue May 31 06:28:43 2022 user.notice mwan3[1337]: Execute ifdown event on interface wan1 (unknown)
Tue May 31 06:28:43 2022 user.info mwan3track[11602]: Detect ifdown event on interface wan1 (pppoe-wan1)
Tue May 31 06:28:43 2022 user.notice mwan3track[11602]: Interface wan1 (pppoe-wan1) is offline
Tue May 31 06:28:46 2022 kern.info kernel: [40116.919128] r8168: eth2: link up
Tue May 31 06:28:46 2022 daemon.notice netifd: Network device 'eth2' link is up
Tue May 31 06:28:46 2022 daemon.notice netifd: Interface 'wan1' has link connectivity
Tue May 31 06:28:46 2022 daemon.notice netifd: Interface 'wan1' is setting up now
Tue May 31 06:28:46 2022 daemon.err insmod: module is already loaded - slhc
Tue May 31 06:28:46 2022 daemon.err insmod: module is already loaded - ppp_generic
Tue May 31 06:28:46 2022 daemon.err insmod: module is already loaded - pppox
Tue May 31 06:28:46 2022 daemon.err insmod: module is already loaded - pppoe
Tue May 31 06:28:46 2022 daemon.info pppd[1719]: Plugin pppoe.so loaded.
Tue May 31 06:28:46 2022 daemon.info pppd[1719]: PPPoE plugin from pppd 2.4.9
Tue May 31 06:28:47 2022 daemon.notice pppd[1719]: pppd 2.4.9 started by root, uid 0
Tue May 31 06:29:02 2022 daemon.warn pppd[1719]: Timeout waiting for PADO packets
Tue May 31 06:29:02 2022 daemon.err pppd[1719]: Unable to complete PPPoE Discovery
Tue May 31 06:29:02 2022 daemon.info pppd[1719]: Exit.
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' is now down
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' is disabled
Tue May 31 06:29:02 2022 kern.info kernel: [40132.080779] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' is enabled
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' is setting up now
Tue May 31 06:29:02 2022 daemon.notice netifd: Network device 'eth2' link is down
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Tue May 31 06:29:02 2022 daemon.err insmod: module is already loaded - slhc
Tue May 31 06:29:02 2022 daemon.err insmod: module is already loaded - ppp_generic
Tue May 31 06:29:02 2022 daemon.err insmod: module is already loaded - pppox
Tue May 31 06:29:02 2022 daemon.err insmod: module is already loaded - pppoe
Tue May 31 06:29:02 2022 daemon.notice netifd: wan1 (1931): Command failed: Permission denied
Tue May 31 06:29:02 2022 daemon.notice netifd: Interface 'wan1' is now down
[... 200 lines cut ...]
[27483.120549] r8168: eth2: link up
[27498.273938] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27498.353167] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27502.480459] r8168: eth2: link up
[27517.633910] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27517.705099] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27520.816366] r8168: eth2: link up
[27535.961754] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27536.049060] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27540.176262] r8168: eth2: link up
[27555.330075] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27555.404775] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27559.536165] r8168: eth2: link up
[27574.673412] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27574.748969] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27578.864067] r8168: eth2: link up
[27594.025372] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27594.100593] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27597.199983] r8168: eth2: link up
[27612.385511] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27612.456425] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27616.591874] r8168: eth2: link up
[27631.749362] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27631.824444] eth2: 0xffffc90000041000, 00:e0:4c:69:51:73, IRQ 16
[27634.927786] r8168: eth2: link up

minlang112, 2022-06-03:
r8168做wan未遇到此情况

coolsnowwolf, 2022-06-07:
ASPM 问题

closed 2022-06-07 by coolsnowwolf