coolsnowwolf/lede #9474: 更新最新源码后出错

mitsukileung, 2022-06-03:
X86

mkdir -p /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/koyko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koyko/lede/build_dir/target-x86_64_musl/luci-app-ttyd/.pkgdir/luci-i18n-ttyd-zh-cn/. /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-ttyd-zh-cn_installed
echo "luci-i18n-ttyd-zh-cn" >> /home/koyko/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-ttyd.default.install
make[3]: Leaving directory '/home/koyko/lede/feeds/luci/applications/luci-app-ttyd'
time: package/feeds/luci/luci-app-ttyd/compile#0.53#0.13#0.64
make[3]: Entering directory '/home/koyko/lede/package/lean/shortcut-fe/shortcut-fe'
rm -f /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built
touch /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built_check
make -j1 -C "/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.43" KCFLAGS="-fmacro-prefix-map=/home/koyko/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/koyko/lede/staging_dir/host/include -I/home/koyko/lede/staging_dir/hostpkg/include -I/home/koyko/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Jun  3 05:46:28 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/koyko/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/nat46.symvers" KERNELRELEASE=5.15.43  M="/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y  modules
make[4]: Entering directory '/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.43'
  CC [M]  /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.o
/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.c: In function 'sfe_cm_post_routing':
/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.c:505:19: error: 'struct netns_ct' has no member named 'sysctl_no_window_check'
  if ((net&&net->ct.sysctl_no_window_check)
                   ^
make[5]: *** [scripts/Makefile.build:288: /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.o] Error 1
make[4]: *** [Makefile:1868: /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
make[4]: Leaving directory '/home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.43'
make[3]: *** [Makefile:92: /home/koyko/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/koyko/lede/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#1.13#0.21#1.35
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/koyko/lede'
make[1]: *** [package/Makefile:110: /home/koyko/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koyko/lede'
make: *** [/home/koyko/lede/include/toplevel.mk:230：world] 错误 2

RodmanWang, 2022-06-03:
因为默认改成了shortcut-fe，编译这个确实出错。

mitsukileung, 2022-06-03:
> 

改成Offloading 就可以？

RodmanWang, 2022-06-03:
你试试不就知道了，我好像能

ricksuzade-maker, 2022-06-03:
我也遇到同样问题了，编译shortcut-fe出错

mitsukileung, 2022-06-03:
> 我也遇到同样问题了，编译shortcut-fe出错

删除之前的配置，make menuconfig 时把SFE改回 OFFLOADING可以编译成功

closed 2022-06-04 by mitsukileung