coolsnowwolf/lede #9482: 最近的版本在PVE虚拟机上偶尔几个小时就崩溃软重启

GreatMichaelLee, 2022-06-04:
如题，在pve上装的lede，另外还有一个windows和一个ubuntu，三个虚拟机本身都没问题，已稳定运行数天，但是lede会莫名其妙的崩溃软重启，每次不会超过12个小时，而另外windows/ubuntu却没问题，所以判定不是虚拟机本身的问题，不知道trigger point是什么，好像每次都是访问web界面就挂了（不确定）.虚拟机各项设置都没什么问题，内存也设置的足够大。

我看最近修了几个kernel crash的issue，最近一个好像还回退了（没修好？），似乎也没有解决问题。因为kernel crash没有日志，所以也没法提供信息。这个问题让我很头疼，麻烦大佬指点迷津。

GreatMichaelLee, 2022-06-05:
因为这个问题的确是没日志，不是不按格式反馈哈，就拿这个问题跟一下那几个kernel crash的fix把，求大佬别关，谢谢。

WYC-2020, 2022-06-05:
建议关闭直通在看

GreatMichaelLee, 2022-06-05:
> 建议关闭直通在看

非直通能支持2.5G吗？直通主要是为了破千兆限制

minlang112, 2022-06-05:
> > 建议关闭直通在看
> 
> 非直通能支持2.5G吗？直通主要是为了破千兆限制

可以啊，PVE可以驱动8125和I225的呀

zhangguanzhang, 2022-06-06:
除了上面几个大佬的建议以外。如果问题还没解决
你自己能编译的话试试把内核 5.4 5.10 5.15的版本编译三个固件做个对照组，看看是不是内核的问题。
死机重启的情况我不知道重启后 logread 命令能否读取到kernal 的 crash 日志，有的话你贴上来看看有大佬能解决不

GreatMichaelLee, 2022-06-09:
这次抓到了一些日志，这几行是不是会引起reboot啊

Thu Jun  9 13:31:25 2022 authpriv.notice dropbear[23899]: Password auth succeeded for 'root' from 10.0.0.5:51732
**Thu Jun  9 13:47:42 2022 kern.notice kernel: [   12.006648] random: crng init done
Thu Jun  9 13:47:42 2022 kern.notice kernel: [   12.007456] random: 56 urandom warning(s) missed due to ratelimiting**
Thu Jun  9 13:47:42 2022 authpriv.info dropbear[2862]: Not backgrounding
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: 8021ad
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: 8021q
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: macvlan
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: veth
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: bridge
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: Network device
Thu Jun  9 13:47:43 2022 user.notice : Added device handler type: tunnel
**Thu Jun  9 13:47:43 2022 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Thu Jun  9 13:47:43 2022 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.**
Thu Jun  9 13:47:43 2022 cron.err crond[3422]: crond (busybox 1.35.0) started, log level 8
Thu Jun  9 13:47:43 2022 daemon.notice procd: /etc/rc.d/S50netserver: Starting netserver with host 'IN(6)ADDR_ANY' port '12865' and fam
Thu Jun  9 13:47:43 2022 kern.info kernel: [   12.985656] br-lan: port 1(eth2) entered blocking state
Thu Jun  9 13:47:43 2022 kern.info kernel: [   12.986599] br-lan: port 1(eth2) entered disabled state
Thu Jun  9 13:47:43 2022 kern.info kernel: [   12.987593] device eth2 entered promiscuous mode
Thu Jun  9 13:47:43 2022 daemon.notice netifd: Interface 'lan' is enabled
Thu Jun  9 13:47:43 2022 daemon.notice netifd: Interface 'lan' is setting up now
Thu Jun  9 13:47:43 2022 daemon.notice netifd: Interface 'lan' is now up
Thu Jun  9 13:47:43 2022 daemon.notice netifd: Interface 'WAN2' is enabled
Thu Jun  9 13:47:43 2022 daemon.notice netifd: Interface 'WAN2' is setting up now

GreatMichaelLee, 2022-06-09:
by the way: 这个autoreboot服务是干嘛的，什么场景下用到？我看他在init服务里

zhangguanzhang, 2022-06-09:
> by the way: 这个autoreboot服务是干嘛的，什么场景下用到？我看他在init服务里

看看开启了没：
```
uci show autoreboot
```
我看了它是在 /etc/crontabs/root 里加定时计划的，你也看下 /etc/crontabs/root
不过估计不是autoreboot的问题

GreatMichaelLee, 2022-06-09:
> > by the way: 这个autoreboot服务是干嘛的，什么场景下用到？我看他在init服务里
> 
> 看看开启了没：
> 
> ```
> uci show autoreboot
> ```
> 
> 我看了它是在 /etc/crontabs/root 里加定时计划的，你也看下 /etc/crontabs/root 不过估计不是autoreboot的问题

autoreboot里没有定时reboot得，倒是有个coremark（之前删了，刷了一次新code不知道怎么又来了），先删了看看。

WYC-2020, 2022-06-10:
这玩意是设置自动重启用的，自己编译了那些插件难道不清楚

closed 2022-06-12 by coolsnowwolf