coolsnowwolf/lede #9483: 勾选Wireguard编译出错，提示缺少libblake2s.ko

librec, 2022-06-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：勾选Wireguard编译出错

### (2) 路由器型号和固件版本

A：竞斗云2.0 使用P&W R619AC 128M

### (3) 详细日志

A：
make[3]: Leaving directory '/home/librec/lede/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.15#0.02#0.17
make[3]: Entering directory '/home/librec/lede/package/kernel/linux'
mkdir -p /home/librec/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp

SHELL= flock /home/librec/lede/tmp/.root-copy.flock -c 'cp -fpR /home/librec/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/packages/.pkgdir/kernel/. /home/librec/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/'
touch /home/librec/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/stamp/.kernel_installed
Package kmod-wireguard is missing dependencies for the following libraries:
libblake2s.ko
make[3]: *** [modules/netsupport.mk:1313: /home/librec/lede/bin/targets/ipq40xx/generic/packages/kmod-wireguard_5.4.195-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[3]: Leaving directory '/home/librec/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.62#0.09#1.74
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/librec/lede'
make[1]: *** [package/Makefile:110: /home/librec/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/librec/lede'
make: *** [/home/librec/lede/include/toplevel.mk:230: world] Error 2

提示少依赖libblake2s.ko

看了下最近更新的内核说明

kernel: bump 5.15 to 5.15.44

package/kernel/linux/modules/netsupport.mk
@@ -1290,7 +1290,6 @@ define KernelPackage/wireguard
  SUBMENU:=$(NETWORK_SUPPORT_MENU)
  TITLE:=WireGuard secure network tunnel
  DEPENDS:= \
	  +kmod-crypto-lib-blake2s \
	  +kmod-crypto-lib-chacha20poly1305 \
	  +kmod-crypto-lib-curve25519 \
	  +kmod-udptunnel4 \

@@ -1290,7 +1290,6 @@ define KernelPackage/wireguard
  SUBMENU:=$(NETWORK_SUPPORT_MENU)
  TITLE:=WireGuard secure network tunnel
  DEPENDS:= \
	  +kmod-crypto-lib-blake2s \
	  +kmod-crypto-lib-chacha20poly1305 \
	  +kmod-crypto-lib-curve25519 \
	  +kmod-udptunnel4 \

大佬帮看一下本地的话怎么手动解决

librec, 2022-06-05:
但是这个应该不影响5.4.195内核啊，感觉可能是我自己配置的问题，全部推倒重来再测一下。

librec, 2022-06-05:
测试完毕，全默认设置仅勾选luci-app-wireguard同样错误

Portan, 2022-06-05:
我这边也是一样的错误，需要大佬指导下如何修复

Bonnie2980, 2022-06-05:
编译k2t，也是一样的错误

librec, 2022-06-05:
先临时用老版本编译吧
先 git log 一下
commit 91b3516ba1c116f6f7d78ffe742f8806c1001511 (HEAD -> master, origin/master, origin/HEAD)
Author: aakkll <94471752+aakkll@users.noreply.github.com>
Date:   Sat Jun 4 23:25:35 2022 +0800

    kernel: bump 5.15 to 5.15.44 (#9457)

    Signed-off-by: aakkll <94471752+aakkll@users.noreply.github.com>

commit 98f703f4b179263607004f50143ffebddb4fb769
Author: AmadeusGhost <42570690+AmadeusGhost@users.noreply.github.com>
Date:   Tue May 24 23:40:20 2022 +0800

    r8168: refresh patches

    Fixes: #9480

commit 78b1611182fb11461ec04b2976c67f460f5c540c
Author: AmadeusGhost <42570690+AmadeusGhost@users.noreply.github.com>
Date:   Tue May 24 23:08:03 2022 +0800

    r8168: update to 8.050.02

commit 9fdd43d6ed3e41c261b2b14cc3ac04bde21bb14a
Author: lean <coolsnowwolf@gmail.com>
Date:   Sat Jun 4 00:16:27 2022 +0800

    generic: 5.15: revert sysctl_no_window_check

看到每一个commit之后有一串字符 哈希值 选定一个
然后 git checkout 这一串字符
我用的 git checkout 9fdd43d6ed3e41c261b2b14cc3ac04bde21bb14a
返回4号凌晨的这个版本 然后正常编译就行了

Jejz168, 2022-06-05:
今天还以为我这边有问题，一直package/kernel/linux failed to build过不去。原来是源码kmod模块被删了部分kbuild

Curious-r, 2022-06-05:
同样的问题
[image]

Jejz168, 2022-06-05:
[image]

删掉这个依赖就没问题了

fengqi, 2022-06-06:
https://github.com/coolsnowwolf/lede/pull/9457 最新的代码已经处理掉了，issue可关闭了。

sjtuross, 2022-06-06:
issue提的是5.4内核，现在还是有问题的

xcf13363175, 2022-06-06:
5.10也有问题

Jejz168, 2022-06-06:
没考虑到arm机器。

WYC-2020, 2022-06-07:
其实根本没必要完全删除这些模块，只是linux5.15 默认编译了libblake2s这个模块而已
[image]
个人感觉完全可以
[image]
这样，加入限制，其他都不用改也不用删除模块，应该就能兼容所有的，其实不加入限制都可以，最多也就是不受控制而已

WYC-2020, 2022-06-07:
如果要编译5.4或者5.10的话，暂时这样修复
[image]
如果5.10升级到5.10.119后可以去掉5.10的限制，如果从来不用5.4 那就可以直接删除这些模块，先撤销5.15.44升级所带来的模块修改,内核升级不用改，在按这个步骤修改，就可以编译通过，如果后期5.4也按5.10这样修改了那就可以完全去掉这些模块了

Jejz168, 2022-06-07:
> 如果这些编译 5.4 或者 10句话，暂时 如果修复 5.10 升级到 5.10.119 后可以直接取消 5.10.10 的限制，如果从始至终使用 5.4 那就可以删除模块，先撤销 5.5.44 升级所带来的修改，内核也升级不用改，在这个修改，就可以通过修改，然后5.4按这样就可以完全按这些模块了 [image]

不要wireguard就可以了。现在ssrp也是。太频繁了。还是不要编译了

WYC-2020, 2022-06-07:
> > 如果这些编译 5.4 或者 10句话，暂时 如果修复 5.10 升级到 5.10.119 后可以直接取消 5.10.10 的限制，如果从始至终使用 5.4 那就可以删除模块，先撤销 5.5.44 升级所带来的修改，内核也升级不用改，在这个修改，就可以通过修改，然后5.4按这样就可以完全按这些模块了 [image]
> 
> 不要wireguard就可以了。现在ssrp也是。太频繁了。还是不要编译了

这玩意我都不用只是给他们提供解决方案

Jejz168, 2022-06-07:
> > > 如果这些编译 5.4 或者 10句话，暂时 如果修复 5.10 升级到 5.10.119 后可以直接取消 5.10.10 的限制，如果从始至终使用 5.4 那就可以删除模块，先撤销 5.5.44 升级所带来的修改，内核也升级不用改，在这个修改，就可以通过修改，然后5.4按这样就可以完全按这些模块了 [image]
> > 
> > 
> > 不要wireguard就可以了。现在ssrp也是。太频繁了。还是不要编译了
> 
> 这玩意我都不用只是给他们提供解决方案

嗯、太频繁的插件。建议不要编译、稳定下来在编译

fengqi, 2022-06-07:
> > 如果这些编译 5.4 或者 10句话，暂时 如果修复 5.10 升级到 5.10.119 后可以直接取消 5.10.10 的限制，如果从始至终使用 5.4 那就可以删除模块，先撤销 5.5.44 升级所带来的修改，内核也升级不用改，在这个修改，就可以通过修改，然后5.4按这样就可以完全按这些模块了 [image]
> 
> 不要编译wireguard就可以了。现在ssrp也是。太频繁了。还是不要编译了

wireguard还是很有必要的，国内连回家、两家组网。

closed 2022-06-07 by AmadeusGhost

WYC-2020, 2022-06-08:
> > 如果要编译5.4或者5.10的话，暂时这样修复 [image] 如果5.10升级到5.10.119后可以去掉5.10的限制，如果从来不用5.4 那就可以直接删除这些模块，先撤销5.15.44升级所带来的模块修改,内核升级不用改，在按这个步骤修改，就可以编译通过，如果后期5.4也按5.10这样修改了那就可以完全去掉这些模块了
> 
> 如图做了修改，编译的时候还是出现一样的错误信息。 [image]

我亲测过没问题，自己哪里没操作对

Curious-r, 2022-06-08:
> > 如果要编译5.4或者5.10的话，暂时这样修复 [image] 如果5.10升级到5.10.119后可以去掉5.10的限制，如果从来不用5.4 那就可以直接删除这些模块，先撤销5.15.44升级所带来的模块修改,内核升级不用改，在按这个步骤修改，就可以编译通过，如果后期5.4也按5.10这样修改了那就可以完全去掉这些模块了
> 
> 如图做了修改，编译的时候还是出现一样的错误信息。 [image]

5.4应该已经在源码里修复了，你尝试在menuconfig里取消wireguard然后再勾选一次，重新save一次.config。如果你上个版本的.config是在源码修复前生成的，那配置里可能会缺少模块的声明。

WYC-2020, 2022-06-08:
> 5.4应该已经在源码里修复了，你尝试在menuconfig里取消wireguard然后再勾选一次，重新save一次.config。如果你上个版本的.config是在源码修复前生成的，那配置里可能会缺少模块的声明。

这种修复我猜测应该5.10编译也不会过，没验证，除非5.10升级到119以上，但是目前还没升所以应该会出现一样错误，你可以试下

klever1988, 2022-06-08:
> 5.4应该已经在源码里修复了，你尝试在menuconfig里取消wireguard然后再勾选一次，重新save一次.config。如果你上个版本的.config是在源码修复前生成的，那配置里可能会缺少模块的声明。

谢谢，我试试看

klever1988, 2022-06-08:
拿掉LINUX_5_10，并且重新make defconfig，可以了

WYC-2020, 2022-06-09:
> 拿掉LINUX_5_10，并且重新make defconfig，可以了

说过那样修复有问题，我刚已经验证，如果5.10没升级到5.10.119以上 就必须加入这个，要不然会空包，l大仓库目前的修复是不行的必须按我那个方式修复，给你来点实际的证据：
这个图是l大修复的方式编译5.10
[image]
这个我的修复方式的截图
[image]

klever1988, 2022-06-09:
> > 拿掉LINUX_5_10，并且重新make defconfig，可以了
> 
> 说过那样修复有问题，我刚已经验证，如果5.10没升级到5.10.119以上 就必须加入这个，要不然会空包，l大仓库目前的修复是不行的必须按我那个方式修复，给你来点实际的证据： 这个图是l大修复的方式编译5.10 [image] 这个我的修复方式的截图 [image]

谢谢！我试试

Hzend, 2022-06-10:
对于arm系统的，现在有解决方案了吗？
看了上面的讨论，还是没能解决，求指点

WYC-2020, 2022-06-10:
> 对于arm系统的，现在有解决方案了吗？ 看了上面的讨论，还是没能解决，求指点

这东西都一样，arm也是这样，再说又不说清楚arm啥问题，是编译不过还是编译运行不了，我那个修复方式是完全没问题的，不管你什么arm还是其他什么，说白了就是5.4和5.10内核和没修改之前是一样的逻辑，5.15除外是内嵌了

Hzend, 2022-06-10:
@WYC-2020 抱歉我没说清楚。
我现在编译arm的时候，报错：Package kmod-wireguard is missing dependencies for the following libraries: libblake2s.ko
按照上面的方法，修改package/kernel/linux/modules/crypto.mk，增加了DEPENDS:=@(LINUX_5_4||LINUX_5_10) ，编译仍然报错。
你提到的这个linux.default.install，arm编译我看是没有的，文件名包含*default*只有一个.default.install。
我也正在尝试修改这个文件，重新编译试一下

Hzend, 2022-06-10:
试过了，在.default.install增加kmod-wireguard，还是相同的报错

WYC-2020, 2022-06-10:
> @WYC-2020 抱歉我没说清楚。 我现在编译arm的时候，报错：Package kmod-wireguard is missing dependencies for the following libraries: libblake2s.ko [...]

自己修改不对，要撤回部分修改，在按我这个修改才可以，把那个lt5.9 全部去除 在加上我说的这个，然后再make -j4 V=s

commit ee7d9cff (2022-06-07): Revert "kernel: bump 5.15 to 5.15.44 (#9457)"

This reverts commit 91b3516, fixes: #9483