coolsnowwolf/lede #9485: x86默认配置编译会报ddns相关依赖的错误

aakkll, 2022-06-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：尝试默认配置编译报错（主要还是想着加入sfe会怎么样的）

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
Configuring ddns-scripts_freedns_42_pl.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency bind-client for ddns-scripts_nsupdate
 * pkg_hash_fetch_best_installation_candidate: Packages for ddns-scripts_nsupdate found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package ddns-scripts_nsupdate.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/leung/kk151/515/lede'
make[1]: *** [package/Makefile:111: /home/leung/kk151/515/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/leung/kk151/515/lede'
make: *** [/home/leung/kk151/515/lede/include/toplevel.mk:230：world] 错误 2

Beginner-Go, 2022-06-05:
有这事？

zhangguanzhang, 2022-06-06:
第一个错误是无法找到依赖`bind-client`，第二个报错说是架构排斥，第一个问题你看下你的`.config`里的`CONFIG_PACKAGE_bind-client` 有没有
第二个问题你的target是什么设备，我看 Makefile里是全架构的：
```
cat feeds/packages/net/ddns-scripts/Makefile
...
PKGARCH:=all
```

coolsnowwolf, 2022-06-06:
rm -rf bin