coolsnowwolf/lede #9493: 云编译R2S最新固件时出现警告提示:luci-app-ssr-plus缺少依赖luci-lib-ipkg

felixxiao99, 2022-06-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译R2S最新固件时出现警告提示:luci-app-ssr-plus缺少依赖luci-lib-ipkg

### (2) 路由器型号和固件版本

A：R2S ARM架构

### (3) 详细日志

A：
[image]

WYC-2020, 2022-06-07:
改成luci-base,l大的根本就没有分离出ipkg

Jejz168, 2022-06-07:
一个wireguard，一个ssrp都不要编译就好了

felixxiao99, 2022-06-07:
> 一个wireguard，一个ssrp都不要编译就好了

这两个插件我都需要啊

WYC-2020, 2022-06-07:
这个告警根本没影响，可以忽略有洁癖的按我说的那样改，wire我又说怎么修复 自己看帖子

Jejz168, 2022-06-07:
wireguard，在arm上面编译不过，ssrp要改sed -i 's/luci-lib-ipkg/luci-base/g'，，麻烦、也是过几天也要改了

lovehackintosh, 2022-06-07:
执行make menuconfig前在lede目录下执行
cd ./package/feeds/helloworld/luci-app-ssr-plus
rm Makefile
wget https://raw.githubusercontent.com/fw876/helloworld/b4ede27bc2665208b7c5b2901abab55ea780d7b3/luci-app-ssr-plus/Makefile
然后正常编译

ClayStan, 2022-06-07:
> 

如果不改的话会编译报错的
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency luci-lib-ipkg for luci-app-ssr-plus
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ssr-plus found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
 * 	luci-lib-ipkg
 * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/claystan/master/lede-master'
make[1]: *** [package/Makefile:111: /home/claystan/master/lede-master/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/claystan/master/lede-master'
make: *** [/home/claystan/master/lede-master/include/toplevel.mk:230：world] 错误 2

Jejz168, 2022-06-07:
sed -i 's/luci-lib-ipkg/luci-base/g' package/luci-app-ssr-plus/Makefile

ihuaer, 2022-06-07:
> wget https://raw.githubusercontent.com/fw876/helloworld/b4ede27bc2665208b7c5b2901abab55ea780d7b3/luci-app-ssr-plus/Makefile

可以用

jtwgo, 2022-06-07:
sed -i 's/luci-lib-ipkg/luci-base/g' package/feeds/helloworld/luci-app-ssr-plus/Makefile

jtwgo, 2022-06-07:
sed -i 's/luci-lib-ipkg/luci-base/g' package/feeds/helloworld/luci-app-ssr-plus/Makefile 执行后编译仍然失败
[image]

Jejz168, 2022-06-07:
最新还要大改。还是不编译编吧。

Jejz168, 2022-06-07:
1.luci-lib-ipkg 在新的 js luci 中已经不再默认包含，故必须写入依赖。我稍后会做一个 workaround。
2.新引入的 sagernet-core 与 v2ray xray core 都冲突。如果之前勾选过 *ray，请先反选。
3.sagernet-core 必须使用 Go 1.18+ 进行编译。

作者说的

lxwq, 2022-06-08:
> 1.luci-lib-ipkg 在新的 js luci 中已经不再默认包含，故必须写入依赖。我稍后会做一个 workaround。 2.新引入的 sagernet-core 与 v2ray xray core 都冲突。如果之前勾选过 *ray，请先反选。 3.sagernet-core 必须使用 Go 1.18+ 进行编译。
> 
> 作者说的
第2是什么意思?sagernet-core取代v2ray xray core吗?而且只能2选一?

WYC-2020, 2022-06-08:
> > 1.luci-lib-ipkg 在新的 js luci 中已经不再默认包含，故必须写入依赖。我稍后会做一个 workaround。 2.新引入的 sagernet-core 与 v2ray xray core 都冲突。如果之前勾选过 *ray，请先反选。 3.sagernet-core 必须使用 Go 1.18+ 进行编译。
> > 作者说的
> > 第2是什么意思?sagernet-core取代v2ray xray core吗?而且只能2选一?

那是在最后打包的时候会冲突，至于为啥冲突就是相同位置安装了相同的文件，所以只能二选一

Myoko, 2022-06-09:
果然是 sagernet-core 与 v2ray xray core  有冲突！ 取消 sagernet-core 编译就正常了

bootli, 2022-06-09:
> 执行make menuconfig前在lede目录下执行 cd ./package/feeds/helloworld/luci-app-ssr-plus rm Makefile wget [https://raw.githubusercontent.com/fw876/helloworld/b4ede27bc2665208b7c5b2901abab55ea780d7b3/luci-app-ssr- plus/Makefile](https://raw.githubusercontent.com/fw876/helloworld/b4ede27bc2665208b7c5b2901abab55ea780d7b3/luci-app-ssr-plus/Makefile) 然后正常编译

可以是可以编译    TROJAN协议的节点用不了

bootli, 2022-06-09:
> 原来是 sagernet-core 和 v2ray xray core 有冲突！取消 sagernet-core 编译就正常了

取消 sagernet-core 编译就正常了  TROJAN协议的节点也用不了

WYC-2020, 2022-06-10:
> > 原来是 sagernet-core 和 v2ray xray core 有冲突！取消 sagernet-core 编译就正常了
> 
> 取消 sagernet-core 编译就正常了 TROJAN协议的节点也用不了

TROJAN 你要选中编译TROJAN支持吧，不是有一个选项嘛

bootli, 2022-06-11:
> > > 原来是 sagernet-core 和 v2ray xray core 有冲突！取消 sagernet-core 编译就正常了
> > 
> > 
> > 取消 sagernet-core 编译就正常了 TROJAN协议的节点也用不了
> 
> TROJAN 你要编译 TROJAN 支持吧，不是有一个选项嘛

编译的 一样

Jejz168, 2022-06-11:
> > > > 原来是 sagernet-core 和 v2ray xray core 有冲突！取消 sagernet-core 编译就正常了
> > > 
> > > 
> > > 取消 sagernet-core 编译就正常了 TROJAN协议的节点也用不了
> > 
> > 
> > TROJAN 你要编译 TROJAN 支持吧，不是有一个选项嘛
> 
> 编译的 一样

只有ssrp和clash，完美没有问题

Jejz168, 2022-06-11:
> 可以是可以编译 TROJAN协议的节点用不了

[image]

bootli, 2022-06-11:
> 编译 TROJAN协议的节

你是怎么做到的?

bootli, 2022-06-11:
> [image]

你是怎么做到的

Jejz168, 2022-06-11:
> 你是怎么做到的

ssrp依赖全部选。 v2ray-core/ xray-core全部 is not set。就可以了

bootli, 2022-06-11:
> ssrp依赖全部选。 v2ray-core/ xray-core全部 is not set。就可以了

你单单编译ssrp吧 别的科学插件没编译吧 单单编译ssrp是没问题的 加多编译个passwall或bypass就不行了

Jejz168, 2022-06-11:
> 你单单编译ssrp吧 别的科学插件没编译吧 单单编译ssrp是没问题的 加多编译个passwall或bypass就不行了

不行尝试了，让那些支持了在编译

WYC-2020, 2022-06-11:
> 你单单编译ssrp吧 别的科学插件没编译吧 单单编译ssrp是没问题的 加多编译个passwall或bypass就不行了

这个我在其他的帖子又说怎么搞

WYC-2020, 2022-06-11:
要么取消helloword中的新增的那个包，要么取消其他的v2和xray,还有就是要么自己修改passwall改用使用新加的那个包，我是直接改pw使用新加那个包编译

iKenzo, 2022-06-11:
首先nano .config ，F6 搜索 xray-core,  把结尾的y改成n

以上

closed 2022-06-12 by AmadeusGhost