coolsnowwolf/lede #9500: 使用sagernet core替换v2ray xray的问题。

harleywong, 2022-06-08:
在使用sagernet core替换v2ray xray之后，因为固件里面没有了这两个核心，luci-app-v2ray-server缺少了核心，无法使用了。
[image]

WYC-2020, 2022-06-08:
等会看下

WYC-2020, 2022-06-08:
[image]
[image]
[image]
简单修复了一下其他没看，没时间，测试运行没问题，功能自测 @harleywong

WYC-2020, 2022-06-08:
几万年的老东西 居然还在用

Jason6111, 2022-06-08:
你点MANUALLY UPDATE试试 没看见版本号

harleywong, 2022-06-08:
> 几万年的老东西 居然还在用

感谢，刚刚试了下，可以了！
主要是ssp自带的服务端没有v2相关的配置，在家里路由上开个服务端，公司用个路由器开ssp可以连回家里访问局域网，也不耽误使用公司的局域网。比其他纯vpn软件要方便点。
[image]

harleywong, 2022-06-08:
> 你点MANUALLY UPDATE试试 没看见版本号

根据大佬修改编译后可以了。

closed 2022-06-09 by harleywong