coolsnowwolf/lede #9502: luci-app-ipsec-vpnd 不能和 luci-app-ipsec-server一起编译了嘛？记得以前可以哦

qqsir-dev, 2022-06-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Collected errors:
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
        But that file is already provided by package  * strongswan-ipsec
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/controller/ipsec-server.lua
        But that file is already provided by package  * luci-app-ipsec-server
 * opkg_install_cmd: Cannot install package luci-app-ipsec-vpnd.
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
        But that file is already provided by package  * strongswan-ipsec
 * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/user/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/controller/ipsec-server.lua
        But that file is already provided by package  * luci-app-ipsec-server
 * opkg_install_cmd: Cannot install package luci-i18n-ipsec-vpnd-zh-cn.

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

WYC-2020, 2022-06-09:
看他们内容是否一样，一样就用一个就行了不要两个都编译

qqsir-dev, 2022-06-09:
嗯，去掉其中一个就可以，不能两个一起

closed 2022-06-09 by qqsir-dev

MgCI2, 2023-05-09:
同样的报错，想问一下路径在哪