coolsnowwolf/lede #9507: 编译报错 很多依赖找不到

lisonglin980, 2022-06-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A： * opkg_install_cmd: Cannot install package iptables-mod-extra.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-iprange:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package iptables-mod-iprange.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-tproxy:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package iptables-mod-tproxy.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
 * pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3 for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 *      kmod-fs-ntfs3
 * opkg_install_cmd: Cannot install package automount.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dnsmasq-full:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package dnsmasq-full.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package ppp.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pppoe:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-openclash:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package luci-app-openclash.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package sagernet-core.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-diskman:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package luci-app-diskman.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-nlbwmon:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package luci-app-nlbwmon.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-turboacc:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 *      kernel (= 5.10.117-1-21649dc79c1adca205c022c5bcbfef40)
 * opkg_install_cmd: Cannot install package luci-proto-bonding.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/lee/lede'
make[1]: *** [package/Makefile:111: /home/lee/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lee/lede'
make: *** [/home/lee/lede/include/toplevel.mk:230: world] Error 2

lisonglin980, 2022-06-10:
是因为更改了编译内核的原因吗，但是昨天我编译5.10也没问题，今天更新了源码就报错了

WYC-2020, 2022-06-10:
删除build linux 内核文件夹 让其重新编译内核
passwall 没有采用 sagernet-core，自己取消passwall的v2ray-core， xray-core  ，同时ssr选中 sagernet-core，或者ssr取消 sagernet-core，主要是passwall 那边还没切，我自己的仓库到是改了，目前你就这样选就能编译通过，自己等passwall那边修改

lisonglin980, 2022-06-10:
您好，是删除\lede\build_dir\target-x86_64_musl\linux-x86_64文件夹吗

Jejz168, 2022-06-10:
> 删除build linux内核文件夹，让其重新编译内核 passwall 没有采用sagernet-core，自己取消passwall的v2ray-core，xray-core，同时ssr取消sagernet-core，或者ssr取消sagernet-core，还主要是passwall没切，我自己的仓库到了就是改了，目前修改这样选通过编译，自己等通过墙

make[3] -C feeds/packages/net/vlmcsd compile
    ERROR: package/feeds/packages/vlmcsd failed to build.

我这是什么问题？一天了，选了vlmcsd插件也这样。这次直接全部删除试试

CincaiLa, 2022-06-12:
我的问题也是和你一样。我昨天第一次编译的。我要安装kmod-usb-net-asix-ax88179可是会弹出这段错误。
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-libphy
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-libphy found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-mii
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-mii found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-nls-base
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-usb-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-usb-net
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.45-1-36366c23928f05b55c3ce67034f899fe) for kmod-usb-net-asix-ax88179
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-asix-ax88179 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-asix-ax88179.
```

closed 2022-10-31 by aiamadeus