coolsnowwolf/lede #9515: luci-app-netdata建议加上中文版

shu0802, 2022-06-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-netdata中文版加入
[image]

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

Jason6111, 2022-06-11:
自己找下有补丁跟用法的 我做过部分汉化
[image]

shu0802, 2022-06-11:
> 自己找下有补丁跟用法的 我做过部分汉化 [image]

https://github.com/sirpdboy/sirpdboy-package/tree/main/luci-app-netdata 这里有已经汉化好的，我没办法把它折腾到openwrt里面来。

suizhenmin, 2022-06-11:
> > 自己找下有补丁跟用法的 我做过部分汉化 [image]
> 
> https://github.com/sirpdboy/sirpdboy-package/tree/main/luci-app-netdata 这里有已经汉化好的，我没办法把它折腾到openwrt里面来。

这个汉化的很完整，而且也比网上繁体汉化的更准确，希望大神能整合进来

shu0802, 2022-06-12:
今天早上发现汉化成功了。
[image]
[image]

closed 2022-06-12 by shu0802