coolsnowwolf/lede #9519: 新三能获取到IP，但登录不了界面bad gateway，

yuzhe1129, 2022-06-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
刷新页面无限[   83.350283] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76

登录IP 显示 bad gateway

The process did not produce any response
编译了2次都是一样的情况，
### (2) 路由器型号和固件版本
新三 D2
A：

### (3) 详细日志

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Newifi D2
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: Winbond W25Q256 (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 2
Autoboot aborted due to button press.

Starting breed built-in shell

breed> Erase completed.
Upgrade completed.
Rebooting in 2 seconds ...
Resetting ...

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Newifi D2
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: Winbond W25Q256 (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-5.4.195
[... 632 lines cut ...]
[   69.900952] jffs2: warning: (3812) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.707550] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.716140] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.724767] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.795211] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.803724] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   78.812227] jffs2: warning: (4400) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.845153] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.853630] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.862091] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.929456] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.937920] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   81.946396] jffs2: warning: (4494) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.676662] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.685150] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.693619] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.760500] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.769010] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   82.777545] jffs2: warning: (4495) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.350283] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.358797] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.367271] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.434248] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.442723] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76
[   83.451167] jffs2: warning: (4496) do_jffs2_setxattr: jffs2_reserve_space()=-28, request=76

ayancn, 2022-06-12:
同样问题很久了

yuzhe1129, 2022-06-12:
> 同样问题很久了

你也是这个问题吗？，什么情况

AmadeusGhost, 2022-06-13:
你的 spi 出问题了

yuzhe1129, 2022-06-13:
> 你的 spi 出问题了

应该不是吧，那为什么我刷别人的固件又可以成功启动呢，我编译固件也是提示成功了的，我试下换闪存试试吧

yuzhe1129, 2022-06-16:
没人知道吗？，这是bug还是我的路由问题？

closed 2022-06-17 by yuzhe1129

dengxin6, 2022-06-23:
不保留配置升级

frehedhuang, 2022-07-11:
是不是只要开某个软件就会这样？比如翻墙软件？若是这样，SSH进去百度一下看cpu内存占用的命令，看看cpu内存是不是太高了。太高了就进不去后台。换其他人的源重新编