coolsnowwolf/lede #9542: socat和ssrplus冲突导致编译报错

barton0426, 2022-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错 socat和ssrplus

### (2) 路由器型号和固件版本

A：Phicomm K3 latest

### (3) 详细日志

A：
Configuring socat.
Collected errors:
Configuring luci-app-socat.
Configuring luci-i18n-socat-zh-cn.
Configuring procd.
Configuring resolveip.
Configuring kmod-tcp-bbr.
Configuring kmod-shortcut-fe.
Configuring kmod-shortcut-fe-cm.
Configuring luci-app-turboacc.
Configuring kmod-nf-nat6.
Configuring kmod-crypto-acompress.
Configuring luci-i18n-autoreboot-zh-cn.
Configuring dns2socks.
Configuring etherwake.
Configuring luci-i18n-turboacc-zh-cn.
Configuring vsftpd-alt.
Configuring luci-app-vsftpd.
Configuring luci-i18n-vsftpd-zh-cn.
Configuring libmbedtls12.
Configuring block-mount.
Configuring luci-app-ddns.
Configuring kmod-ipt-iprange.
Configuring iptables-mod-iprange.
Configuring miniupnpd.
Configuring dns2tcp.
Configuring kmod-ipt-conntrack-extra.
Configuring dnsmasq-full.
Configuring iptables-mod-tproxy.
Configuring microsocks.
Configuring tcping.
Configuring libcurl4.
Configuring curl.
Configuring shadowsocks-rust-sslocal.
Configuring luci-app-ssr-plus.
Configuring luci-newapi.
Configuring odhcp6c.
Configuring kmod-crypto-ghash.
Configuring unzip.
Configuring uci.
Configuring luci-proto-ipv6.
Configuring ip6tables.
Configuring kmod-ipt-nat6.
Configuring odhcpd-ipv6only.
Configuring ipv6helper.
Configuring dropbear.
Configuring oseama.
Configuring luci-compat.
Configuring mtd.
Configuring hostapd-common.
Configuring wpad.
Configuring wireless-regdb.
Configuring kmod-cfg80211.
Configuring kmod-brcmutil.
Configuring brcmfmac-firmware-usb.
Configuring kmod-brcmfmac.
Configuring luci-i18n-arpbind-zh-cn.
Configuring kmod-crypto-ccm.
Configuring nvram.
Configuring urandom-seed.
Configuring otrx.
Configuring kmod-crypto-arc4.
Configuring kmod-crypto-sha1.
Configuring kmod-crypto-ecb.
Configuring kmod-mppe.
Configuring ppp.
Configuring kmod-leds-gpio.
Configuring luci-i18n-ssr-plus-zh-cn.
Configuring kmod-gpio-button-hotplug.
Configuring logd.
Configuring luci-app-accesscontrol.
Configuring luci-i18n-accesscontrol-zh-cn.
Configuring kmod-crypto-cmac.
Configuring kmod-crypto-gcm.
Configuring kmod-mac80211.
Configuring luci-i18n-ddns-zh-cn.
Configuring kmod-nf-nathelper-extra.
Configuring libcares.
Configuring ca-certificates.
Configuring wol.
Configuring luci-app-wol.
Configuring luci-i18n-wol-zh-cn.
Configuring luci-app-upnp.
Configuring iptables-mod-conntrack-extra.
Configuring default-settings.
Configuring k3wifi.
Configuring luci-i18n-ttyd-zh-cn.
Configuring coreutils-nohup.
Configuring shadowsocks-libev-ss-server.
Configuring xray-core.
Configuring luci-app-passwall.
Configuring luci-i18n-passwall-zh-cn.
Configuring luci-i18n-base-zh-cn.
Configuring kmod-b43.
Configuring kmod-lib-lzo.
Configuring osafeloader.
Configuring luci-i18n-upnp-zh-cn.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for: 	xray-core * 
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for: 	xray-core * 
 * opkg_install_cmd: Cannot install package sagernet-core.
 * resolve_conffiles: Existing conffile /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/runner/work/lede/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:111: /home/runner/work/lede/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

EnnawYang, 2022-06-12:
您再看看是什么和什么冲突？

barton0426, 2022-06-12:
> 

写出歧义了 socat出现了冲突  ssrplus也出现了冲突 sagernet-core和xray-core

WYC-2020, 2022-06-13:
这个 隔壁不是有啥详细的解决方案 自己去看

zhiwen1987, 2022-06-14:
> 隔壁不是有啥详细的解决方案
有链接吗 我看看

KevinMX, 2022-06-14:
https://github.com/fw876/helloworld/issues/912
https://github.com/fw876/helloworld/issues/902
https://github.com/fw876/helloworld/issues/893

d975xbx, 2022-06-14:
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_SagerNet_Core is not set
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Xray=y

d975xbx, 2022-06-14:
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_SagerNet_Core is not set

zzy19868, 2022-06-14:
> > .config xray-core=n
> 
> 没用的

我确实是这样解决的，ssrp下级选项全选，然后改config就可以。

barton0426, 2022-06-14:
目前的解决方案       # CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_SagerNet_Core is not set
          # CONFIG_PACKAGE_sagernet-core is not set

closed 2022-06-14 by barton0426