coolsnowwolf/lede #9576: 4530r硬改32M闪存固件如何编译，路过大神麻烦指点

jhx80kg, 2022-06-15:
硬改了32M闪存，编译成功后没有输出，网上找了一大堆也看得云里雾里
劳烦路过的大佬指点一下都修改那些地方能输出32M的固件，万分感谢
32M闪存芯片型号25q256fvfg

之前按照网上编译16M成功，但是教程里32始终不行

jhx80kg, 2022-06-16:
> 引用TPlink 对应容量的机型就行了
翻阅一些配置文件似乎没有找到有32m的路由

WowRouter, 2022-06-16:
还真有人折腾16M  32M的固件。。。

lsq726, 2022-06-20:
> 还真有人折腾16M 32M的固件。。。

是啊正在找K2P的32M的编译方法，已经替换闪存，
因为客户端很少  3台手机，2台电视  1台电脑，，1智能音箱 ，几2个智能家电这路由除了在线看视频，几乎闲置状态 ，升级32不为提升性能，只为安装插件，因为没空间了相信折腾闪存都是如此

lsq726git, 2022-06-25:
我已经找到K2P的编译方法，你的设备应该也是相通的，正在编译没有验证你自己试试，祝你好运

sed -i '/spi-max-frequency/a\\t\tbroken-flash-reset;' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/<0xa0000 0xf60000>/<0xa0000 0x1fb0000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/15744k/32448k/g' target/linux/ramips/image/mt7621.mk

原文连接
https://www.bgegao.com/2020/11/1885.html

jhx80kg, 2022-06-25:
我依此借鉴试试，再次感谢朋友的热心帮助

coolsnowwolf, 2026-09-05:
硬改 32 MiB Flash 的问题是设备分区、bootloader 和镜像布局定制咨询，没有具体 DTS/分区表或编译错误，不能安全给出通用源码修改，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf