coolsnowwolf/lede #9588: CR6606 R22.6.16 在 LuCI 界面点重启，LAN无法连接。

iseeyou, 2022-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

LuCI 界面点重启，LAN无法连接，wifi可以连接。

## 2. 详细叙述

LuCI 界面点重启，LAN无法连接，wifi可以连接。断电重启可以正常启动。

### (2) 路由器型号和固件版本

CR6606 R22.6.16 内核版本 | 5.4.195

### (3) 详细日志

通过wifi连接上去获取的日志

kernel: [    0.781655] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
Thu Jun 16 09:47:58 2022 kern.info kernel: [    0.788774] NET: Registered protocol family 1
Thu Jun 16 09:47:58 2022 kern.info kernel: [    0.793103] PCI: CLS 0 bytes, default 32
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.030200] 4 CPUs re-calibrate udelay(lpj = 2924544)
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    1.035548] random: fast init done
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.037150] workingset: timestamp_bits=14 max_order=16 bucket_order=2
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.057741] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.063535] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.076431] mt7621_gpio 1e000600.gpio: registering 32 gpios
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.082356] mt7621_gpio 1e000600.gpio: registering 32 gpios
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.088168] mt7621_gpio 1e000600.gpio: registering 32 gpios
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.094711] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.102445] printk: console [ttyS0] disabled
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.106696] 1e000c00.uartlite: ttyS0 at MMIO 0x1e000c00 (irq = 19, base_baud = 3125000) is a 16550A
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.115703] printk: console [ttyS0] enabled
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.123962] printk: bootconsole [early0] disabled
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.135725] mt7621-nand 1e003000.nand: Using programmed access timing: 31c07388
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.143326] nand: device found, Manufacturer ID: 0xc8, Chip ID: 0xd1
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.149646] nand: ESMT PSU1GA30DT
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.152970] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.160531] mt7621-nand 1e003000.nand: ECC strength adjusted to 4 bits
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.167060] mt7621-nand 1e003000.nand: Using programmed access timing: 21005134
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.174373] mt7621-nand 1e003000.nand: Using programmed access timing: 21005134
Thu Jun 16 09:47:58 2022 kern.info kernel: [    1.181667] Scanning device for bad blocks
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.166343] 8 fixed-partitions partitions found on MTD device mt7621-nand
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.173117] Creating 8 MTD partitions on "mt7621-nand":
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.178324] 0x000000000000-0x000000080000 : "Bootloader"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.185114] 0x000000080000-0x0000000c0000 : "Nvram"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.191399] 0x0000000c0000-0x000000100000 : "Bdata"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.197551] 0x000000100000-0x000000180000 : "Factory"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.204006] 0x000000180000-0x0000001c0000 : "crash"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.210186] 0x0000001c0000-0x000000200000 : "crash_log"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.216821] 0x000000200000-0x000000600000 : "kernel"
Thu Jun 16 09:47:58 2022 kern.notice kernel: [    3.223157] 0x000000600000-0x000007f80000 : "ubi"
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.270969] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.281404] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 22
Thu Jun 16 09:47:58 2022 kern.err kernel: [    3.292273] mt7621-pci 1e140000.pcie: Parsing DT failed
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.299994] NET: Registered protocol family 10
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.305910] Segment Routing with IPv6
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.309660] NET: Registered protocol family 17
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.314440] 8021q: 802.1Q VLAN Support v1.8
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.322155] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.337918] mt7530 mdio-bus:1f lan1 (uninitialized): PHY [dsa-0.0:00] driver [Generic PHY]
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.347800] mt7530 mdio-bus:1f lan2 (uninitialized): PHY [dsa-0.0:01] driver [Generic PHY]
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.357616] mt7530 mdio-bus:1f lan3 (uninitialized): PHY [dsa-0.0:02] driver [Generic PHY]
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.367635] mt7530 mdio-bus:1f wan (uninitialized): PHY [dsa-0.0:04] driver [Generic PHY]
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.377416] mt7530 mdio-bus:1f: configuring for fixed/rgmii link mode
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.388647] DSA: tree 0 setup
Thu Jun 16 09:47:58 2022 kern.err kernel: [    3.392012] rt2880-pinmux pinctrl: pcie is already enabled
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.397629] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
Thu Jun 16 09:47:58 2022 kern.info kernel: [    3.405224] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
Thu Jun 16 09:47:58 2022 kern.err kernel: [    3.412689] mt7621-pci 1e140000.pcie: failed to parse bus ranges property: -22
[... 337 lines cut ...]
Thu Jun 16 09:48:31 2022 daemon.notice netifd: wan (3281): uci: Entry not found
Thu Jun 16 09:48:31 2022 daemon.notice netifd: wan (3281): sh: out of range
Thu Jun 16 09:48:31 2022 daemon.info pppd[3329]: Plugin pppoe.so loaded.
Thu Jun 16 09:48:31 2022 daemon.info pppd[3329]: PPPoE plugin from pppd 2.4.9
Thu Jun 16 09:48:31 2022 daemon.notice pppd[3329]: pppd 2.4.9 started by root, uid 0
Thu Jun 16 09:48:32 2022 daemon.info hostapd: wlan1: STA 7e:41:1c:24:a8:f9 IEEE 802.11: authenticated
Thu Jun 16 09:48:32 2022 daemon.info hostapd: wlan1: STA 7e:41:1c:24:a8:f9 IEEE 802.11: associated (aid 2)
Thu Jun 16 09:48:33 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 7e:41:1c:24:a8:f9
Thu Jun 16 09:48:33 2022 daemon.info hostapd: wlan1: STA 7e:41:1c:24:a8:f9 RADIUS: starting accounting session 16C658FC90A6A906
Thu Jun 16 09:48:33 2022 daemon.info hostapd: wlan1: STA 7e:41:1c:24:a8:f9 WPA: pairwise key handshake completed (RSN)
Thu Jun 16 09:48:33 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 7e:41:1c:24:a8:f9
Thu Jun 16 09:48:37 2022 daemon.warn odhcpd[1819]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Jun 16 09:48:40 2022 daemon.notice procd: /etc/rc.d/S96led: setting up led internet
Thu Jun 16 09:48:40 2022 daemon.notice procd: /etc/rc.d/S96led: setting up led system
Thu Jun 16 09:48:41 2022 daemon.info procd: - init complete -
Thu Jun 16 09:48:45 2022 daemon.info hostapd: wlan0: STA 7c:c2:94:2f:ea:14 IEEE 802.11: authenticated
Thu Jun 16 09:48:45 2022 daemon.info hostapd: wlan0: STA 7c:c2:94:2f:ea:14 IEEE 802.11: associated (aid 4)
Thu Jun 16 09:48:45 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 7c:c2:94:2f:ea:14
Thu Jun 16 09:48:45 2022 daemon.info hostapd: wlan0: STA 7c:c2:94:2f:ea:14 RADIUS: starting accounting session 8030DE1A18E3DADB
Thu Jun 16 09:48:45 2022 daemon.info hostapd: wlan0: STA 7c:c2:94:2f:ea:14 WPA: pairwise key handshake completed (RSN)
Thu Jun 16 09:48:45 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 7c:c2:94:2f:ea:14
Thu Jun 16 09:48:46 2022 daemon.warn odhcpd[1819]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Jun 16 09:48:47 2022 daemon.warn pppd[3329]: Timeout waiting for PADO packets
Thu Jun 16 09:48:47 2022 daemon.err pppd[3329]: Unable to complete PPPoE Discovery
Thu Jun 16 09:48:47 2022 daemon.info pppd[3329]: Exit.

ayancn, 2022-06-18:
同样问题很久了

ayancn, 2022-06-19:
ip显示6x.x.x.x LEAN R2.2.5.5之后最新多个版本都有这个问题重启后出现,断电重启才能进, 
immortalwrt master 和 18.06K5.4版本也有这个问题是直接编译好了电脑就不能进(显示拒绝连接) 
然后都是wifi手机可以进 
目前就是官方版本 和 Lienol 版本正常 , 希望能修复 谢谢
路由器是CR6608 (CR660X)

closed 2022-06-27 by iseeyou