coolsnowwolf/lede #9597: kern.warn kernel: [36777.941983] test 512

drw2005, 2022-06-17:
### (2) 路由器型号和固件版本
R2S   内核版本： 5.10.117
-- | --

### (3) 系统日志
Fri Jun 17 07:59:05 2022 kern.warn kernel: [40170.968844] test 512
Fri Jun 17 07:59:05 2022 kern.warn kernel: [40170.973741] test 512
Fri Jun 17 07:59:05 2022 kern.warn kernel: [40170.977104] test 512
Fri Jun 17 07:59:05 2022 kern.warn kernel: [40171.031947] test 512
Fri Jun 17 07:59:14 2022 kern.warn kernel: [40171.032971] test 512
Fri Jun 17 07:59:18 2022 kern.warn kernel: [40180.073304] test 512
Fri Jun 17 07:59:20 2022 kern.warn kernel: [40184.230610] test 512
Fri Jun 17 07:59:24 2022 kern.warn kernel: [40186.642059] test 512
Fri Jun 17 07:59:25 2022 kern.warn kernel: [40190.212397] test 512
Fri Jun 17 07:59:40 2022 kern.warn kernel: [40191.618401] test 512
Fri Jun 17 07:59:59 2022 kern.warn kernel: [40206.614145] test 512
Fri Jun 17 07:59:59 2022 kern.warn kernel: [40225.040760] test 512
Fri Jun 17 07:59:59 2022 kern.warn kernel: [40225.041298] test 512
Fri Jun 17 08:00:09 2022 kern.warn kernel: [40225.041727] test 512
Fri Jun 17 08:00:14 2022 kern.warn kernel: [40235.390615] test 512
Fri Jun 17 08:00:20 2022 kern.warn kernel: [40240.234353] test 512
Fri Jun 17 08:00:40 2022 kern.warn kernel: [40246.646169] test 512
Fri Jun 17 08:00:41 2022 kern.warn kernel: [40265.849305] test 512
Fri Jun 17 08:00:44 2022 kern.warn kernel: [40267.004772] test 512
Fri Jun 17 08:00:49 2022 kern.warn kernel: [40270.003515] test 512
Fri Jun 17 08:00:49 2022 kern.warn kernel: [40275.265969] test 512
Fri Jun 17 08:00:54 2022 kern.warn kernel: [40275.630260] test 512
Fri Jun 17 08:00:56 2022 kern.warn kernel: [40280.693184] test 512
Fri Jun 17 08:00:57 2022 kern.warn kernel: [40282.507894] test 512
Fri Jun 17 08:01:17 2022 kern.warn kernel: [40282.874627] test 512
Fri Jun 17 08:01:20 2022 kern.warn kernel: [40302.950984] test 512
Fri Jun 17 08:01:29 2022 kern.warn kernel: [40306.669691] test 512
Fri Jun 17 08:01:29 2022 kern.warn kernel: [40314.816888] test 512
Fri Jun 17 08:01:34 2022 kern.warn kernel: [40314.831527] test 512
Fri Jun 17 08:01:49 2022 kern.warn kernel: [40319.934804] test 512
Fri Jun 17 08:01:54 2022 kern.warn kernel: [40335.129465] test 512
Fri Jun 17 08:01:56 2022 kern.warn kernel: [40340.239904] test 512

内核日志：
[   18.788459] r8152 4-1:1.0 eth1: carrier on
[   19.765336] pppoe-wan: renamed from ppp0
[   21.845565] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.846428] br-lan: port 1(eth0) entered blocking state
[   21.846902] br-lan: port 1(eth0) entered forwarding state
[   24.050439] test 512
[   24.201893] test 512
[   24.203378] test 512
[   24.214549] test 512
[   24.214917] test 512
[   24.394477] test 512
[   24.434298] test 512
[   24.484833] test 512
[   24.499913] test 512
[   24.501377] test 512
[   24.504133] test 512
[   24.622470] test 512
[   24.638948] test 512
[   24.967397] test 512
[   25.007752] test 512
[   25.139956] r8152 4-1:1.0 eth1: carrier on
[   25.219882] r8152 4-1:1.0 eth1: carrier on
[   25.252028] r8152 4-1:1.0 eth1: carrier on
[   25.364311] r8152 4-1:1.0 eth1: carrier on
[   25.412265] r8152 4-1:1.0 eth1: carrier on
[   25.460078] r8152 4-1:1.0 eth1: carrier on
[   25.572190] r8152 4-1:1.0 eth1: carrier on
[   25.620237] r8152 4-1:1.0 eth1: carrier on
[   25.674877] r8152 4-1:1.0 eth1: carrier on
[   25.716490] r8152 4-1:1.0 eth1: carrier on
[   26.085128] pppoe-wan: renamed from ppp0
[   29.071533] test 512
[   29.479931] test 512
[   29.595319] test 512
[   29.822136] test 512
[   30.077353] test 512
[   30.598982] test 512
[   30.605273] test 512
[   30.670291] test 512
[   30.677815] test 512
[   30.678994] test 512
[   30.998874] test 512
[   31.006205] test 512
[   31.285535] test 512
[   31.511344] test 512
[   31.512156] test 512
[   31.605038] test 512
[   31.625245] test 512
[   31.939142] test 512
[   32.087777] test 512
[   32.437676] test 512
[   32.702437] test 512

系统一直输出上面的内容，请教大佬是何原因？

modelsun, 2022-06-17:
同样遇到了，记得当时2.4G 好像闪断了一下

Benjamin23333, 2022-06-17:
对，就是这个，我妈和我姐刚才微信视频突然显示网络异常然后掉线了，然后我看系统日志里全是这个，铺满了一页

coolsnowwolf, 2022-06-17:
具体型号？

Benjamin23333, 2022-06-17:
> 具体型号？

我和楼主一样是r2s，只不过我的内核版本是5.10.120

coolsnowwolf, 2022-06-17:
这个是因为你的  SFE 和 Flow 一起开启了，互相争夺冲突了，最好恢复出厂重新配置下

coolsnowwolf, 2022-06-17:
/etc/config/firewall

config defaults
        option syn_flood '1'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option fullcone '1'
        option flow_offloading '1'
        option flow_offloading_hw '1'

这个配置两个 flow 改为 0， 然后重启系统，即可解决

Benjamin23333, 2022-06-17:
> 这个是因为你的 SFE 和 Flow 一起开启了，互相争夺冲突了，最好恢复出厂重新配置下

可是我用的固件没有编译sfe啊😂，我用的这个https://github.com/haiibo/OpenWrt

Benjamin23333, 2022-06-17:
> /etc/config/firewall
> 
> config defaults option syn_flood '1' option input 'ACCEPT' option output 'ACCEPT' option forward 'REJECT' option fullcone '1' option flow_offloading '1' option flow_offloading_hw '1'
> 
> 这个配置两个 flow 改为 0， 然后重启系统，即可解决

我明天试一下，谢谢lean大

modelsun, 2022-06-17:
> 具体型号？

我的HC6962 ， mt7621 ，版本好像是5.10.117

Benjamin23333, 2022-06-18:
> 

刚才试了一下，改完之后重启完了之后，看下系统日志，还是一堆
Sat Jun 18 11:58:26 2022 kern.warn kernel: [  141.918811] test 512
Sat Jun 18 11:58:32 2022 kern.warn kernel: [  152.073971] test 512
Sat Jun 18 11:58:33 2022 kern.warn kernel: [  157.900031] test 512
Sat Jun 18 11:58:36 2022 kern.warn kernel: [  158.841041] test 512
Sat Jun 18 11:58:36 2022 kern.warn kernel: [  162.266015] test 512
Sat Jun 18 11:58:36 2022 kern.warn kernel: [  162.717580] test 512
Sat Jun 18 11:58:42 2022 kern.warn kernel: [  162.718226] test 512
Sat Jun 18 11:58:46 2022 kern.warn kernel: [  167.832819] test 512
Sat Jun 18 11:58:46 2022 kern.warn kernel: [  172.434003] test 512
Sat Jun 18 11:58:56 2022 kern.warn kernel: [  172.565326] test 512
这样的。
又去/etc/config/firewall看了下，改过了部分恢复原状了😂

Benjamin23333, 2022-06-18:
> 这个是因为你的 SFE 和 Flow 一起开启了，互相争夺冲突了，最好恢复出厂重新配置下
我也重新恢复出厂了，一样会出现
Sat Jun 18 13:39:25 2022 kern.warn kernel: [ 6210.492331] test 512
检查了我用这个固件确实没有sfe，只有shortcut-fe
https://github.com/haiibo/OpenWrt/blob/77251927282e7792f72c8f745e68031395da2b1f/nanopi/r2s.config#L253

drw2005, 2022-06-18:
> 我用这个固件确实没有shortcut-fe，只有Flow Offload

> 这个是因为你的 SFE 和 Flow 一起开启了，互相争夺冲突了，最好恢复出厂重新配置下

我的固件也没有shortcut-fe，只有Flow Offload

CkovMk, 2022-06-18:
x86_64固件使用5.10.117内核也有同样的问题，且路由功能正常，但CPU Load较高。
同样没有编译SFE功能。前面修改两个flow的设置也没有解决问题。

yin666666, 2022-06-29:
AX6S     5.10.125内核，也遇到test 512，没有shortcut-fe，只有Flow Offload，路由其它功能暂未发现问题

uosX, 2022-09-20:
一样有这问题，设置为0了，依然出现这问题，sfe，或者turbo acc没法用了

xxp2277, 2022-11-08:
这个bug还没有解决么，关于turboacc加速的，只要打开就会不断出现test 512的错误。
wndr3800
OpenWrt  R22.11.11
内核版本 | 5.10.153

drw2005, 2022-11-09:
> 这个bug还没有解决么，关于turboacc加速的，只要打开就会不断出现test 512的错误。 wndr3800 OpenWrt R22.11.11 内核版本 | 5.10.153

我现在用的内核5.15.70，OpenWrt R22.9.1，没有出现这个情况了。

QYKing233, 2022-11-12:
插眼

xbins7210, 2022-12-14:
插眼

Ais-Danny, 2024-01-06:
插眼