coolsnowwolf/lede #9598: x86/64编译make file多种报错，不提示插件冲突，没出现插件相关字段

a2945517217, 2022-06-17:
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 147551.03 Kbytes (144.09 Mbytes)
	28.44% of uncompressed filesystem size (518845.99 Kbytes)
Inode table size 43924 bytes (42.89 Kbytes)
	21.08% of uncompressed inode table size (208326 bytes)
Directory table size 61434 bytes (59.99 Kbytes)
	41.41% of uncompressed directory table size (148364 bytes)
Number of duplicate files found 655
Number of inodes 6254
Number of files 5309
Number of fragments 113
Number of symbolic links  537
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 407
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
rm -f /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
cp /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/bzImage /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
cp /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin /home/ubuntu/op_lean/lede/bin/targets/x86/64/openwrt-x86-64-generic-kernel.bin
[ -f /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin -a -f /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs ]
dd if=/home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs >> /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img
295102+1 records in
295102+1 records out
151092258 bytes (151 MB, 144 MiB) copied, 0.689631 s, 219 MB/s
dd if=/home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img of=/home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.new bs=3221225472 conv=sync
make[5]: *** [Makefile:164: /home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img] Killed
make[5]: *** Deleting file '/home/ubuntu/op_lean/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img'
make[5]: Leaving directory '/home/ubuntu/op_lean/lede/target/linux/x86/image'
make[4]: *** [Makefile:27: install] Error 2
make[4]: Leaving directory '/home/ubuntu/op_lean/lede/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ubuntu/op_lean/lede/target/linux'
time: target/linux/install#395.43#7.81#240.73
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ubuntu/op_lean/lede'
make[1]: *** [target/Makefile:19: /home/ubuntu/op_lean/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ubuntu/op_lean/lede'
make: *** [/home/ubuntu/op_lean/lede/include/toplevel.mk:230: world] Error 2

a393310872, 2022-06-17:
如果是多线程编译的.切换单线程继续.试试.现成开多了.有时也会报错

a2945517217, 2022-06-17:
> 

我看到make file failed ，我怀疑是4G内存不够，内存不足，然后我尝试了换上自己的16G的电脑装了个ubuntu进行编译，就成功了，可能是内存不够的问题。
失败的编译环境是腾讯云的香港地区轻量服务器，2C4G,ubuntu 20.04lts。

Gokou-Ruri, 2024-07-24:
先开单线程跑一遍确定不是固件或依赖等有问题再来考虑是具体什么问题，而且报错只能看单线程下的输出，你贴个多线程编译的报错输出毫无意义。

coolsnowwolf, 2026-09-05:
原始日志明确是 4 GiB 云主机在生成 x86 镜像时被 OOM killer 终止（`make[5]: *** ... Killed`），触发点是旧版 x86 image Makefile 用 `dd ... bs=3221225472 conv=sync` 生成大文件，并非插件冲突或依赖缺失。当前 master 的 x86 image 流程已改为标准 `append-rootfs | pad-to $(ROOTFS_PARTSIZE)`，不再使用该 3 GiB 单块内存分配方式；评论也确认增加内存后成功。现行源码已覆盖此问题，关闭为 completed。

closed 2026-09-05 by coolsnowwolf