coolsnowwolf/lede #9603: 树莓派3b+编译失败， ubuntu lts 22

matianhe3, 2022-06-18:
原版编译失败了三次都是 wolfssl编译失败，取消了所有插件测试还是同样错误。
[image]

hcym, 2022-06-18:
4B一直可以，多半是细节问题吧，

matianhe3, 2022-06-18:
换了debian11 还是一样。
```
4.0_musl_eabi/usr/lib -L/home/debian/Downloads/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro -flto  -o src/libwolfssl.la -rpath /usr/lib                                           wolfcrypt/src/src_libwolfssl_la-hmac.lo  wolfcrypt/src/src_libwolfssl_la-hash.lo wolfcrypt/src/src_libwolfssl_la-cpuid.lo wolfcrypt/src/src_libwolfssl_la-kdf.lo wolfcrypt/src/src_libwolfssl_la-random.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-sha256.lo        wolfcrypt/src/src_libwolfssl_la-rsa.lo          wolfcrypt/src/src_libwolfssl_la-aes.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-aes.lo  wolfcrypt/src/src_libwolfssl_la-cmac.lo wolfcrypt/src/src_libwolfssl_la-des3.lo wolfcrypt/src/src_libwolfssl_la-sha.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-sha512.lo  wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-sha512-asm.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-32-sha512-asm.lo       wolfcrypt/src/src_libwolfssl_la-logging.lo wolfcrypt/src/src_libwolfssl_la-wc_port.lo wolfcrypt/src/src_libwolfssl_la-error.lo wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo wolfcrypt/src/src_libwolfssl_la-signature.lo wolfcrypt/src/src_libwolfssl_la-wolfmath.lo wolfcrypt/src/src_libwolfssl_la-memory.lo wolfcrypt/src/src_libwolfssl_la-dh.lo wolfcrypt/src/src_libwolfssl_la-asn.lo wolfcrypt/src/src_libwolfssl_la-coding.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-poly1305.lo wolfcrypt/src/src_libwolfssl_la-poly1305.lo  wolfcrypt/src/src_libwolfssl_la-arc4.lo wolfcrypt/src/src_libwolfssl_la-md4.lo wolfcrypt/src/src_libwolfssl_la-md5.lo wolfcrypt/src/src_libwolfssl_la-pwdbased.lo wolfcrypt/src/src_libwolfssl_la-pkcs12.lo wolfcrypt/src/src_libwolfssl_la-dsa.lo       wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-chacha.lo   wolfcrypt/src/src_libwolfssl_la-chacha20_poly1305.lo  wolfcrypt/src/src_libwolfssl_la-tfm.lo  wolfcrypt/src/src_libwolfssl_la-ecc.lo   wolfcrypt/src/src_libwolfssl_la-curve25519.lo     wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-32-curve25519.lo wolfcrypt/src/port/arm/src_libwolfssl_la-armv8-curve25519.lo                  src/src_libwolfssl_la-internal.lo src/src_libwolfssl_la-wolfio.lo src/src_libwolfssl_la-keys.lo src/src_libwolfssl_la-ssl.lo src/src_libwolfssl_la-tls.lo src/src_libwolfssl_la-tls13.lo src/src_libwolfssl_la-ocsp.lo src/src_libwolfssl_la-crl.lo              -lm   -lpthread
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc -shared  -fPIC -DPIC  wolfcrypt/src/.libs/src_libwolfssl_la-hmac.o wolfcrypt/src/.libs/src_libwolfssl_la-hash.o wolfcrypt/src/.libs/src_libwolfssl_la-cpuid.o wolfcrypt/src/.libs/src_libwolfssl_la-kdf.o wolfcrypt/src/.libs/src_libwolfssl_la-random.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-sha256.o wolfcrypt/src/.libs/src_libwolfssl_la-rsa.o wolfcrypt/src/.libs/src_libwolfssl_la-aes.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-aes.o wolfcrypt/src/.libs/src_libwolfssl_la-cmac.o wolfcrypt/src/.libs/src_libwolfssl_la-des3.o wolfcrypt/src/.libs/src_libwolfssl_la-sha.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-sha512.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-sha512-asm.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-32-sha512-asm.o wolfcrypt/src/.libs/src_libwolfssl_la-logging.o wolfcrypt/src/.libs/src_libwolfssl_la-wc_port.o wolfcrypt/src/.libs/src_libwolfssl_la-error.o wolfcrypt/src/.libs/src_libwolfssl_la-wc_encrypt.o wolfcrypt/src/.libs/src_libwolfssl_la-signature.o wolfcrypt/src/.libs/src_libwolfssl_la-wolfmath.o wolfcrypt/src/.libs/src_libwolfssl_la-memory.o wolfcrypt/src/.libs/src_libwolfssl_la-dh.o wolfcrypt/src/.libs/src_libwolfssl_la-asn.o wolfcrypt/src/.libs/src_libwolfssl_la-coding.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-poly1305.o wolfcrypt/src/.libs/src_libwolfssl_la-poly1305.o wolfcrypt/src/.libs/src_libwolfssl_la-arc4.o wolfcrypt/src/.libs/src_libwolfssl_la-md4.o wolfcrypt/src/.libs/src_libwolfssl_la-md5.o wolfcrypt/src/.libs/src_libwolfssl_la-pwdbased.o wolfcrypt/src/.libs/src_libwolfssl_la-pkcs12.o wolfcrypt/src/.libs/src_libwolfssl_la-dsa.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-chacha.o wolfcrypt/src/.libs/src_libwolfssl_la-chacha20_poly1305.o wolfcrypt/src/.libs/src_libwolfssl_la-tfm.o wolfcrypt/src/.libs/src_libwolfssl_la-ecc.o wolfcrypt/src/.libs/src_libwolfssl_la-curve25519.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-32-curve25519.o wolfcrypt/src/port/arm/.libs/src_libwolfssl_la-armv8-curve25519.o src/.libs/src_libwolfssl_la-internal.o src/.libs/src_libwolfssl_la-wolfio.o src/.libs/src_libwolfssl_la-keys.o src/.libs/src_libwolfssl_la-ssl.o src/.libs/src_libwolfssl_la-tls.o src/.libs/src_libwolfssl_la-tls13.o src/.libs/src_libwolfssl_la-ocsp.o src/.libs/src_libwolfssl_la-crl.o   -L/home/debian/Downloads/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/debian/Downloads/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -lm -lpthread  -pthread -O2 -Os -mfloat-abi=hard -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -flto -flto   -pthread -Wl,-soname -Wl,libwolfssl.so.5.3.0.31258522 -o src/.libs/libwolfssl.so.5.3.0.31258522
/home/debian/Downloads/lede/tmp/ccTKx4TI.s: Assembler messages:
/home/debian/Downloads/lede/tmp/ccTKx4TI.s:2595: Error: first transfer register must be even -- `ldrd r11,r10,[r14,#4*14]'
lto-wrapper: fatal error: arm-openwrt-linux-muslgnueabi-gcc returned 1 exit status
compilation terminated.
[... cut ...]
time: package/libs/wolfssl/compile#28.56#4.77#32.95
    ERROR: package/libs/wolfssl failed to build.
make[2]: *** [package/Makefile:116: package/libs/wolfssl/compile] Error 1
make[2]: Leaving directory '/home/debian/Downloads/lede'
make[1]: *** [package/Makefile:110: /home/debian/Downloads/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/Downloads/lede'
make: *** [/home/debian/Downloads/lede/include/toplevel.mk:230: world] Error 2
```

EnnawYang, 2022-06-18:
编译64位的试试？

hcym, 2022-06-18:
操作系统无所谓了，d11，u20，u22都编译过，说不清楚，只能自己慢慢排查，最关键是要看懂出错信息，我是怕看字母，😂

Neverends4, 2022-06-26:
64位3B+刚编译没问题

closed 2022-06-26 by matianhe3