coolsnowwolf/lede #9605: R2s编译错误

jayshen594, 2022-06-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译错误

### (2) 路由器型号和固件版本

A：R2s

### (3) 详细日志

cp /home/jay/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux /home/jay/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/friendlyarm_nanopi-r2s-kernel.bin
Pseudo file "/dev" exists in source filesystem "/home/jay/lede/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 1743 blocks, out of space?
make[5]: *** [/home/jay/lede/include/image.mk:336: /home/jay/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[5]: *** Waiting for unfinished jobs....
[=============================================================================================================/] 6517/6517 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 130116.29 Kbytes (127.07 Mbytes)
	25.69% of uncompressed filesystem size (506559.53 Kbytes)
Inode table size 50288 bytes (49.11 Kbytes)
	21.01% of uncompressed inode table size (239351 bytes)
Directory table size 65722 bytes (64.18 Kbytes)
	39.57% of uncompressed directory table size (166094 bytes)
Number of duplicate files found 706
Number of inodes 7207
Number of files 6158
Number of fragments 77
Number of symbolic links  571
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 477
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
make[5]: Leaving directory '/home/jay/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/jay/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/jay/lede/target/linux'
time: target/linux/install#268.53#5.32#46.13
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/jay/lede'
make[1]: *** [target/Makefile:19: /home/jay/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/jay/lede'
make: *** [/home/jay/lede/include/toplevel.mk:230：world] 错误 2

zhangguanzhang, 2022-06-18:
```
CONFIG_TARGET_ROOTFS_PARTSIZE=645
```
CONFIG_TARGET_ROOTFS_PARTSIZE 太小了

jayshen594, 2022-06-19:
> `error: ext4_allocate_best_fit_partial: failed to allocate 1743 blocks, out of space`
> 你看看你的`CONFIG_TARGET_ROOTFS_PARTSIZE` 是不是太小了

是的 解决了

zhangguanzhang, 2022-06-19:
那就把issue关闭吧

closed 2022-06-19 by jayshen594