coolsnowwolf/lede #9610: 极路由B70使用新驱动编译后WiFi连接设备网络不正常现象

mcgivefenngg, 2022-06-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
极路由B70使用新版mt7621 HWNAT驱动编译刷入后，使用WiFi连接的设备网络不正常，表现为无法完整开启网页，聊天软件无法连接例如微信,telegram。但网线连接的设备网络完全正常，所有功能可用。

### (2) 路由器型号和固件版本
极路由B70 OpenWrt R22.6.16 / LuCI Master (git-22.168.62806-d59f9ce) 内核5.4.199

### (3) 详细日志

<details><summary>kernel</summary>
<p>
[    0.000000] Linux version 5.4.199 (runner@fv-az212-476) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4589-f01d99d61)) #0 SMP Fri Jun 17 16:34:04 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is HiWiFi HC5962
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27024 r8192 d22128 u57344
[    0.000000] pcpu-alloc: s27024 r8192 d22128 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00006200
[    0.000000] Readback ErrCtl register=00006200
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250084K/262144K available (6169K kernel code, 273K rwdata, 1300K rodata, 1268K init, 236K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007801] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073977] pid_max: default: 32768 minimum: 301
[    0.078768] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085968] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096493] rcu: Hierarchical SRCU implementation.
[    0.101831] smp: Bringing up secondary CPUs ...
[    9.744613] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.744624] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.744636] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.744741] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.166624] Synchronize counters for CPU 1: done.
[    0.008833] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 300 lines cut ...]
[   34.829396] extif_set_dev(wlan0)
[   34.846299] br-lan: port 5(wlan0) entered blocking state
[   34.851772] br-lan: port 5(wlan0) entered disabled state
[   34.858019] device wlan0 entered promiscuous mode
[   34.863305] br-lan: port 5(wlan0) entered blocking state
[   34.868713] br-lan: port 5(wlan0) entered forwarding state
[   34.874762] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   35.560455] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[   35.567976] br-lan: port 3(lan3) entered blocking state
[   35.573262] br-lan: port 3(lan3) entered forwarding state
[   35.640152] mt7530 mdio-bus:1f wan: Link is Down
[   35.804955] br-lan: port 5(wlan0) entered disabled state
[   35.836515] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   35.843596] br-lan: port 4(wlan1) entered blocking state
[   35.849032] br-lan: port 4(wlan1) entered forwarding state
[   36.080536] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   36.087566] 8021q: adding VLAN 0 to HW filter on device wan
[   36.306067] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   36.312942] br-lan: port 5(wlan0) entered blocking state
[   36.318373] br-lan: port 5(wlan0) entered forwarding state
[   39.449888] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   39.457167] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   40.102538] pppoe-wan: renamed from ppp0
</p>
</details>

asiaon123, 2022-06-19:
同样不正常

britneyjeanspears, 2022-06-19:
测试的时候见不到人，就想等别人测完了再来享受。

iseeyou, 2022-06-19:
cr66x 7621 wifi也不行

EnnawYang, 2022-06-19:
~~测试贴发了那么久不见几个人测试，合并了之后问题就都来了~~

britneyjeanspears, 2022-06-19:
还是别管这些人得了，有问题的自行用老版本，反正也不参与测试。

mcgivefenngg, 2022-06-19:
> 测试贴发了那么久不见几个人测试，合并了之后问题就都来了

这B70是手里唯一的设备，拿来测试的话就没网用了，合并后刷了一次才发现Wi-Fi的问题- -

britneyjeanspears, 2022-06-19:
薛定谔的“一台设备”。测试的时候“一台设备”是测试会没网用，合并后“一台设备”就能测了。

Hiyajomaho-num9, 2022-06-20:
这排版不想看

jvesky, 2022-06-22:
同样的问题，HC5962/B70 ，WIFI有信号，可以连接。
但是测试下载，没速度（或者非常低）。

jvesky, 2022-06-22:
`Wed Jun 22 20:21:15 2022 daemon.err uhttpd[2957]: sh: write error: Broken pipe
Wed Jun 22 20:25:58 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: authenticated
Wed Jun 22 20:25:58 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: associated (aid 2)
Wed Jun 22 20:25:58 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 5c:ad:cf:5a:7f:96
Wed Jun 22 20:25:58 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 RADIUS: starting accounting session 1BECC454CA068442
Wed Jun 22 20:25:58 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 WPA: pairwise key handshake completed (RSN)
Wed Jun 22 20:25:58 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 5c:ad:cf:5a:7f:96
Wed Jun 22 20:26:41 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 5c:ad:cf:5a:7f:96
Wed Jun 22 20:26:41 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: disassociated
Wed Jun 22 20:26:42 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jun 22 20:26:58 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED ee:83:72:2d:37:4d
Wed Jun 22 20:26:58 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: disassociated
Wed Jun 22 20:26:59 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jun 22 20:27:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: authenticated
Wed Jun 22 20:27:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: associated (aid 1)
Wed Jun 22 20:27:00 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ee:83:72:2d:37:4d
Wed Jun 22 20:27:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d RADIUS: starting accounting session 18CA12F428E2E396
Wed Jun 22 20:27:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d WPA: pairwise key handshake completed (RSN)
Wed Jun 22 20:27:00 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED ee:83:72:2d:37:4d
Wed Jun 22 20:33:54 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED ee:83:72:2d:37:4d
Wed Jun 22 20:33:54 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: disassociated
Wed Jun 22 20:33:55 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jun 22 20:35:43 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: authenticated
Wed Jun 22 20:35:43 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 IEEE 802.11: associated (aid 1)
Wed Jun 22 20:35:43 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 5c:ad:cf:5a:7f:96
Wed Jun 22 20:35:43 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 RADIUS: starting accounting session 92467F6DEA0B2D08
Wed Jun 22 20:35:43 2022 daemon.info hostapd: wlan1: STA 5c:ad:cf:5a:7f:96 WPA: pairwise key handshake completed (RSN)
Wed Jun 22 20:35:43 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 5c:ad:cf:5a:7f:96
Wed Jun 22 20:39:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: authenticated
Wed Jun 22 20:39:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d IEEE 802.11: associated (aid 2)
Wed Jun 22 20:39:00 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED ee:83:72:2d:37:4d
Wed Jun 22 20:39:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d RADIUS: starting accounting session BA58E80F0722DC0D
Wed Jun 22 20:39:00 2022 daemon.info hostapd: wlan1: STA ee:83:72:2d:37:4d WPA: pairwise key handshake completed (RSN)
Wed Jun 22 20:39:00 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED ee:83:72:2d:37:4d`

dengxin6, 2022-06-23:
这些通通都是不看说明的傻逼们，  不保留配置升级， 或者先重置

sonytxl, 2022-06-25:
有解决的办法了吗？

xiangfeigao, 2022-07-06:
无线速度太慢了，我还以为是单位网咯设置呢，快点修复

asiaon123, 2022-07-06:
> 无线速度太慢了，我还以为是单位网咯设置呢，快点修复

你怎么这么牛逼

closed 2022-07-27 by aiamadeus