coolsnowwolf/lede #9615: NTFS文件系统无法正常挂载

lariboo, 2022-06-19:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

我的移动硬盘文件系统格式是NTFS，接到x86的USB口后不能正常使用，经多次测试发现可复现的bug：1.如果是使用kmod-fs-ntfs，磁盘能够自动挂载，但磁盘是只读的，且不显示中文目录和中文文件；2.如果是使用ntfs-3g，则磁盘不能被自动挂载，需要手动执行mount -t ntfs-3g -o rw /dev/sdb1 /mnt/sdb1挂载命令，这样挂载之后的磁盘是正常的，但这只是临时挂载，重启后失效。该bug从4月下旬出现，所有内核版本均能复现。

JusticeEER, 2022-06-19:
/etc/hotplug.d/block/10-mount 这里加上脚本试过了吗？

blkdev=`dirname $DEVPATH`
if [ `basename $blkdev` != "block" ]; then
    device=`basename $DEVPATH`
    case "$ACTION" in
        add)
                mkdir -p /mnt/$device

                # vfat & ntfs-3g check
                if  [ `which fdisk` ]; then
                        isntfs=`fdisk -l | grep $device | grep NTFS`
                        isvfat=`fdisk -l | grep $device | grep FAT`
                        isfuse=`lsmod | grep fuse`
                        isntfs3g=`which ntfs-3g`
                else
                        isntfs=""
                        isvfat=""
                fi

                # mount with ntfs-3g if possible, else with default mount
                if [ "$isntfs" -a "$isfuse" -a "$isntfs3g" ]; then
                       ntfs-3g -o nls=utf8 /dev/$device /mnt/$device
                elif [ "$isvfat" ]; then
                       mount -t vfat -o iocharset=utf8,rw,sync,umask=0000,dmask=0000,fmask=0000 /dev/$device /mnt/$device
                else
                       mount -o nls=utf8,umask=0000 /dev/$device /mnt/$device
                fi
                if [ -f /dev/${device}/swapfile ]; then
                       mkswap /dev/${device}/swapfile
                       swapon /dev/${device}/swapfile
                fi
        ;;
        remove)
                if [ -f /dev/${device}/swapfile ]; then
                       swapoff /dev/${device}/swapfile
                fi
                umount /dev/$device
        ;;
    esac
fi

lariboo, 2022-06-20:
> /etc/hotplug.d/block/10-mount 这里加电容了吗？
> 
> blkdev= `dirname $DEVPATH` if [ `basename $blkdev`!= "block" ]; 然后 device= `basename $DEVPATH` case "$ACTION" in add) mkdir -p /mnt/$device
> 
> ```
>             # vfat & ntfs-3g check
>             if  [ `which fdisk` ]; then
>                     isntfs=`fdisk -l | grep $device | grep NTFS`
>                     isvfat=`fdisk -l | grep $device | grep FAT`
>                     isfuse=`lsmod | grep fuse`
>                     isntfs3g=`which ntfs-3g`
>             else
>                     isntfs=""
>                     isvfat=""
>             fi
> 
>             # mount with ntfs-3g if possible, else with default mount
>             if [ "$isntfs" -a "$isfuse" -a "$isntfs3g" ]; then
>                    ntfs-3g -o nls=utf8 /dev/$device /mnt/$device
>             elif [ "$isvfat" ]; then
>                    mount -t vfat -o iocharset=utf8,rw,sync,umask=0000,dmask=0000,fmask=0000 /dev/$device /mnt/$device
>             else
>                    mount -o nls=utf8,umask=0000 /dev/$device /mnt/$device
>             fi
>             if [ -f /dev/${device}/swapfile ]; then
>                    mkswap /dev/${device}/swapfile
>                    swapon /dev/${device}/swapfile
>             fi
>     ;;
>     remove)
>             if [ -f /dev/${device}/swapfile ]; then
>                    swapoff /dev/${device}/swapfile
>             fi
>             umount /dev/$device
>     ;;
> esac
> ```
> 
> 菲

> /etc/hotplug.d/block/10-mount 这里加上脚本试过了吗？
>
> blkdev=`dirname $DEVPATH` if [ `basename $blkdev` != "block" ]; then device=`basename $DEVPATH` case "$ACTION" in add) mkdir -p /mnt/$device [...]

未曾使用过脚本。用老兄的脚本做测试，依旧无效，可能是我姿势不对。老兄能否上传下10-mount这个文件？不胜感激。

binge8, 2022-06-20:
同样的问题，最新编译的R22.6.16 / LuCI Master (git-22.168.62806-d59f9ce)出问题的

JusticeEER, 2022-06-25:
> 未曾使用过脚本。用老兄的脚本做测试，依旧无效，可能是我姿势不对。老兄能否上传下10-mount这个文件？不胜感激。

[10-mount.txt](https://github.com/coolsnowwolf/lede/files/8984465/10-mount.txt)

lariboo, 2022-06-26:
> > > /etc/hotplug.d/block/10-mount 这里加加电容了吗？
> > > blkdev= `dirname $DEVPATH`if [ `basename $blkdev`!= block ]; 然后 device= `basename $DEVPATH`case "$ACTION" in add) mkdir -p /mnt/$device
> > > ```
> > >             # vfat & ntfs-3g check
> > >             if  [ `which fdisk` ]; then
> > >                     isntfs=`fdisk -l | grep $device | grep NTFS`
> > >                     isvfat=`fdisk -l | grep $device | grep FAT`
> > >                     isfuse=`lsmod | grep fuse`
> > >                     isntfs3g=`which ntfs-3g`
> > >             else
> > >                     isntfs=""
> > >                     isvfat=""
> > >             fi
> > > 
> > >             # mount with ntfs-3g if possible, else with default mount
> > >             if [ "$isntfs" -a "$isfuse" -a "$isntfs3g" ]; then
> > >                    ntfs-3g -o nls=utf8 /dev/$device /mnt/$device
> > >             elif [ "$isvfat" ]; then
> > >                    mount -t vfat -o iocharset=utf8,rw,sync,umask=0000,dmask=0000,fmask=0000 /dev/$device /mnt/$device
> > >             else
> > >                    mount -o nls=utf8,umask=0000 /dev/$device /mnt/$device
> > >             fi
> > >             if [ -f /dev/${device}/swapfile ]; then
> > >                    mkswap /dev/${device}/swapfile
> > >                    swapon /dev/${device}/swapfile
> > >             fi
> > >     ;;
> > >     remove)
> > >             if [ -f /dev/${device}/swapfile ]; then
> > >                    swapoff /dev/${device}/swapfile
> > >             fi
> > >             umount /dev/$device
> > >     ;;
> > > esac
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 菲
> > 
> > 
> > 依旧未曾使用过脚本，用老兄的脚本，可能我的姿势不对。老兄可以做上传下10-文件？
> 
> [10-mount.txt](https://github.com/coolsnowwolf/lede/files/8984465/10-mount.txt)
果然是我的姿势不对，用老兄的文件已解决。谢谢！

wlshdjj, 2022-08-21:
static const match_table_t tokens = {
	{antfs_opt_utf8, "utf8"},
	{antfs_opt_uid, "uid=%u"},
	{antfs_opt_gid, "gid=%u"},
	{antfs_opt_umask, "umask=%u"},
	{antfs_opt_err, NULL}
};
这是antfs的参数（antfs来源于ntfs-3g，ntfs-3g一样是这几个参数，这两驱动的参数一直没改变，而到了ntfs3（依然来源于ntfs-3g）就被改了两次，最后确定在了国际惯例的iocharset=utf8），所以ntfs-3g只支持-o utf8这个参数，其它乱加的***=utf8的都不被调用。这是这三个东西的正经用法：
mount -t ntfs3 -o iocharset=utf8 "$@"
mount -t antfs -o utf8 "$@"
/usr/bin/ntfs-3g -o utf8 "$@"
只需要去改mount.ntfs就好了，哪需要那么多麻烦。

lariboo, 2022-08-26:
> static const match_table_t tokens = { {antfs_opt_utf8, "utf8"}, {antfs_opt_uid, "uid=%u"}, {antfs_opt_gid, "gid=%u"}, {antfs_opt_umask, "umask=%u"}, {antfs_opt_err, NULL} }; [...]

感谢指点迷津。那么这个10-mount文件应该如何写？

closed 2022-08-29 by lariboo