coolsnowwolf/lede #9621: 安装编译依赖出错

jgsports, 2022-06-20:
用虚拟机ubuntu 20.04 lts编译lean openwrt，以前没问题，这次虚拟机重装，安装编译依赖时，提示下列错误
Ign:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fonts-gfs-porson all 1.1-7
Ign:2 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fastjar amd64 2:0.98-7
Ign:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fonts-gfs-porson all 1.1-7
Ign:2 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fastjar amd64 2:0.98-7
Ign:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fonts-gfs-porson all 1.1-7
Ign:2 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fastjar amd64 2:0.98-7
Ign:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fonts-gfs-porson all 1.1-7
Err:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fonts-gfs-porson all 1.1-7
  Connection failed [IP: 91.189.91.38 80]
Err:2 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 fastjar amd64 2:0.98-7
  Connection failed [IP: 91.189.91.38 80]
E: Failed to fetch http://us.archive.ubuntu.com/ubuntu/pool/universe/f/fonts-gfs-porson/fonts-gfs-porson_1.1-7_all.deb  Connection failed [IP: 91.189.91.38 80]
E: Failed to fetch http://us.archive.ubuntu.com/ubuntu/pool/universe/f/fastjar/fastjar_0.98-7_amd64.deb  Connection failed [IP: 91.189.91.38 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

换国内的源也这样，网上也有碰到这类问题的

EnnawYang, 2022-06-20:
所以编译机的问题也能来这发issue了嘛

jgsports, 2022-06-20:
楼上，网络是全局没问题，这种口气说话就表明你是大佬吗？

EnnawYang, 2022-06-20:
咱不是大佬，但至少咱环境和编译都没有问题

1054009064, 2022-06-20:
我昨天就装了一遍 20.04，没有出现这个问题。

sbwcwfhv001, 2022-06-20:
自己百度去吧

jgsports, 2022-06-20:
谷歌查了很久，百度就免了

sbwcwfhv001, 2022-06-20:
用国产系统也能编译，deepin,我一直用的这个系统编译的

jgsports, 2022-06-20:
谢谢

qqli007, 2022-08-22:
网络问题，需要外网安装fastjar