coolsnowwolf/lede #9624: NanoPi R2S PPPOE拨号 上行速率异常

Neutralization, 2022-06-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络设备是光猫改桥接，接R2S拨号，后接AP。

拨号可正常连接，下行正常，上行在建立任何下载连接时能在任务管理器看到一个4MB~8MB/s的峰值，然后基本处在2MB/s以下的速度。
拔掉R2S，光猫后直接接电脑拨号正常，上下行数据一切正常。
R2S待机时44℃，产生持续上下行数据时目前观测到温度峰值会一路飙升至76℃。

改桥接前是光猫正常拨号，后接R2S，后接AP。上下行速度一切正常（上行可稳定在7MB/s左右），温度正常。

想请教如何定位问题原因，是否有解决办法。

### (2) 路由器型号和固件版本

A：
NanoPi R2S
LuCI git-22.161.56906-efa5cfe / OpenWrt R22.6.16

编译参数在 https://github.com/Neutralization/OpenWrt/blob/main/.config

### (3) 详细日志

A：系统日志

Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Mon Jun 20 21:29:16 2022 kern.notice kernel: [    0.000000] Linux version 5.10.120 (n@null) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4573-23f225e84) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP PREEMPT Wed Jun 15 15:05:59 2022
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Machine model: FriendlyElec NanoPi R2S
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Zone ranges:
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000]   DMA32    empty
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000]   Normal   empty
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Early memory node ranges
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 261632
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000]   DMA zone: 4088 pages used for memmap
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000]   DMA zone: 261632 pages, LIFO batch:63
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] cma: Reserved 8 MiB at 0x000000003f400000
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s46360 r8192 d31464 u86016
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s46360 r8192 d31464 u86016 alloc=21*4096
Mon Jun 20 21:29:16 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Mon Jun 20 21:29:16 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] Memory: 1002448K/1046528K available (8062K kernel code, 2362K rwdata, 2320K rodata, 1728K init, 725K bss, 35888K reserved, 8192K cma-reserved)
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Mon Jun 20 21:29:16 2022 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
[... 431 lines cut ...]
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 3 br-lan 192.168.20.1:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 4 pppoe-wan 114.88.163.224:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 5 utun 198.18.0.1:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 6 ztppirzd7j 192.168.196.240:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 7 lo [::1]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 8 eth0 [fe80::84e0:c0ff:feea:faa9%2]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 9 br-lan [fd73:f5ca:bc48::1]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 10 br-lan [fe80::4419:b3ff:fe10:2842%4]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 11 pppoe-wan [fe80::b9e5:b066:1dca:8518%5]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 12 utun [fe80::c85b:3c90:3466:528c%7]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listen normally on 13 ztppirzd7j [fe80::bcb7:6bff:fe26:c9af%8]:123
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: Listening on routing socket on fd #62 for interface updates
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Mon Jun 20 21:30:03 2022 daemon.info ntpd[6352]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Mon Jun 20 21:30:14 2022 user.notice firewall: Reloading firewall due to ifup of Zerotier (ztppirzd7j)
Mon Jun 20 21:30:16 2022 daemon.info ntpd[6352]: Deleting interface #5 utun, 198.18.0.1#123, interface stats: received=0, sent=0, dropped=0, active_time=13 secs
Mon Jun 20 21:30:16 2022 daemon.info ntpd[6352]: Deleting interface #12 utun, fe80::c85b:3c90:3466:528c%7#123, interface stats: received=0, sent=0, dropped=0, active_time=13 secs
Mon Jun 20 21:30:17 2022 cron.err crond[7451]: crond (busybox 1.35.0) started, log level 8
Mon Jun 20 21:30:18 2022 cron.err crond[7664]: crond (busybox 1.35.0) started, log level 8
Mon Jun 20 21:30:18 2022 daemon.info ntpd[6352]: Listen normally on 14 utun 198.18.0.1:123
Mon Jun 20 21:30:18 2022 daemon.info ntpd[6352]: Listen normally on 15 utun [fe80::c85b:3c90:3466:528c%7]:123
Mon Jun 20 21:30:18 2022 daemon.debug ntpd[6352]: new interface(s) found: waking up resolver
Mon Jun 20 21:30:49 2022 cron.err crond[8378]: crond (busybox 1.35.0) started, log level 8
Mon Jun 20 21:30:49 2022 cron.err crond[8591]: crond (busybox 1.35.0) started, log level 8
Mon Jun 20 21:35:51 2022 daemon.info ntpd[6352]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized

LPDDR6-10000MHz, 2022-06-20:
有没有想过[openclash](https://github.com/Neutralization/OpenWrt/blob/main/.config#L44)开了会过高？还有kmod-crypto*模块为什么一定要关着？

Neutralization, 2022-06-21:
因为公司有放着同一个映像刷写的第二台R2S，上下行和温度相比之下一切正常。两者唯一的区别是家中的wan口pppoe拨号，所以没有去怀疑openclash的问题。
kmod-crypto*模块没有去手动取消，应该是取消了部分luci-app*后连带取消的，我之后试一下手动加回来会不会有用。

Neutralization, 2022-06-21:
> 有没有想过[openclash](https://github.com/Neutralization/OpenWrt/blob/main/.config#L44)开了会过高？还有kmod-crypto*模块为什么一定要关着？

kmod-crypto*全部加入编译后问题上行解决，感谢。

closed 2022-06-21 by Neutralization