coolsnowwolf/lede #9627: 无线网卡驱动rtl8821cu曾正常工作，最近反而不能用了

szdosar, 2022-06-21:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
我有一个usb-wifi卡，芯片驱动是rtl8821cu。

### (1) 具体问题
我曾在2023年3月26日，使用贵仓库成功编译，并正常使用；
当时的内核版本 | 5.15.31
version.buildinfo | r4370-6d687a132
kmod-rtl8821cu | 5.15.31+2021-11-14-ef..1-1

用了最新的代码编译后，
刷机，只要开启这个usb-wifi卡，
web界面就会卡死，
ssh无法取得任何ip（如ip addr 无响应），
web界面停止响应，
无法进行任何操作，
直到我把这个usb-wifi卡拔除，
再次重启路由器后，
方可正常操控路由器。
A：

### (2) 路由器型号和固件版本
x86-j1900\友善-r2s\vmware-x86等均尝试过
比如，Linux version 5.15.48  (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4589-f01d99d61) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Fri Jun 17 16:34:04 2022
我尝试过贵源编译的很多版本，
都发现次问题。
ps，我把这个驱动移植到lienol的仓库下（lienol源无此驱动），
编译后反而可以正常使用（ta的内核是5.4）
但是，我尝试用你的5.4内核编译，依然无解
A：

### (3) 详细日志
[异常日志.txt](https://github.com/coolsnowwolf/lede/files/8945346/default.txt)
[正常日志.txt](https://github.com/coolsnowwolf/lede/files/8945295/default.txt)
我分别提供异常日志（前者）和正常日志供你参阅。
谢谢你花宝贵的时间来解答此问题。
A：

WYC-2020, 2022-06-21:
可以试下回滚到，这个提交在试下
[image]

WYC-2020, 2022-06-21:
如果还不行 你就 一个月一个月的回滚 具体是那个提交导致的这就不好确定了，要么就等大佬给你修 哈哈，如果能找到那个提交点的话，应该修起来没那么麻烦 @szdosar

szdosar, 2022-06-21:
> 可以试下回滚到，这个提交在试下 [image]

哈哈，咋样回滚？小白求具体步骤或教程。

1582130940, 2022-06-22:
> > 可以试下回滚到，这个提交在试下 [image]
> 
> 哈哈，咋样回滚？小白求具体步骤或教程。

git reset --hard f6f103b1cb4bfb09ff8d690618b0fe9132e07f96

closed 2022-06-25 by AmadeusGhost

wlshdjj, 2022-08-21:
3月29号更新了hostapd、mac80211：
https://github.com/coolsnowwolf/lede/commit/a152752dc0bbebee89487222825b459f6d3f4a91
https://github.com/coolsnowwolf/lede/commit/cc601320047c07ae464695b76258d857a614a981
问题一定出在mac80211，对比日志发现一句明显异常：
[  118.570911] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
由于没有提供wireless配置文件，猜测是wireless初始化配置异常：path:0。band可以不要，只是会丧失少量功能，path不能没有，否则连用都没法用。这个初始化配置由mac80211中的脚本产生。你只需要回滚mac80211这一个模块应该就可以解决的了。如果wireless配置是正常的，那么问题就出在hostapd，这些参数由hostapd中的脚本获取并驱动无线，但别的无线网卡正常的话这个可能性很小。

szdosar, 2022-09-28:
> 

请大神指点一下，怎样只回滚mac80211一个模块？非常感谢！！