coolsnowwolf/lede #9630: 关于OpenWRT的DNS解析服务异常（缓慢或者超时无响应）的问题反馈

AutoCONFIG, 2022-06-22:
##1.lede的内置DNS解析服务（dnsmasq）偶尔会出现dns解析异常（大概率情况下是超时无响应）的情况

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：内置的DNS解析服务偶尔会出现dns解析问题，反映在电脑或者手机上即浏览器提示DNS解析服务无响应或者网页加载异常（无图片和网站样式等）或者刷视频时视频或者评论等无法正常显示。目前测试会在P2P下载或者短时间内反复nslookup（1分钟内执行10-20次）时会出现上述情况，即短时较大量dns解析请求会触发dns拒绝服务。未开启TurboACC的DNS缓存时会出现上述状况，开启后出现上述状况的概率会小幅增加。尝试过去除TurboACC组件编译，上述状况依然会出现。

### (2) 路由器型号和固件版本

A：路由器型号：x86-64 实达 D2550 双千兆
固件版本：OpenWrt R22.6.16 / LuCI Master (git-22.168.62806-d59f9ce)
编译时config: 去掉很多自己用不到的luci-app-application，除<*>必选外只选择了turboacc，samba，upnp，kms插件
也尝试过使用默认选项编译，但是依旧出现上述dns解析异常的问题

### (3) 详细日志

A：内核日志和系统日志在出现上述问题时并无相应的日志，不过有时出现dns解析异常时会有uhttpd的debug：
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: Plugin pppoe.so loaded.
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: PPPoE plugin from pppd 2.4.9
Wed Jun 22 06:00:00 2022 daemon.notice pppd[23462]: pppd 2.4.9 started by root, uid 0
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: PPP session is 4126
Wed Jun 22 06:00:00 2022 daemon.warn pppd[23462]: Connected to 70:44:00:4f:ff:04 via interface eth1
Wed Jun 22 06:00:00 2022 kern.info kernel: [117031.749649] pppoe-wan: renamed from ppp0
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: Renamed interface ppp0 to pppoe-wan
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: Using interface pppoe-wan
Wed Jun 22 06:00:00 2022 daemon.notice pppd[23462]: Connect: pppoe-wan <--> eth1
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: syncppp not active
Wed Jun 22 06:00:00 2022 daemon.info pppd[23462]: Remote message: Authentication Successful,Welcome!
Wed Jun 22 06:00:00 2022 daemon.notice pppd[23462]: PAP authentication succeeded
Wed Jun 22 06:00:00 2022 daemon.notice pppd[23462]: peer from calling number 70:44:00:4F:FF:04 authorized
Wed Jun 22 06:00:01 2022 daemon.notice pppd[23462]: local  IP address 100.106.80.241
Wed Jun 22 06:00:01 2022 daemon.notice pppd[23462]: remote IP address 100.106.0.1
Wed Jun 22 06:00:01 2022 daemon.notice pppd[23462]: primary   DNS address 211.138.156.66
Wed Jun 22 06:00:01 2022 daemon.notice pppd[23462]: secondary DNS address 211.138.151.161
Wed Jun 22 06:00:01 2022 daemon.info avahi-daemon[1858]: Joining mDNS multicast group on interface pppoe-wan.IPv4 with address 100.106.80.241.
Wed Jun 22 06:00:01 2022 daemon.info avahi-daemon[1858]: New relevant interface pppoe-wan.IPv4 for mDNS.
Wed Jun 22 06:00:01 2022 daemon.info avahi-daemon[1858]: Registering new address record for 100.106.80.241 on pppoe-wan.IPv4.
Wed Jun 22 06:00:01 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Wed Jun 22 06:00:01 2022 daemon.notice netifd: Interface 'wan' is now up
Wed Jun 22 06:00:01 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Wed Jun 22 10:32:29 2022 daemon.err uhttpd[1734]: cat: can't open '/etc/bench.log': No such file or directory
Wed Jun 22 10:32:29 2022 daemon.err uhttpd[1734]: cat: can't open '/etc/bench.log': No such file or directory
Wed Jun 22 10:46:41 2022 daemon.err uhttpd[1734]: cat: can't open '/etc/bench.log': No such file or directory
Wed Jun 22 10:47:24 2022 daemon.err uhttpd[1734]: cat: can't open '/etc/bench.log': No such file or directory
Wed Jun 22 10:48:11 2022 authpriv.info dropbear[4453]: Child connection from 192.168.1.244:7003
Wed Jun 22 10:48:13 2022 authpriv.warn dropbear[4453]: Bad password attempt for 'root' from 192.168.1.244:7003
Wed Jun 22 10:48:16 2022 authpriv.notice dropbear[4453]: Password auth succeeded for 'root' from 192.168.1.244:7003
Wed Jun 22 10:48:49 2022 authpriv.info dropbear[4453]: Exit (root) from <192.168.1.244:7003>: Disconnect received

AutoCONFIG, 2022-06-25:
经过几天的调试，大概率是turboacc插件中DNS相关组件的问题（包括依赖的pdnsd）。剔除掉turboacc中DNS相关的组件后，lede再也没有出现过dns解析异常的问题。
关于turboacc，就算从未启用过DNS缓存，也会导致DNS解析异常，必须在编译前完全剔除DNS相关组件才行。

qyh214, 2022-06-27:
> 经过简单的调试，大概率是turboacc插件中DNS相关组件的问题（包括依赖的pdnsd）。剔除掉turboacc中DNS相关的组件后，lede再也没有出现过dns解析异常的问题。 关于turboacc，就算从未启用过DNS缓存，也会导致DNS解析异常，必须在编译前完全剔除turboacc中DNS相关代码才行。

那么如何剔除呢，需要修改源码还是只需要对配置文件做修改

AutoCONFIG, 2022-06-28:
> > 经过简单的调试，大概率是turboacc插件中DNS相关组件的问题（包括依赖的pdnsd）。剔除掉turboacc中DNS相关的组件后，lede再也没有出现过dns解析异常的问题。 关于turboacc，就算从未启用过DNS缓存，也会导致DNS解析异常，必须在编译前完全剔除turboacc中DNS相关代码才行。
> 
> 那么如何剔除呢，需要修改源码还是只需要对配置文件做修改

修改源码，把DNS相关的代码和脚本删除。我这边已经将修改后的代码上传到仓库：https://github.com/AutoCONFIG/luci-app-turboacc，编译时先rm -rf feeds/luci/applications/luci-app-turboacc文件夹，然后git clone https://github.com/AutoCONFIG/luci-app-turboacc.git feeds/luci/applications/luci-app-turboacc，最后再生成config就行了。

AutoCONFIG, 2022-07-09:
除了turboacc模块会使得DNS暂时性解析异常外，miniupnpd组件也会使DNS暂时性解析异常。不过miniupnpd的问题可能跟firewall有关。

feifan1113, 2022-07-15:
> 须在编译前完全剔除turboacc中DNS相关代码才行。

可以适配arm64吗  aarch的

closed 2023-01-24 by AutoCONFIG