coolsnowwolf/lede #9641: K2P最新固件，端口转发中的“启用 NAT 环回”功能失效，内网无法访问转发端口

MageDelfador, 2022-06-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：端口转发中的“启用 NAT 环回”功能失效，内网无法访问转发端口。外网可以正常访问转发端口。（不保留配置升级，默认选项编译）

### (2) 路由器型号和固件版本

A：Phicomm K2P  OpenWrt R22.6.16 / LuCI Master (git-22.174.19578-bdf3e99)

### (3) 详细日志

A：没有产生端口转发相关的日志

MageDelfador, 2022-06-25:
刚刚发现内网无线可以访问转发端口，但速度很慢。内网通过网线则完全不能访问

另外发个内核日志吧
<details>
<summary>内核日志</summary>
<pre class="notranslate">
[    0.000000] Linux version 5.4.199 (zhen@zhen-ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4599-94e2af1bc)) #0 SMP Thu Jun 23 05:27:22 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Phicomm K2P
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s27024 r8192 d22128 u57344
[    0.000000] pcpu-alloc: s27024 r8192 d22128 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000040
[    0.000000] Readback ErrCtl register=00000040
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120264K/131072K available (6166K kernel code, 272K rwdata, 1296K rodata, 1276K init, 236K bss, 10808K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015494] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.087833] pid_max: default: 32768 minimum: 301
[    0.097168] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.111575] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.129844] rcu: Hierarchical SRCU implementation.
[    0.139996] smp: Bringing up secondary CPUs ...
[    0.138389] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.138399] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.138413] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.138522] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.209615] Synchronize counters for CPU 1: done.
[    9.648823] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.648831] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.648838] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.648894] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.330177] Synchronize counters for CPU 2: done.
[    9.767925] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.767934] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.767942] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.768006] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.449753] Synchronize counters for CPU 3: done.
[    0.509359] smp: Brought up 1 node, 4 CPUs
[    0.522025] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 827 lines cut ...]
[ 5611.467114] Start Seq = 00000000
[ 5612.442406] Rcv Wcid(3) AddBAReq
[ 5612.448834] Start Seq = 00000000
[ 5627.005673] Rcv Wcid(3) AddBAReq
[ 5627.012142] Start Seq = 00000000
[ 5637.840933] Rcv Wcid(3) AddBAReq
[ 5637.847517] Start Seq = 00000000
[ 5667.537625] Rcv Wcid(3) AddBAReq
[ 5667.544099] Start Seq = 00000000
[ 5693.494427] Rcv Wcid(3) AddBAReq
[ 5693.500913] Start Seq = 00000000
[ 5713.806518] Rcv Wcid(3) AddBAReq
[ 5713.813054] Start Seq = 00000000
[ 5749.406874] Rcv Wcid(3) AddBAReq
[ 5749.413658] Start Seq = 00000000
[ 5755.093491] Rcv Wcid(3) AddBAReq
[ 5755.100074] Start Seq = 00000000
[ 5756.544597] Rcv Wcid(3) AddBAReq
[ 5756.551025] Start Seq = 00000000
[ 5868.195144] Rcv Wcid(3) AddBAReq
[ 5868.201961] Start Seq = 00000000
[ 5878.838349] Rcv Wcid(3) AddBAReq
[ 5878.844801] Start Seq = 00000000
</pre>
</details>

sonytxl, 2022-06-25:
你的WIFI正常吗？我的有线OK，WiFi默认编译的，速度几乎没有

MageDelfador, 2022-06-25:
> 你的WIFI正常吗？我的有线OK，WiFi默认编译的，速度几乎没有

正常，手机跑个宽带测速下载600M左右，应该还能更高。
可以看下交换机设置里，VLAN 1里的CPU (eth0)  要选“未标记”  否则WIFI就不行。也可以重置下路由器恢复默认配置

sonytxl, 2022-06-26:
> > 你的WIFI正常吗？我的有线OK，WiFi默认编译的，速度几乎没有
> 
> 正常，手机跑个宽带测速下载600M左右，应该还能更高。 可以看下交换机设置里，VLAN 1里的CPU (eth0) 要选“未标记” 否则WIFI就不行。也可以重置下路由器恢复默认配置

兄弟你好，再请教一下，我的是k2p和你一样，我写的固件里面只有无线和接口设置，没有交换机设置，在接口设置里面WAN默认的就是WAN，eth0默认是没有打钩的，也找不到CPU这个选项，请问一下是再哪个选项下面可以找到？谢谢你。。。

MageDelfador, 2022-06-26:
> > > 你的WIFI正常吗？我的有线OK，WiFi默认编译的，速度几乎没有
> > 
> > 
> > 正常，手机跑个宽带测速下载600M左右，应该还能更高。 可以看下交换机设置里，VLAN 1里的CPU (eth0) 要选“未标记” 否则WIFI就不行。也可以重置下路由器恢复默认配置
> 
> 兄弟你好，再请教一下，我的是k2p和你一样，我写的固件里面只有无线和接口设置，没有交换机设置，在接口设置里面WAN默认的就是WAN，eth0默认是没有打钩的，也找不到CPU这个选项，请问一下是再哪个选项下面可以找到？谢谢你。。。

可能是编译的时候配置选项有问题，make distclean后再重新设置编译一下试试吧

sonytxl, 2022-06-26:
> 
谢谢

sonytxl, 2022-06-26:
> > > > 你的WIFI正常吗？我的有线OK，WiFi默认编译的，速度几乎没有
> > > 
> > > 
> > > 正常，手机跑个宽带测速下载600M左右，应该还能更高。 可以看下交换机设置里，VLAN 1里的CPU (eth0) 要选“未标记” 否则WIFI就不行。也可以重置下路由器恢复默认配置
> > 
> > 
> > 兄弟你好，再请教一下，我的是k2p和你一样，我写的固件里面只有无线和接口设置，没有交换机设置，在接口设置里面WAN默认的就是WAN，eth0默认是没有打钩的，也找不到CPU这个选项，请问一下是再哪个选项下面可以找到？谢谢你。。。
> 
> 可能是编译的时候配置选项有问题，make distclean后再重新设置编译一下试试吧

再次请教一下，我在重新编译的时候luci找不到交换机的选项再哪里,请问交换机选项再哪个位置可以找到，再次感谢。。

MageDelfador, 2022-06-26:
> 再次请教一下，我在重新编译的时候luci找不到交换机的选项再哪里,请问交换机选项再哪个位置可以找到，再次感谢。。

不需要额外修改，默认编译，路由器刷完固件里面就有了

Subdue0, 2022-07-05:
我也遇到了这个问题，使用无线网络访问本地的服务，速度只有几十KB，有线网络一切正常，这要怎么解决呢？

Subdue0, 2022-07-05:
@sonytxl 你的问题解决了吗？

MageDelfador, 2023-01-29:
修改/etc/sysctl.conf，添加：
```
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0
```
然后执行sysctl -p
也可以在编译前直接修改/lede/package/base-files/files/etc/sysctl.conf
搜索了一下ISSUES，好像这些配置总是随着内核变来变去，
所以最佳方案应该就是这样修改sysctl.conf文件了

closed 2023-01-29 by MageDelfador

steamch, 2023-01-29:
> 修改/etc/sysctl.conf，添加：
> 
> ```
> net.bridge.bridge-nf-call-arptables=0
> net.bridge.bridge-nf-call-ip6tables=0
> net.bridge.bridge-nf-call-iptables=0
> ```
> 
> 然后执行sysctl -p 也可以在编译前直接修改/lede/package/base-files/files/etc/sysctl.conf 搜索了一下ISSUES，好像这些配置总是随着内核变来变去， 所以最佳方案应该就是这样修改sysctl.conf文件了

sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-ip6tables' is an unknown key
sysctl: error: 'net.bridge/bridge-nf-call-iptables' is an unknown key
会出现这个。这个方法以前有效，现在不行了

MageDelfador, 2023-01-29:
我的是K2P，默认配置编译，5.4内核
可能是变斜杠的问题，找找变斜杠的原因吧