coolsnowwolf/lede #9657: 最近一直编译失败，而且一直在这个地方

mzl0408, 2022-06-27:
CC      fs/io-wq.o
  CC      fs/locks.o
  CC      fs/binfmt_script.o
  CC      fs/binfmt_elf.o
  CC      fs/compat_binfmt_elf.o
  CC      fs/mbcache.o
  CC      fs/posix_acl.o
  CC      fs/coredump.o
  CC      fs/drop_caches.o
  CC      fs/fhandle.o
  AR      fs/built-in.a
make[5]: Leaving directory '/home/meng/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.50'
make[4]: *** [Makefile:27: /home/meng/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.50/.modules] Error 2
make[4]: Leaving directory '/home/meng/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/meng/lede/target/linux'
time: target/linux/compile#1159.50#105.81#121.02
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/meng/lede'
make[1]: *** [target/Makefile:18: /home/meng/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/meng/lede'
make: *** [/home/meng/lede/include/toplevel.mk:230：world] 错误 2

zhangguanzhang, 2022-06-28:
make V=s 看下完整的详细的日志

HollyMadCow, 2022-06-28:
我也出现类似问题，后来发现是make clean的时候清理掉了一些配置文件，这个文件在重新编译的时候不会重新生成，我的解决办法是删除真个重新拉重新编译

closed 2022-07-27 by aiamadeus