coolsnowwolf/lede #9660: Xiaomi cr660x 配置二级路由后wifi网络问题

Guxing0321, 2022-06-28:
Hello Team,

发现一个问题，就是xiaomi cr6608 路由器刷了lede固件后，配置二级路由，网线连接是没问题的，但是wifi基本上网页打不开
[image]
[image]

[image]

目前发现的问题，
wifi 连接后ping 114.114.114.114 这种外网地址是可以通的，但是ping www.baidu.com 有时候可以，有时候解析不出来

日志我拷贝出来了，目前没看到明显的相关的报错
[log.txt](https://github.com/coolsnowwolf/lede/files/8997081/log.txt)

希望大佬们帮忙看下，是不是哪边需要改下测试，或者怎么样测试下

BR
Robin

286700234, 2022-06-28:
我一级路由都WiFi跟你一样的情况

Guxing0321, 2022-06-29:
> 我一级路由都WiFi跟你一样的情况

hello， 这个应该不对呀，我记得之前就是这么配置的，也没出啥问题
感觉新的固件版本可能有某个地方需要改配置啥的

BR
Robin

Guxing0321, 2022-06-30:
我发现了一个解决的办法，
就是回退版本，realease tag: **20220505**

我用的是action编译，
[image]

BR
Robin

closed 2022-06-30 by Guxing0321