coolsnowwolf/lede #9664: lede全默认设置编译失败

adamgonglj, 2022-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

[image]

adamgonglj, 2022-06-29:
参数是全部默认的，仍然编译失败。

adamgonglj, 2022-06-29:
编译环境是腾讯云8G，Ubuntu 22.04 LTS x86_64(Py3.7.8)

EnnawYang, 2022-06-29:
默认编译无任何问题，另外能不能不要重复开issue
[image]

adamgonglj, 2022-06-29:
我看报错内容不一样，所以开的两个issue.

adamgonglj, 2022-06-29:
> 默认编译无任何问题，另外能不能不要重复开issue [image]

那我应该如何排障呢？

138398wpf, 2022-06-29:
你的报错日志介绍的很清楚，error while loading shared libraries 缺少必要的库，建议查一下/usr/local/lib是否包含提示的文件。

adamgonglj, 2022-06-30:
换系统后编译成功。

closed 2022-06-30 by adamgonglj