coolsnowwolf/lede #9670: 大佬你好，发现严重问题：lede17.01版编译ar71xx的固件时一直报错，编译失败，好像是选了luci-app-frpc的问题，请大佬指点（由于lede17.01无法发起issues只能到这里提交了）。

jacelift, 2022-06-29:
大佬你好，发现严重问题：lede17.01版编译ar71xx的固件时一直报错，编译失败，好像是选了luci-app-frpc的问题，请大佬指点（由于lede17.01无法发起issues只能到这里提交了）。

详细日志：
make[3]: Entering directory '/workdir/openwrt/package/libs/elfutils'
mkdir -p /workdir/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp
echo '1' | cmp -s - /workdir/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/libelf.version || echo '1' > /workdir/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/libelf.version
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mips_24kc_musl/elfutils-0.177/.pkgdir/libelf/. /workdir/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/'
touch /workdir/openwrt/staging_dir/target-mips_24kc_musl/root-ar71xx/stamp/.libelf_installed
echo "libelf" >> /workdir/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/elfutils.default.install
make[3]: Leaving directory '/workdir/openwrt/package/libs/elfutils'
time: package/libs/elfutils/compile#0.17#0.04#0.20
make[3]: Entering directory '/workdir/openwrt/package/libs/libmnl'
make[3]: Leaving directory '/workdir/openwrt/package/libs/libmnl'
time: package/libs/libmnl/compile#0.11#0.03#0.14
make[3]: Entering directory '/workdir/openwrt/package/network/utils/iproute2'
mkdir -p /workdir/openwrt/bin/targets/ar71xx/generic/packages /workdir/openwrt/build_dir/target-mips_24kc_musl/iproute2-full/iproute2-5.0.0/ipkg-mips_24kc/ip-full/CONTROL /workdir/openwrt/staging_dir/target-mips_24kc_musl/pkginfo
install -d -m0755 /workdir/openwrt/build_dir/target-mips_24kc_musl/iproute2-full/iproute2-5.0.0/ipkg-mips_24kc/ip-full/usr/libexec
install -m0755 /workdir/openwrt/build_dir/target-mips_24kc_musl/iproute2-full/iproute2-5.0.0/ip/ip /workdir/openwrt/build_dir/target-mips_24kc_musl/iproute2-full/iproute2-5.0.0/ipkg-mips_24kc/ip-full/usr/libexec/ip-full
find /workdir/openwrt/build_dir/target-mips_24kc_musl/iproute2-full/iproute2-5.0.0/ipkg-mips_24kc/ip-full -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package ip-full is missing dependencies for the following libraries:
libcap.so.2
make[3]: *** [/workdir/openwrt/bin/packages/mips_24kc/base/ip-full_5.0.0-1_mips_24kc.ipk] Error 1
Makefile:202: recipe for target '/workdir/openwrt/bin/packages/mips_24kc/base/ip-full_5.0.0-1_mips_24kc.ipk' failed
make[2]: *** [package/network/utils/iproute2/compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/network/utils/iproute2'
make[1]: *** [/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
time: package/network/utils/iproute2/full/compile#0.17#0.05#0.20
make: *** [world] Error 2
package/Makefile:111: recipe for target 'package/network/utils/iproute2/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
Error: Process completed with exit code 2.

jacelift, 2022-07-01:
【已经自行解决】：修改源码package/network/utils/iproute2/Makefile：
给Package/ip-full和Package/tc添加上+libcap，make menuconfig—>libraries：选中libcap

closed 2022-07-29 by jacelift

hcym, 2022-07-29:
真的很有效，感谢