coolsnowwolf/lede #9676: 咨询 ssr影响steam自动登录

MrVicent, 2022-07-01:
ssr开启后 steam保存登陆状态无效 无法自动登陆账号
关闭ssr之后steam自动登录恢复正常  
求教下这种情况怎么解决

WYC-2020, 2022-07-01:
隔壁

laaArmy, 2022-07-02:
把Steam所有域名和IP代理/直连

162.254.198.130
162.254.198.104
155.133.252.39
162.254.198.133
155.133.226.75
162.254.197.39
155.133.252.54
162.254.198.131
162.254.198.44
162.254.198.132
155.133.226.78
162.254.197.54
155.133.248.38
155.133.248.39
185.25.182.77
185.25.182.76
162.254.196.84
162.254.196.83
162.254.196.68
162.254.196.67
155.133.246.68
155.133.246.69
155.133.230.34
155.133.246.35
155.133.246.51
155.133.230.50
146.66.155.54
146.66.155.38
155.133.239.20
155.133.239.51
155.133.245.36
155.133.245.37
162.254.192.71
162.254.192.87
162.254.192.101
162.254.192.109
162.254.192.100
162.254.192.108
162.254.193.102
162.254.193.74
155.133.253.50
155.133.253.34
162.254.195.71
162.254.195.87
162.254.195.66
162.254.195.82
162.254.195.83
162.254.195.67
205.196.6.67
205.196.6.75
153.254.86.181
153.254.86.179
153.254.86.163
153.254.86.29
103.10.124.162
103.10.124.164
103.10.124.165
103.10.124.163
103.10.125.164
103.10.125.165
155.133.238.164
155.133.238.165
203.80.149.67
180.101.192.199
205.185.194.213
205.185.194.197
155.133.255.164
155.133.255.100
155.133.249.197
155.133.249.196
190.217.33.68
190.217.33.69

steampowered.com

coolsnowwolf, 2026-09-05:
SSR 影响 Steam 自动登录只有应用层现象，没有 DNS/代理规则、Steam 连接日志或最小复现；属于第三方代理策略兼容性问题，无法在 LEDE 源码中定位，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf