coolsnowwolf/lede #9682: 660x拨号、链接都正常，就是打不开网页，WiFi链接也是这样

286700234, 2022-07-01:
Fri Jul  1 22:24:05 2022 kern.notice kernel: [    0.000000] Linux version 5.4.199 (lyus@lyus) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4470-8a7297041)) #0 SMP Fri Jul 1 11:31:03 2022
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router CR660x
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Zone ranges:
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000]   HighMem  empty
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Early memory node ranges
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27024 r8192 d22128 u57344
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27024 r8192 d22128 u57344 alloc=14*4096
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Fri Jul  1 22:24:05 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00030000
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00030000
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] Memory: 250084K/262144K available (6194K kernel code, 269K rwdata, 1304K rodata, 1244K init, 236K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Jul  1 22:24:05 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.007804] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.073981] pid_max: default: 32768 minimum: 301
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.078774] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.085974] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.096541] rcu: Hierarchical SRCU implementation.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.101896] smp: Bringing up secondary CPUs ...
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.256732] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.256743] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.256755] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    3.256862] CPU1 revision is: 0001992f (MIPS 1004Kc)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.166690] Synchronize counters for CPU 1: done.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.350043] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.350052] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.350059] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    3.350115] CPU2 revision is: 0001992f (MIPS 1004Kc)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.257438] Synchronize counters for CPU 2: done.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.437401] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.437410] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    3.437418] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    3.437483] CPU3 revision is: 0001992f (MIPS 1004Kc)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.342626] Synchronize counters for CPU 3: done.
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.372492] smp: Brought up 1 node, 4 CPUs
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.381182] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.390981] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.397983] pinctrl core: initialized pinctrl subsystem
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.404790] NET: Registered protocol family 16
Fri Jul  1 22:24:05 2022 kern.debug kernel: [    0.421830] FPU Affinity set after 11720 emulations
Fri Jul  1 22:24:05 2022 kern.warn kernel: [    0.440078] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
Fri Jul  1 22:24:05 2022 kern.info kernel: [    0.451319] clocksource: Switched to clocksource GIC
[... 386 lines cut ...]
Fri Jul  1 22:24:28 2022 daemon.err odhcp6c[4474]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Fri Jul  1 22:24:29 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Jul  1 22:24:29 2022 daemon.notice netifd: Interface 'wan_6' is now up
Sat Jul  2 01:25:45 2022 daemon.info hostapd: wlan1: STA e8:6f:38:cf:c7:df IEEE 802.11: authenticated
Sat Jul  2 01:25:45 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED e8:6f:38:cf:c7:df 2
Sat Jul  2 01:25:45 2022 daemon.info hostapd: wlan1: STA e8:6f:38:cf:c7:df IEEE 802.11: associated (aid 3)
[... cut ...]
Sat Jul  2 01:33:27 2022 daemon.notice miniupnpd[12631]: Listening for NAT-PMP/PCP traffic on port 5351
Sat Jul  2 01:33:27 2022 daemon.notice miniupnpd[12631]: shutting down MiniUPnPd
Sat Jul  2 01:33:28 2022 daemon.err miniupnpd[12678]: could not open lease file: /var/upnp.leases
Sat Jul  2 01:33:28 2022 daemon.notice miniupnpd[12678]: HTTP listening on port 5000
Sat Jul  2 01:33:28 2022 daemon.notice miniupnpd[12678]: Listening for NAT-PMP/PCP traffic on port 5351
Sat Jul  2 01:34:03 2022 cron.err crond[13141]: crond (busybox 1.35.0) started, log level 5

wuzhuang123, 2022-07-02:
无线废了，受影响的都是mt7621。无线能ping通，内外网任何端口速度都奇慢，有线正常，可能是两星期前更新mt76造成的。小米mini的mt7620正常。我极路由4重启后WAN口ipv6和ipv4冲突造成无法获取ip4地址也是有问题。

liwenjie119, 2022-07-02:
合并前测试过几次反馈了这个问题，后来实在影响使用也因为忙没时间测试了。之后就直接合并了，也没找到啥原因

anbylan, 2022-07-02:
原来大家都有相同的问题，无线真的废了，求赶紧更新解决啊！

closed 2022-07-09 by coolsnowwolf