coolsnowwolf/lede #9683: Vmess 订阅在带有alterId时被丢弃，并在无法通过导入手动添加识别参数并正常使用。

Fddh2012, 2022-07-01:
## 1.关于你要提交的问题
Vmess 订阅在带有alterId时被丢弃，并在无法通过导入手动添加识别参数并正常使用。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如题

### (2) 路由器型号和固件版本

A：X86 OpenWrt R22.6.16 / LuCI Master (git-22.182.67615-30db966)

### (3) 详细日志

A：
Sat Jul  2 04:21:04 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:21:37 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:21:42 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:21:47 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:21:52 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:21:59 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.
Sat Jul  2 04:22:05 2022 daemon.err uhttpd[2083]: Warning: hk10.**.com resolves to multiple addresses: using only the first one returned by the resolver.

Michaol, 2022-07-02:
alterId不是被遗弃了么？还在用？

Michaol, 2022-07-02:
Vmess的alterID，上年11月就全默认为0（v2fly官方设定），继而不再支持设置了。所以不用再问了，不会倒退回去的。

Fddh2012, 2022-07-02:
> Vmess的alterID，上年11月就全默认为0（v2fly官方设定），继而不再支持设置了。所以不用再问了，不会倒退回去的。

那只能提工单给机场升级了。哈哈，好吧。不过有的客户端又还是支持这个alterID。

closed 2022-07-02 by Fddh2012