coolsnowwolf/lede #9691: R2S编译失败

adamgonglj, 2022-07-04:
反馈bug/问题模板，提建议请删除

系统环境：ubuntu 20.04.4 LTS x86_64(Py3.7.9)

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有找到
## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：R2S

### (3) 详细日志

A：
[image]

menphy1989, 2022-07-04:
x86也是这个报错，更新了5.15.52内核之后出现的

felixxiao99, 2022-07-04:
我的没有报错

pangfeng161, 2022-07-04:
> x86也是这个报错，更新了5.15.52内核之后出现的

前段时间也是更新内核就会编译报错,现在又重现

mitsukileung, 2022-07-04:
> x86也是这个报错，更新了5.15.52内核之后出现的
[image]
X86没有报错，可以编译

menphy1989, 2022-07-04:
make[3]: Entering directory '/home/menphy/lede/package/kernel/ksmbd'
mkdir -p /home/menphy/lede/dl
SHELL= flock /home/menphy/lede/tmp/.ksmbd-3.4.5.tar.gz.flock -c '  	/home/menphy/lede/scripts/download.pl "/home/menphy/lede/dl" "ksmbd-3.4.5.tar.gz" "2873c8ba1027fc5b04c5f5344804ef1469ad7019a033456c16ca4aa3f2c161f0" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.4.5?"    '
touch /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.prepared_cf9a91aee8c7adac5421e4ecc11daef4_6664517399ebbbc92a37c5bb081b5c53_check
. /home/menphy/lede/include/shell.sh; gzip -dc /home/menphy/lede/dl/ksmbd-3.4.5.tar.gz | tar -C /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5 -xf - --strip-components 1
[ ! -d ./src/ ] || cp -fpR ./src/. /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5

Applying ./patches/01-keep_kmod_metadata.patch using plaintext: 
patching file glob.h
touch /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.prepared_cf9a91aee8c7adac5421e4ecc11daef4_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.configured_*
rm -f /home/menphy/lede/staging_dir/target-x86_64_musl/stamp/.ksmbd_installed
(cd /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/./; if [ -x ./configure ]; then find /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/ -name config.guess | xargs -r chmod u+w; find /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/ -name config.guess | xargs -r -n1 cp --remove-destination /home/menphy/lede/scripts/config.guess; find /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/ -name config.sub | xargs -r chmod u+w; find /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/ -name config.sub | xargs -r -n1 cp --remove-destination /home/menphy/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5=ksmbd-3.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5=ksmbd-3.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DCONFIG_SMB_INSECURE_SERVER=1" CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5=ksmbd-3.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DCONFIG_SMB_INSECURE_SERVER=1" CPPFLAGS="-I/home/menphy/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/menphy/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/menphy/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/menphy/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/menphy/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.built
touch /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.built_check
make -C /home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.52 KCFLAGS="-fmacro-prefix-map=/home/menphy/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/menphy/lede/staging_dir/host/include -I/home/menphy/lede/staging_dir/hostpkg/include -I/home/menphy/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jul  3 09:45:04 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/menphy/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.52 M="/home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[4]: Entering directory '/home/menphy/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.52'
[... cut ...]
make[3]: Leaving directory '/home/menphy/lede/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.40#0.11#0.48
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/menphy/lede'
make[1]: *** [package/Makefile:110: /home/menphy/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/menphy/lede'
make: *** [/home/menphy/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-07-05:
自己看另外一个帖子 我又说怎么解决

closed 2022-07-05 by aiamadeus