coolsnowwolf/lede #9694: aarch64_cortex-a53+crypto 编译 mbedtls出错

sKyissKy, 2022-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mbedtls编译出错

### (2) 路由器型号和固件版本

A：ipq60xx

### (3) 详细日志

A：
make[3]: Entering directory '/home/ubuntu/lede/package/libs/mbedtls'
mkdir -p /home/ubuntu/lede/dl
SHELL= flock /home/ubuntu/lede/tmp/.mbedtls-2.28.0.tar.gz.flock -c '  	/home/ubuntu/lede/scripts/download.pl "/home/ubuntu/lede/dl" "mbedtls-2.28.0.tar.gz" "6519579b836ed78cc549375c7c18b111df5717e86ca0eeff4cb64b2674f424cc" "" "https://codeload.github.com/ARMmbed/mbedtls/tar.gz/v2.28.0?"    '
rm -f /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/.built
touch /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/.built_check
MAKEFLAGS="" /home/ubuntu/lede/staging_dir/host/bin/ninja  -j1 -C /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0 
ninja: Entering directory `/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0'
[1/299] Building C object library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o
FAILED: library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o 
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc  -I/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/include -I/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0=mbedtls-2.28.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -O2 -MD -MT library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -MF library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o.d -o library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -c /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c: In function 'mbedtls_armv8ce_aes_crypt_ecb':
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:66:13: note: called from here
         x = vaeseq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
[... 30 lines cut ...]
                 ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:79:13: note: called from here
         x = vaesdq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:30:
/home/ubuntu/lede/staging_dir/toolchain-aarch64_cortex-a53+crypto_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/library/armv8ce_aes.c:79:13: note: called from here
         x = vaesdq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:179: /home/ubuntu/lede/build_dir/target-aarch64_cortex-a53+crypto_musl/mbedtls-2.28.0/.built] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.10#0.35#0.43
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53+crypto_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:230：world] 错误 2

sKyissKy, 2022-07-11:
> [immortalwrt/immortalwrt#718](https://github.com/immortalwrt/immortalwrt/discussions/718) 看这个，我之前遇到过解决的

这个禁用掉以后aarch64的AES硬件加速就没有了

zhangguanzhang, 2022-07-11:
> 

这个我看op官方的包里没这个选项，https://github.com/openwrt/openwrt/blob/openwrt-21.02/package/libs/mbedtls/Makefile 是最新的包里才有 `LIBMBEDTLS_HAVE_ARMV8CE_AES`  ， 新的包里才有这个，不过这个是包的问题，估计还得包的人看下

closed 2022-10-31 by aiamadeus

sbwml, 2023-01-23:
哈哈，这个 issues 应该是过时了。太尴尬了，刚刚也遇到了同样的问题。

这个问题出现的原因是 TARGET_OPTIMIZATIO 被覆盖了，它必须包含 -march=armv8/xxx-a`+crypto`

另外它还有延伸的玄学问题，假全模块编译 `CONFIG_ALL_KMODS=y` 或 使用 glibc  `CONFIG_LIBC_USE_GLIBC=y`

这个东西还会报隐式声明函数错误，这种情况只需要给 `library/gcm.c` 添加 `#include "mbedtls/armv8ce_aes.h"` 头文件就可以愉快编译过去了 https://github.com/sbwml/r4s_build_script/commit/1a5098be8ad9e978551d839e025c5a3ffbf8de53#diff-465650b0abf0fe9f93554fa938bfe3a8a01b5278b2b46464141439cca8d4250eR384

AES硬件加速继续愉快！！！