coolsnowwolf/lede #9695: ksmd的错误吗？这个怎么解决？

qqsir-dev, 2022-07-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/kernel/ksmbd'
mkdir -p /home/runner/work/AutoLede/AutoLede/openwrt/dl
SHELL= flock /home/runner/work/AutoLede/AutoLede/openwrt/tmp/.ksmbd-3.4.5.tar.gz.flock -c '  	/home/runner/work/AutoLede/AutoLede/openwrt/scripts/download.pl "/home/runner/work/AutoLede/AutoLede/openwrt/dl" "ksmbd-3.4.5.tar.gz" "2873c8ba1027fc5b04c5f5344804ef1469ad7019a033456c16ca4aa3f2c161f0" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.4.5?"    '
rm -f /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.built
touch /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.built_check
make -C /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.52 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/include -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/hostpkg/include -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jul  3 09:45:04 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.52 M="/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[3]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.52'
  CC [M]  /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/unicode.o
In file included from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/vfs.h:16,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/vfs_cache.h:16,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/glob.h:15,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/unicode.c:12:
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h: In function 'posix_acl_uid_translate':
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h:227:9: error: implicit declaration of function 'kuid_into_mnt'; did you mean 'i_uid_into_mnt'? [-Werror=implicit-function-declaration]
  kuid = kuid_into_mnt(mnt_userns, pace->e_uid);
         ^~~~~~~~~~~~~
         i_uid_into_mnt
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h:227:7: error: incompatible types when assigning to type 'kuid_t' {aka 'struct <anonymous>'} from type 'int'
  kuid = kuid_into_mnt(mnt_userns, pace->e_uid);
       ^
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h: In function 'posix_acl_gid_translate':
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h:247:9: error: implicit declaration of function 'kgid_into_mnt'; did you mean 'i_gid_into_mnt'? [-Werror=implicit-function-declaration]
  kgid = kgid_into_mnt(mnt_userns, pace->e_gid);
         ^~~~~~~~~~~~~
         i_gid_into_mnt
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/smbacl.h:247:7: error: incompatible types when assigning to type 'kgid_t' {aka 'struct <anonymous>'} from type 'int'
  kgid = kgid_into_mnt(mnt_userns, pace->e_gid);
       ^
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:288: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/unicode.o] Error 1
make[3]: *** [Makefile:1871: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5] Error 2
make[3]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.52'
make[2]: *** [Makefile:76: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.5/.built] Error 2
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.81#0.25#1.04
    ERROR: package/kernel/ksmbd failed to build.
make[1]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: package/compile] Error 2
Error: Process completed with exit code 2.

### (2) 路由器型号和固件版本

A：软路由

### (3) 详细日志

A：

WYC-2020, 2022-07-05:
更新ksmbd

WYC-2020, 2022-07-05:
[02-fix-ksmbd-build-with-kernel-5.15.patch.txt](https://github.com/coolsnowwolf/lede/files/9042924/02-fix-ksmbd-build-with-kernel-5.15.patch.txt)
这个补丁放ksmbd 目录，几个吧后缀改为patch @[qqsir-dev](https://github.com/qqsir-dev)

qqsir-dev, 2022-07-05:
谢谢各位！

closed 2022-07-05 by aiamadeus